bdring/FluidNC #1444: Cannot configure stepper motor with GPIO for step and I2SO for direction in config.yaml (ESP32 Dev)

OmidAyar, 2025-02-02:
### Wiki Search Terms

I'm working with an ESP32 Dev board and trying to configure a stepper motor in config.yaml for FluidNC. Due to the limited number of pins on the ESP32 Dev board, I want to use a GPIO pin for the step signal and an I2SO pin for the direction signal. However, I am unable to configure this as FluidNC doesn't seem to allow both signals to be configured in this way.

I do not want to use the I2SO pins for the step signal, but instead for the direction pin. I would also be willing to assist with setting up the ESP32 S3 if needed, as I am working on a project that requires the increased number of pins on this microcontroller.

Is it possible to configure the stepper motor this way, and if so, how can I achieve this in the config.yaml file?

Thank you for your help!

### Controller Board

4-pack

### Machine Description

this a project for myself

### Input Circuits

```yaml

```

### Configuration file

```yaml
board: esp32dev
name: Default (Test Drive no I/O)
meta: 
stepping:
  engine: RMT
  idle_ms: 255
  pulse_us: 4
  dir_delay_us: 0
  disable_delay_us: 0
  segments: 12

  

i2so:
  bck_pin: gpio.27
  data_pin: gpio.12
  ws_pin: gpio.14

spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

sdcard:
  cs_pin: gpio.5
  card_detect_pin: NO_PIN
  frequency_hz: 8000000

kinematics:
  Cartesian:

axes:
  shared_stepper_disable_pin: NO_PIN
  x:
    steps_per_mm: 100.000
    max_rate_mm_per_min: 5000.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 300.000
    soft_limits: false
    homing:
      cycle: 2
      positive_direction: true
      mpos_mm: 0.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 200.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.36
      limit_pos_pin: gpio.39
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1.000
      
      standard_stepper:
        step_pin: gpio.32
        direction_pin: i2so.1
        disable_pin: NO_PIN
             
       

  y:
    steps_per_mm: 100.000
    max_rate_mm_per_min: 5000.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 300.000
    soft_limits: false
    homing:
      cycle: 2
      positive_direction: true
      mpos_mm: 0.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 200.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.34
      limit_pos_pin: gpio.35
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1.000
      
      standard_stepper:
        step_pin: gpio.33
        direction_pin: i2so.2
        disable_pin: NO_PIN

  z:
    steps_per_mm: 100.000
    max_rate_mm_per_min: 5000.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 300.000
    soft_limits: false
    homing:
      cycle: 2
      positive_direction: true
      mpos_mm: 0.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 200.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1.000
      
      standard_stepper:
        step_pin: gpio.25
        direction_pin: i2so.3
        disable_pin: NO_PIN

control:
  safety_door_pin: NO_PIN
  reset_pin: NO_PIN
  feed_hold_pin: NO_PIN
  cycle_start_pin: NO_PIN
  macro0_pin: NO_PIN
  macro1_pin: NO_PIN
  macro2_pin: NO_PIN
  macro3_pin: NO_PIN
  fault_pin: NO_PIN
  estop_pin: NO_PIN

coolant:
  flood_pin: NO_PIN
  mist_pin: NO_PIN
  delay_ms: 0

probe:
  pin: NO_PIN
  toolsetter_pin: NO_PIN
  check_mode_start: true
  hard_stop: false

macros:
  startup_line0:
  startup_line1:
  Macro0:
  Macro1:
  Macro2:
  Macro3:
  after_homing:
  after_reset:
  after_unlock:

start:
  must_home: false
  deactivate_parking: false
  check_limits: true

parking:
  enable: false
  axis: Z
  target_mpos_mm: -5.000000
  rate_mm_per_min: 800.000000
  pullout_distance_mm: 5.000000
  pullout_rate_mm_per_min: 250.000000

user_outputs:
  analog0_pin: NO_PIN
  analog1_pin: NO_PIN
  analog2_pin: NO_PIN
  analog3_pin: NO_PIN
  analog0_hz: 5000
  analog1_hz: 5000
  analog2_hz: 5000
  analog3_hz: 5000
  digital0_pin: NO_PIN
  digital1_pin: NO_PIN
  digital2_pin: NO_PIN
  digital3_pin: NO_PIN
  digital4_pin: NO_PIN
  digital5_pin: NO_PIN
  digital6_pin: NO_PIN
  digital7_pin: NO_PIN

user_inputs:
  analog0_pin: NO_PIN
  analog1_pin: NO_PIN
  analog2_pin: NO_PIN
  analog3_pin: NO_PIN
  digital0_pin: NO_PIN
  digital1_pin: NO_PIN
  digital2_pin: NO_PIN
  digital3_pin: NO_PIN
  digital4_pin: NO_PIN
  digital5_pin: NO_PIN
  digital6_pin: NO_PIN
  digital7_pin: NO_PIN

arc_tolerance_mm: 0.002000
junction_deviation_mm: 0.010000
verbose_errors: true
report_inches: false
enable_parking_override_control: false
use_line_numbers: false
planner_blocks: 16
```

### Startup Messages

```text
[MSG:INFO: FluidNC v3.9.4 https://github.com/bdring/FluidNC]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.7-dirty]
[MSG:INFO: Local filesystem type is littlefs]
[MSG:INFO: Configuration file:config.yaml]
[MSG:ERR: Validation error at /axes/x/motor0/standard_stepper: Step pin must be an I2SO pin]
[MSG:ERR: Validation error at /axes/y/motor0/standard_stepper: Step pin must be an I2SO pin]
[MSG:ERR: Validation error at /axes/z/motor0/standard_stepper: Step pin must be an I2SO pin]
[MSG:INFO: Machine Default (Test Drive no I/O)]
[MSG:INFO: Board esp32dev]
[MSG:INFO: I2SO BCK:gpio.27 WS:gpio.14 DATA:gpio.12Min Pulse:2us]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
[MSG:INFO: SD Card cs_pin:gpio.5 detect:NO_PIN freq:8000000]
[MSG:INFO: Stepping:I2S_STREAM Pulse:4us Dsbl Delay:0us Dir Delay:0us Idle Delay:255ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Axis X (-300.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:gpio.32 Dir:I2SO.1 Disable:NO_PIN]
[MSG:INFO:  X Neg Limit gpio.36]
[MSG:INFO:  X Pos Limit gpio.39]
[MSG:INFO: Axis Y (-300.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:gpio.33 Dir:I2SO.2 Disable:NO_PIN]
[MSG:INFO:  Y Neg Limit gpio.34]
[MSG:INFO:  Y Pos Limit gpio.35]
[MSG:INFO: Axis Z (-300.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:gpio.25 Dir:I2SO.3 Disable:NO_PIN]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: STA SSID is not set]
[MSG:INFO: AP SSID FluidNC IP 192.168.0.1 mask 255.255.255.0 channel 1]
[MSG:INFO: AP started]
[MSG:INFO: WiFi on]
[MSG:INFO: Captive Portal Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]
```

### What happened?

Cannot configure stepper motor with GPIO for step and I2SO for direction in config.yaml (ESP32 Dev)

bdring, 2025-02-02:
You cannot use i2so pins with the RMT engine type. You should use all `gpio` or all `i2so` pins for the motors. 

```yaml
board: esp32dev
name: Default (Test Drive no I/O)
meta: 
stepping:
  engine: RMT
  idle_ms: 255
```

MitchBradley, 2025-02-02:
Dev boards do not have the external shift registers that are necessary for I2SO stepping.

bdring, 2025-02-03:
We have some work going for an I/O extender. It works for inputs, digital I/O and PWM.  

- [Discord Discussion](https://discord.com/channels/780079161460916227/1333171112883912715)
- [Wiki page](http://wiki.fluidnc.com/en/hardware/official/STM32_IO_Extender)

closed 2025-02-11 by bdring