bdring/FluidNC #1446: Problem: TMC Driver not detected

NEWTech-Creative, 2025-02-04:
### Wiki Search Terms

 TMC Driver not detected, TMC5160, r_sense_ohms

### Controller Board

Experimental board: Root Controller Pro (6axis TMC5160 board). 

### Machine Description

6 motors attached for testing- no machine

### Input Circuits

```yaml
No inputs at this stage
```

### Configuration file

```yaml
board: Root Controller STEP
name: TMC5160 XYZABC
stepping:
  engine: I2S_STREAM
  idle_ms: 255
  pulse_us: 4
  dir_delay_us: 0
  disable_delay_us: 0

axes:
  shared_stepper_disable_pin: I2SO.4:low
  x:
    steps_per_mm: 400  
    max_rate_mm_per_min: 5000.000
    acceleration_mm_per_sec2: 50.000
    max_travel_mm: 800
    soft_limits: true
    homing:
      cycle: 2
      positive_direction: true
      mpos_mm: 0.000
      feed_mm_per_min: 500.000
      seek_mm_per_min: 200.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: gpio.39
      hard_limits: false
      pulloff_mm: 1.500
      tmc_5160:
        cs_pin: i2so.2
        r_sense_ohms: 0.05
        run_amps: 1.0
        hold_amps: 0.5
        microsteps: 32
        stallguard: 0
        stallguard_debug: false
        toff_disable: 0
        toff_stealthchop: 5
        toff_coolstep: 3
        run_mode: StealthChop
        homing_mode: StealthChop
        use_enable: false
        step_pin: I2SO.1
        direction_pin: I2SO.0:low
[... 307 lines cut ...]
After troubleshooting, I rolled back to previous versions of FluidNC and found that v3.8.2 was the last working version where all 6 of the TMC5160 drivers functioned correctly. I’m using a 6-axis setup.

I’ve tried several solutions, including:
    •    Erasing and reinstalling FluidNC
    •    Running $MI
    •    Testing different variations of the YAML configuration file

However, none of these attempts have resolved the issue.

Example error: 

[MSG: ERR: X Axis TMC driver not detected - expected 0x30 got 0x0]
[MSG: INFO: Y Axis driver test passed]
[MSG: ERR: Z Axis TMC driver not detected - expected 0x30 got 0xff]
[MSG: ERR: A Axis TMC driver not detected - expected 0x30 got 0xfe]
[MSG: INFO: B Axis driver test passed]
[MSG: INFO: C Axis driver test passed]

I’ve had the board for a year and no issues until updated to newer firmware. 

Any suggestions on what might be causing this?

### GCode File

N/A

bdring, 2025-02-04:
Have you contacted the supplier of the controller?

NEWTech-Creative, 2025-02-05:
Thank you for your guidance, Bart.

I have already discussed this issue with RootCNC and suggested posting here since the board has been working reliably for the past year but started encountering issues with the later versions.

To troubleshoot, I updated my YAML configuration to include the following:

i2so:
  bck_pin: gpio.22
  data_pin: gpio.12
  ws_pin: gpio.21
  min_pulse_us: 4

However, after adding this configuration, all TMC drivers were reported as non-responsive, returning the following errors:

[MSG:ERR: X Axis TMC driver not detected - expected 0x30, got 0xff]
[MSG:ERR: Y Axis TMC driver not detected - expected 0x30, got 0xff]
[MSG:ERR: Z Axis TMC driver not detected - expected 0x30, got 0x0]
[MSG:ERR: A Axis TMC driver not detected - expected 0x30, got 0x0]
[MSG:ERR: B Axis TMC driver not detected - expected 0x30, got 0xff]
[MSG:ERR: C Axis TMC driver not detected - expected 0x30, got 0xff]

I then experimented with different min_pulse_us values:
- 4 and 2 µs: The X-axis motor driver seemed to function, but the other axes only moved slightly before becoming unresponsive.
- 1 µs: All TMC driver tests passed, but there was no motor movement at all.

Finally, I rolled back to v3.8.2, and everything returned to normal—all TMC drivers passed successfully, and movement was restored without errors.

This board uses the same SN74HCS595DR shift registers as the Root Controller ISO, so I suspect the issue might be related to firmware changes affecting the driver communication.

Let me know if I've missed anything in testing or require more information?

bdring, 2025-02-05:
Try sending `$Motors/Init`

http://wiki.fluidnc.com/en/features/commands_and_settings#motorsinit-or-mi

If that works it could be related to a timing problem. Also try I2S_STATIC instead of I2S_STREAM

MitchBradley, 2025-02-05:
We don't have access to a Root Controller so debugging it is going to be very difficult.  I for one do not have time to debug by remote control with a human intermediary.  If I had a Root Controller and they sponsored the project that they make money from, I would be happy to try and figure out what is going wrong.

NEWTech-Creative, 2025-02-07:
Thank you for your response, Bart.

I have tested using I2S_STATIC as suggested, but unfortunately, it did not resolve the issue. I agree that this appears to be a timing-related problem, possibly linked to changes in TMC behavior in FluidNC 3.9.x and beyond.

As you know, I have always supported and promoted FluidNC purely out of passion and have never sought direct assistance beyond community discussions. I truly appreciate the time and effort you and the team dedicate to its development.

I understand that debugging hardware remotely can be challenging. Given the circumstances, I will continue using an older version of FluidNC that remains stable for my setup.

Thank you again for your help.

3404gerber, 2025-02-07:
Hi all,

It's been a couple of months now that I switched to Linuxcnc with my TMC5160 drivers, but I spent many hours reading the FluidNC source code last years. I was wondering: could it be a timing problem? I know from the datasheet that the CS pin has to be low for a given time before you can start to send data on the SPI bus. If you use a "slow" I2S pin as chip select, maybe there should be a short wait time in the code before you start to send the SPI message.

MitchBradley, 2025-02-07:
It is probably something like that, but I for one don't have time to try stuff without access to hardware to test on a short debug cycle. If I had the hardware I could just scope it and see what is wrong.  But making code changes and having someone else tell me if it worked, whenever they can get around to it, takes more time than I can spare given all the other things I am working on NM.

3404gerber, 2025-02-08:
I understand, I was rather thinking that NEWTech could give it a try.

msetilley, 2025-02-13:
I have the same issue with a 6-Pack controller board. Upgraded from 3.7.10 to 3.9.4 and started seeing the same problem with TMC2130 drivers. My 6-pack controller is a bit older (V1P5). I have a set of 5160 drivers I can try to see if they also have the same issue, but this machine worked fine (I used it just prior to the update in case I screwed something up).

Initial config
```yaml
board: 6 Pack
name: TMC2130 XYZ
meta: 
stepping:
  engine: I2S_STATIC
  idle_ms: 250
  pulse_us: 4
  dir_delay_us: 0
  disable_delay_us: 0
  segments: 12

spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

axes:
  shared_stepper_disable_pin: NO_PIN
  x:
    steps_per_mm: 1600.000
    max_rate_mm_per_min: 1500.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 374.000
    soft_limits: true
    homing:
      cycle: 2
      positive_direction: true
      mpos_mm: 0.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 200.000
      seek_scaler: 1.100
      feed_scaler: 1.100
    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: gpio.33:low
      hard_limits: true
      pulloff_mm: 2.000
      tmc_2130:
        cs_pin: i2so.3
        r_sense_ohms: 0.110
        run_amps: 1.00
        hold_amps: 0.750
        microsteps: 16
        stallguard: 0
        stallguard_debug: false
        toff_disable: 0
        toff_stealthchop: 5
        toff_coolstep: 3
        run_mode: CoolStep
        homing_mode: CoolStep
        use_enable: false
        step_pin: I2SO.2
        direction_pin: I2SO.1
        disable_pin: I2SO.0

  y:
    steps_per_mm: 1600.000
    max_rate_mm_per_min: 1500.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 392.000
    soft_limits: true
    homing:
      cycle: 2
      positive_direction: true
      mpos_mm: 0.000
[... 181 lines cut ...]
  data_pin: gpio.21
  ws_pin: gpio.17
  min_pulse_us: 4
```
This didn't help either, it seemed worse since it was no longer random and no drivers passed at all no matter how many times I repeated it.

Issue is the same, drivers randomly fail their test, although I have had all three pass and work for long enough to home / jog the machine around but its uncommon.

```
$MI
[MSG:INFO: X Axis driver test passed]
[MSG:ERR: Y Axis TMC driver not detected - expected 0x11 got 0xfc]
[MSG:ERR: Z Axis TMC driver not detected - expected 0x11 got 0xff]
ok
$MI
[MSG:ERR: X Axis TMC driver not detected - expected 0x11 got 0xfe]
[MSG:INFO: Y Axis driver test passed]
[MSG:ERR: Z Axis TMC driver not detected - expected 0x11 got 0xff]
ok
$MI
[MSG:INFO: X Axis driver test passed]
[MSG:INFO: Y Axis driver test passed]
[MSG:INFO: Z Axis driver test passed]
ok
```

msetilley, 2025-02-14:
I installed v3.8.3 to verify the original post and confirmed TMC2130 drivers all pass and everything works as expected. 

The v3.8.3 config had min_pulse_us removed and some other config that caused a startup error, config used is below.

```yaml
board: 6 Pack
name: TMC2130 XYZ
meta: 
stepping:
  engine: I2S_STATIC
  idle_ms: 250
  pulse_us: 4
  dir_delay_us: 0
  disable_delay_us: 0
  segments: 12

spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

axes:
  shared_stepper_disable_pin: NO_PIN
  x:
    steps_per_mm: 1600.000
    max_rate_mm_per_min: 1500.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 374.000
    soft_limits: true
    homing:
      cycle: 2
      positive_direction: true
      mpos_mm: 0.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 200.000
      seek_scaler: 1.100
      feed_scaler: 1.100
    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: gpio.33:low
      hard_limits: true
      pulloff_mm: 2.000
      tmc_2130:
        cs_pin: i2so.3
        r_sense_ohms: 0.110
        run_amps: 1.00
        hold_amps: 0.750
        microsteps: 16
        stallguard: 0
        stallguard_debug: false
        toff_disable: 0
        toff_stealthchop: 5
        toff_coolstep: 3
        run_mode: CoolStep
        homing_mode: CoolStep
        use_enable: false
        step_pin: I2SO.2
        direction_pin: I2SO.1
        disable_pin: I2SO.0

  y:
    steps_per_mm: 1600.000
    max_rate_mm_per_min: 1500.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 392.000
    soft_limits: true
    homing:
      cycle: 2
      positive_direction: true
      mpos_mm: 0.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 200.000
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: gpio.32:low
      hard_limits: true
      pulloff_mm: 2.000
      tmc_2130:
        cs_pin: i2so.6
        r_sense_ohms: 0.110
        run_amps: 1.00
        hold_amps: 0.750
        microsteps: 16
        stallguard: 0
        stallguard_debug: false
        toff_disable: 0
        toff_stealthchop: 5
        toff_coolstep: 3
        run_mode: CoolStep
        homing_mode: CoolStep
        use_enable: false
        step_pin: I2SO.5
        direction_pin: I2SO.4
        disable_pin: I2SO.7

  z:
    steps_per_mm: 1600.000
    max_rate_mm_per_min: 1500.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 100.000
    soft_limits: true
    homing:
      cycle: 2
      positive_direction: true
      mpos_mm: 0.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 200.000
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: gpio.35:low
      hard_limits: false
      pulloff_mm: 2.000
      tmc_2130:
        cs_pin: i2so.11
        r_sense_ohms: 0.110
        run_amps: 1.00
        hold_amps: 0.750
        microsteps: 16
        stallguard: 0
        stallguard_debug: false
        toff_disable: 0
        toff_stealthchop: 5
        toff_coolstep: 3
        run_mode: CoolStep
        homing_mode: CoolStep
        use_enable: false
        step_pin: I2SO.10
        direction_pin: I2SO.9
        disable_pin: I2SO.8

i2so:
  bck_pin: gpio.22
  data_pin: gpio.21
  ws_pin: gpio.17

sdcard:
  cs_pin: NO_PIN
  card_detect_pin: NO_PIN
  frequency_hz: 8000000

kinematics:
  Cartesian:

control:
  safety_door_pin: NO_PIN
  reset_pin: NO_PIN
  feed_hold_pin: NO_PIN
  cycle_start_pin: NO_PIN
  macro0_pin: NO_PIN
  macro1_pin: NO_PIN
  macro2_pin: NO_PIN
  macro3_pin: NO_PIN
  fault_pin: NO_PIN
  estop_pin: NO_PIN

coolant:
  flood_pin: NO_PIN
  mist_pin: NO_PIN
  delay_ms: 0

probe:
  pin: gpio.34
  check_mode_start: true

macros:
  startup_line0:
  startup_line1:
  Macro0:
  Macro1:
  Macro2:
  Macro3:
  after_homing:
  after_reset:
  after_unlock:

start:
  must_home: true

YL620:
  uart:
    txd_pin: gpio.26
    rxd_pin: gpio.16
    rts_pin: gpio.4
    baud: 9600
    mode: 8N1
  modbus_id: 1
  tool_num: 0
  speed_map: 0=0% 0=25% 6000=25% 24000=100%

parking:
  enable: false

arc_tolerance_mm: 0.002000
junction_deviation_mm: 0.010000
verbose_errors: true
report_inches: false
enable_parking_override_control: false
use_line_numbers: false
planner_blocks: 16
```

bdring, 2025-02-14:
I think I have enough TMC2130s to try some experiments on a 6 pack. I'll look at the timing on a logic analyzer.

My desk is covered in other setups right now. This may take a few days.

bdring, 2025-02-14:
I created a 5x 2130 setup on a 6 pack. On the latest release I get similar results to you. Sometimes all work and sometimes 1 or 2 don't pass.

3.8.3 works fine. I will put it on a logic analyzer when I get time.

bdring, 2025-02-14:
As a complete hack to test the timing I added a delay when enabling the CS.

TMC2130Stepper.cpp in the TMCStepper library.

```cpp
__attribute__((weak))
void TMC2130Stepper::switchCSpin(bool state) {
  digitalWrite(_pinCS, state);
  if (!state) {
      delayMicroseconds(10);
  }
}
```

Now it works every time.

```
$mi
[MSG:INFO: X Axis driver test passed]
[MSG:INFO: Y Axis driver test passed]
[MSG:INFO: Z Axis driver test passed]
[MSG:INFO: A Axis driver test passed]
[MSG:INFO: B Axis driver test passed]
ok
$mi
[MSG:INFO: X Axis driver test passed]
[MSG:INFO: Y Axis driver test passed]
[MSG:INFO: Z Axis driver test passed]
[MSG:INFO: A Axis driver test passed]
[MSG:INFO: B Axis driver test passed]
ok
$mi
[MSG:INFO: X Axis driver test passed]
[MSG:INFO: Y Axis driver test passed]
[MSG:INFO: Z Axis driver test passed]
[MSG:INFO: A Axis driver test passed]
[MSG:INFO: B Axis driver test passed]
ok
$mi
[MSG:INFO: X Axis driver test passed]
[MSG:INFO: Y Axis driver test passed]
[MSG:INFO: Z Axis driver test passed]
[MSG:INFO: A Axis driver test passed]
[MSG:INFO: B Axis driver test passed]
ok
$mi
[MSG:INFO: X Axis driver test passed]
[MSG:INFO: Y Axis driver test passed]
[MSG:INFO: Z Axis driver test passed]
[MSG:INFO: A Axis driver test passed]
[MSG:INFO: B Axis driver test passed]
ok
$mi
[MSG:INFO: X Axis driver test passed]
[MSG:INFO: Y Axis driver test passed]
[MSG:INFO: Z Axis driver test passed]
[MSG:INFO: A Axis driver test passed]
[MSG:INFO: B Axis driver test passed]
ok
```

Ideally we don't modify the library and only add the delay if it is a i2so pin. The function is weak, so there probably better ways to do it.

MitchBradley, 2025-02-14:
The override is in src/Pins/PinMapper.cpp:
```c++
void IRAM_ATTR digitalWrite(pinnum_t pin, uint8_t val) {
...
        thePin->synchronousWrite(val);
```
For I2SO, that delegates to src/Pins/I2SOPinDetail.cpp:
```c++
void IRAM_ATTR I2SOPinDetail::synchronousWrite(int high) {
...
      i2s_out_write(_index, _inverted ^ (bool)high);
      i2s_out_delay();
```
which invokes esp32/i2s_engine.c:
```c
void i2s_out_delay() {
    // Depending on the timing, it may not be reflected immediately,
    // so wait twice as long just in case.
    uint32_t wait_counts = timer_running ? FIFO_THRESHOLD + FIFO_RELOAD : 2;
    delay_us(i2s_frame_us * wait_counts);
}
```
I tried to calculate the delay based on the depth of the FIFO but maybe I made a mistake, or maybe a little extra slop is needed.  It would be nice to know, based on logic analyzer measurements, whether the existing delay is sufficient for the write to clear the FIFO.

bdring, 2025-02-14:
With 3.8.3 the timing does not change much.

[image]

With 3.9.4 it is all over the place. This was a particularly bad one.

```
$MI
[MSG:ERR: X Axis TMC driver not detected - expected 0x11 got 0xff]
[MSG:ERR: Y Axis TMC driver not detected - expected 0x11 got 0xfc]
[MSG:INFO: Z Axis driver test passed]
[MSG:INFO: A Axis driver test passed]
[MSG:ERR: B Axis TMC driver not detected - expected 0x11 got 0xff]
ok
```

[image]

MitchBradley, 2025-02-14:
It is hard to compare those two with their differing time bases.

MitchBradley, 2025-02-14:
Also why is the clock not toggling consistently?  Too slow sampling?

MitchBradley, 2025-02-14:
If it is missing clocks due to slow sampling, I wonder if it is also missing data, in which case the measurements are questionable.

bdring, 2025-02-14:
I was capturing on the CS of the first chip. This one is the least likely to fail. I moved it to the second chip for this plot and set the rate at 20MHz. It is obviously out of sync here.

[image]

bdring, 2025-02-14:
I can attach the capture file if you want to load it in DS Scope.

MitchBradley, 2025-02-14:
What happens if you change the 2 in the next line to, say, 6?  (esp32/i2s_engine.c) ? 
```c
   uint32_t wait_counts = timer_running ? FIFO_THRESHOLD + FIFO_RELOAD : 2;
```

3404gerber, 2025-02-22:
> As a complete hack to test the timing I added a delay when enabling the CS.
> 
> TMC2130Stepper.cpp in the TMCStepper library.
> 
> __attribute__((weak))
> void TMC2130Stepper::switchCSpin(bool state) {
>   digitalWrite(_pinCS, state);
>   if (!state) {
>       delayMicroseconds(10);
>   }
> }
> 
> Now it works every time.
> ...

> 
> Ideally we don't modify the library and only add the delay if it is a i2so pin. The function is weak, so there probably better ways to do it.

Why not simply add a delay in esp32/tmc_spi.cpp, after each switchCSpin? Maybe not the most elegant way to do it, but bullet proof. 

>void TMC2130Stepper::write(uint8_t reg, uint32_t data) {
>    log_verbose("TMC reg " << to_hex(reg) << " write " << to_hex(data));
>    tmc_spi_bus_setup();
>
>    switchCSpin(0);
> add a short delay here?
>    tmc_spi_rw_reg(reg | 0x80, data, link_index);
> and here?
>    switchCSpin(1);
}

Another question: do I understand it correctly, if I read in the file esp32/tmc_spi_support.c that the spi bus is initialised each time with 2MHz frequency? Here is the code:

>if (clk_reg_val == 0) {
>        spi_ll_master_cal_clock(SPI_LL_PERIPH_CLK_FREQ, 2000000, 128, &clk_reg_val);
>    }

Because on the other side, in theTMC5160Driver.cpp file, there is a part of code "saying" that in case we use a I2S CS pin, the SPI speed should be only 100kHz:

>        // use slower speed if I2S
>        if (_cs_pin.capabilities().has(Pin::Capabilities::I2S)) {
>            tmc5160->setSPISpeed(_spi_freq);
>        }

I remember that I also had communication issues in the beginning, but it was with increased number of daisy chained drivers, and I ended up overwritting the value in the tmc_spi_support.c file.

abcpibot, 2025-04-05:
Hello everyone, I'm glad to see this discussion thread. I'm also encountering the same issue. When using version 3.9.4, sometimes the TMC5160 is detected and sometimes it isn't. After reverting to version 3.8.3, the TMC5160 works normally, but I still ran into another problem: after using the TMC5160 driver, my SD card stops responding（3.9.4 and 3.8.3 same）. If I switch back to the original en, dir, clk, and gnd setup, everything works fine again. Below is my YAML test file.

```
board: PiBotV49PB
name: TMC5160_Test
stepping:
  engine: I2S_STATIC
  idle_ms: 254
  pulse_us: 8
  dir_delay_us: 1
  disable_delay_us: 0

axes:
  shared_stepper_disable_pin: NO_PIN
  x:
    steps_per_mm: 800.000
    max_rate_mm_per_min: 5000.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 300.000
    soft_limits: false
    homing:
      cycle: 2
      positive_direction: false
      mpos_mm: 150.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 200.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.33:low
      limit_pos_pin: gpio.32:low
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1.000      
      tmc_5160:        
        r_sense_ohms: 0.1
        run_amps: 0.75
        hold_amps: 0.1
        microsteps: 32
        stallguard: 0
        stallguard_debug: false
        toff_disable: 0
        toff_stealthchop: 5
        toff_coolstep: 3
        run_mode: CoolStep
        homing_mode: CoolStep
        use_enable: false
        cs_pin: i2so.16
        step_pin: I2SO.2
        direction_pin: I2SO.1
        disable_pin: I2SO.0

        
     #####ms1 ms2 use the board capnut
        
  y:
    steps_per_mm: 800.000
    max_rate_mm_per_min: 5000.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 300.000
    soft_limits: false
    homing:
      cycle: 2
      positive_direction: true
      mpos_mm: 150.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 200.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.35:low
      limit_pos_pin: gpio.34:low
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1.000      
      tmc_5160:        
        r_sense_ohms: 0.1
        run_amps: 0.75
        hold_amps: 0.1
        microsteps: 32
        stallguard: 0
        stallguard_debug: false
        toff_disable: 0
        toff_stealthchop: 5
        toff_coolstep: 3
        run_mode: StealthChop
        homing_mode: StealthChop
        use_enable: false
        cs_pin: i2so.18
        step_pin: I2SO.5
        direction_pin: I2SO.4
        disable_pin: I2SO.7

  z:
    steps_per_mm: 800.000
    max_rate_mm_per_min: 5000.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 300.000
    soft_limits: false
    homing:
      cycle: 1
      positive_direction: true
      mpos_mm: 150.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 800.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1.000
      tmc_5160:        
        r_sense_ohms: 0.075
        run_amps: 0.75
        hold_amps: 0.1
        microsteps: 32
        stallguard: 0
        stallguard_debug: false
        toff_disable: 0
        toff_stealthchop: 5
        toff_coolstep: 3
        run_mode: CoolStep
        homing_mode: CoolStep
        use_enable: false
        cs_pin: I2SO.17
        step_pin: I2SO.10
        direction_pin: I2SO.9
        disable_pin: I2SO.8

i2so:
  bck_pin: gpio.22
  data_pin: gpio.21
  ws_pin: gpio.17

spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

sdcard:
  card_detect_pin: NO_PIN
  cs_pin: gpio.5
  frequency_hz: 1000000

probe:
  pin: gpio.2:pu
  toolsetter_pin: NO_PIN
  check_mode_start: true
  hard_stop: false

start:
  must_home: false

# Begin relay
relay:
  direction_pin: NO_PIN
  output_pin: gpio.26
  enable_pin: NO_PIN
  disable_with_s0: false
  s0_with_disable: true
  spinup_ms: 0
  spindown_ms: 0
  tool_num: 0
  speed_map: 0=0.000% 0=100.000% 1=100.000%
  off_on_alarm: false

# Begin PWM  
pwm:
  pwm_hz: 5000
  direction_pin: NO_PIN
  output_pin: gpio.4
  enable_pin: NO_PIN
  disable_with_s0: false
  s0_with_disable: true
  spinup_ms: 0
  spindown_ms: 0
  tool_num: 1
  speed_map: 0=0.000% 10000=100.000%
  off_on_alarm: false
  
# Begin Laser 
Laser:
  pwm_hz: 5000
  output_pin: gpio.12
  enable_pin: NO_PIN
  disable_with_s0: false
  s0_with_disable: true
  tool_num: 2
  speed_map: 0=0.000% 255=100.000%
  off_on_alarm: true

# Begin 10V
10V:
  forward_pin: NO_PIN
  reverse_pin: NO_PIN
  pwm_hz: 5000
  output_pin: gpio.13
  enable_pin: NO_PIN
  direction_pin: NO_PIN
  disable_with_s0: false
  s0_with_disable: true
  spinup_ms: 0
  spindown_ms: 0
  tool_num: 3
  speed_map: 0=0.000% 1000=0.000% 24000=100.000%
  off_on_alarm: false
  
# Begin mpg

uart1:
  txd_pin: gpio.27
  rxd_pin: gpio.25
  rts_pin: NO_PIN
  cts_pin: NO_PIN
  baud: 115200
  mode: 8N1

uart_channel1:
  report_interval_ms: 75
  uart_num: 1
```

abcpibot, 2025-04-05:
Grbl 3.8 [FluidNC v3.8.3 (wifi) '$' for help]
$SD/LIST
[MSG:ERR: sdmmc_card_init failed code 0x107]
[MSG:ERR: filesystem error: SD card is inaccessible: ESP_ERR_TIMEOUT]
error:60
[MSG:ERR: Failed to mount device]

abcpibot, 2025-04-05:
I change the frequency_hz for SDCARD， but still not works， thanks

abcpibot, 2025-04-05:
> I installed v3.8.3 to verify the original post and confirmed TMC2130 drivers all pass and everything works as expected.
> 
> The v3.8.3 config had min_pulse_us removed and some other config that caused a startup error, config used is below.
> 
> board: 6 Pack
> name: TMC2130 XYZ
> meta: 
> stepping:
>   engine: I2S_STATIC
>   idle_ms: 250
>   pulse_us: 4
>   dir_delay_us: 0
>   disable_delay_us: 0
>   segments: 12
> 
> spi:
>   miso_pin: gpio.19
>   mosi_pin: gpio.23
>   sck_pin: gpio.18
> 
> axes:
>   shared_stepper_disable_pin: NO_PIN
>   x:
>     steps_per_mm: 1600.000
>     max_rate_mm_per_min: 1500.000
>     acceleration_mm_per_sec2: 100.000
>     max_travel_mm: 374.000
>     soft_limits: true
>     homing:
>       cycle: 2
>       positive_direction: true
>       mpos_mm: 0.000
>       feed_mm_per_min: 100.000
>       seek_mm_per_min: 200.000
>       seek_scaler: 1.100
>       feed_scaler: 1.100
>     motor0:
>       limit_neg_pin: NO_PIN
>       limit_pos_pin: NO_PIN
>       limit_all_pin: gpio.33:low
>       hard_limits: true
>       pulloff_mm: 2.000
>       tmc_2130:
>         cs_pin: i2so.3
>         r_sense_ohms: 0.110
>         run_amps: 1.00
>         hold_amps: 0.750
>         microsteps: 16
>         stallguard: 0
>         stallguard_debug: false
>         toff_disable: 0
>         toff_stealthchop: 5
>         toff_coolstep: 3
>         run_mode: CoolStep
>         homing_mode: CoolStep
>         use_enable: false
>         step_pin: I2SO.2
>         direction_pin: I2SO.1
>         disable_pin: I2SO.0
> 
>   y:
>     steps_per_mm: 1600.000
>     max_rate_mm_per_min: 1500.000
>     acceleration_mm_per_sec2: 100.000
>     max_travel_mm: 392.000
>     soft_limits: true
>     homing:
>       cycle: 2
>       positive_direction: true
>       mpos_mm: 0.000
>       feed_mm_per_min: 100.000
>       seek_mm_per_min: 200.000
>       seek_scaler: 1.100
>       feed_scaler: 1.100
> 
>     motor0:
>       limit_neg_pin: NO_PIN
>       limit_pos_pin: NO_PIN
>       limit_all_pin: gpio.32:low
>       hard_limits: true
>       pulloff_mm: 2.000
>       tmc_2130:
>         cs_pin: i2so.6
>         r_sense_ohms: 0.110
>         run_amps: 1.00
>         hold_amps: 0.750
>         microsteps: 16
>         stallguard: 0
>         stallguard_debug: false
>         toff_disable: 0
>         toff_stealthchop: 5
>         toff_coolstep: 3
>         run_mode: CoolStep
>         homing_mode: CoolStep
>         use_enable: false
>         step_pin: I2SO.5
>         direction_pin: I2SO.4
>         disable_pin: I2SO.7
> 
>   z:
>     steps_per_mm: 1600.000
>     max_rate_mm_per_min: 1500.000
>     acceleration_mm_per_sec2: 100.000
>     max_travel_mm: 100.000
>     soft_limits: true
>     homing:
>       cycle: 2
>       positive_direction: true
>       mpos_mm: 0.000
>       feed_mm_per_min: 100.000
>       seek_mm_per_min: 200.000
>       seek_scaler: 1.100
>       feed_scaler: 1.100
> 
>     motor0:
>       limit_neg_pin: NO_PIN
>       limit_pos_pin: NO_PIN
>       limit_all_pin: gpio.35:low
>       hard_limits: false
>       pulloff_mm: 2.000
>       tmc_2130:
>         cs_pin: i2so.11
>         r_sense_ohms: 0.110
>         run_amps: 1.00
>         hold_amps: 0.750
>         microsteps: 16
>         stallguard: 0
>         stallguard_debug: false
>         toff_disable: 0
>         toff_stealthchop: 5
>         toff_coolstep: 3
>         run_mode: CoolStep
>         homing_mode: CoolStep
>         use_enable: false
>         step_pin: I2SO.10
>         direction_pin: I2SO.9
>         disable_pin: I2SO.8
> 
> i2so:
>   bck_pin: gpio.22
>   data_pin: gpio.21
>   ws_pin: gpio.17
> 
> 
> sdcard:
>   cs_pin: NO_PIN
>   card_detect_pin: NO_PIN
>   frequency_hz: 8000000
> 
> kinematics:
>   Cartesian:
> 
> control:
>   safety_door_pin: NO_PIN
>   reset_pin: NO_PIN
>   feed_hold_pin: NO_PIN
>   cycle_start_pin: NO_PIN
>   macro0_pin: NO_PIN
>   macro1_pin: NO_PIN
>   macro2_pin: NO_PIN
>   macro3_pin: NO_PIN
>   fault_pin: NO_PIN
>   estop_pin: NO_PIN
> 
> coolant:
>   flood_pin: NO_PIN
>   mist_pin: NO_PIN
>   delay_ms: 0
> 
> probe:
>   pin: gpio.34
>   check_mode_start: true
> 
> macros:
>   startup_line0:
>   startup_line1:
>   Macro0:
>   Macro1:
>   Macro2:
>   Macro3:
>   after_homing:
>   after_reset:
>   after_unlock:
> 
> start:
>   must_home: true
> 
> YL620:
>   uart:
>     txd_pin: gpio.26
>     rxd_pin: gpio.16
>     rts_pin: gpio.4
>     baud: 9600
>     mode: 8N1
>   modbus_id: 1
>   tool_num: 0
>   speed_map: 0=0% 0=25% 6000=25% 24000=100%
> 
> parking:
>   enable: false
> 
> arc_tolerance_mm: 0.002000
> junction_deviation_mm: 0.010000
> verbose_errors: true
> report_inches: false
> enable_parking_override_control: false
> use_line_numbers: false
> planner_blocks: 16

I see your sdcard is nopin，have you test with sdcard？

commit 1bc76d16 (2025-04-16): Fix I2S delay - fixes #1446 (#1475)

* Fix I2S delay

I misunderstood how the I2S FIFO works and thus had miscalculated
the delay from writing the FIFO until the data appears on the output.
That resulted in sporadic TMC driver initialization failures.

* Set SDCard spi clock on every mount so $MI does not change it
closed 2025-04-16 by MitchBradley