bdring/FluidNC #1448: Problem: problem with wiki "forgot password?" page

RichMorin, 2025-02-08:
### Wiki Search Terms

forgot password

### Controller Board

6-pack

### Machine Description

no such machine

### Input Circuits

```yaml

```

### Configuration file

```yaml
no such file
```

### Startup Messages

```text
no such messages
```

### What happened?

Note: Sorry about filing this on GitHub; I couldn't find any other way to contact the project.

When I tried to register for the wiki, I was told that there was already an account for that email address. So, I clicked on the "forgot password?" link and filled in the resulting form. I was then presented with this nastygram:

139775324228400:error:1408F10B:SSL routines:ssl3_get_record:wrong version number:../deps/openssl/openssl/ssl/record/ssl3_record.c:331:

In case it helps, my name is Rich Morin and my email address is rdm@cfcl.com

bdring, 2025-02-08:
You account has been approved.

Here is the normal process to follow.

http://wiki.fluidnc.com/en/wiki_contributions

closed 2025-02-13 by bdring

RichMorin, 2025-02-13:
I just went to http://wiki.fluidnc.com/en/wiki_contributions and followed the "Discord" link (to http://wiki.fluidnc.com/en/support/discord).  There, I clicked on the "link to an invitation" and reached https://discord.com/invite/MDsRDeNsTE, where I saw "Bart Dring (bartdring) invited you to join FluidNC" and an "Accept Invite" button. Clicking that caused the banner to read "Unable to accept invite".  Clicking "Accept Invite" again had no perceptible effect. In short, it's still broken (for me).

-r

https://discord.com/invite/MDsRDeNsTE

bdring, 2025-02-13:
I'll email you the name and password.

RichMorin, 2025-02-13:
Thanks; sorry for the hassle (and thanks for your efforts on this and more generally, FluidNC).  FYI, I'm an aspiring Maslow 4.1 user (still waiting for my machine).  I've been chatting with the folks on the Maslow forum about my notions of adding a RasPi 5 and some associated devices (e.g., cameras, laser) to the sled.  After I review some FluidNC info, I'll probably pester folks for follow-up information (no good deed goes unpunished...).

-r

MitchBradley, 2025-02-13:
In my experience, anytime a Pi is involved, add Linux system administration to whatever list of tasks and problems you already have.  And I say this as someone who has been using Unix and its descendants since 1978.

RichMorin, 2025-02-13:
In 1978 I was still using Drum Scope on a CDC 3800 and 6400, RT-11 on my home LSI-11's, and VMS on a VAX 780. I touched my first Unix terminal in 1982 (and mostly use macOS these daze). That said, I agree that adding a Pi (and Linux) into the mix will definitely add tasks and problems.

However, some of the stuff I'd like to do will require a machine and OS that can support fairly arbitrary sets of code (e.g., in Elixir and/or Python). So, "needs must when the devil drives". In other news, I really hope that the Maslow and FluidNC code bases can be brought into alignment, as that would (IMHO) benefit them both.

MitchBradley, 2025-02-13:
For you own use, Pi/Raspbian/whatever is great, but supporting it for a customer base is a totally different ballgame.  For awhile I maintained the Pi version of CNCjs, and eventually gave up.  Version churn in packages and OS releases, coupled with most people's unfamiliarity with shells, init scripts and stuff like crontab, just got to be too tedious to withstand.

RichMorin, 2025-02-13:
Thanks for the feedback; it sounds all too plausible. Although I don't know whether I'll go there, the [Nerves](https://nerves-project.org/) folks have had great success in supporting large sets of fail-soft embedded processors via remote administration. Then again, that's a rather unusual "customer base"...