bdring/FluidNC #1451: Problem: controller crashes after running-code

Russellfam, 2025-02-13:
### Wiki Search Terms

So when I've finished running g-code more often than not the controller crashes the serial interface and I can move the machine in some directions but not all, the positioning does not update on the g-code sender which is open builds. I've tried several gcode files and I get the same issue

I get no errors on the console and I need to connect via webui and reboot the controller several times.  
It seems that the Webui works fine, I can home and move the machine. 

Whilst in this state the controller can receive data from PC via USB as I can get it to move back to it's zero but none of the co-ordinates update. It looks like serial comms is dying in one direction. I have tried different computers and USB cables no change. 

If I disconnect the USB and try reconnect in openbuilds without reboot and reconnect I get the following errors that can only be resolved with several reboots. as you can see the job completes with no errors, but there is no further information from the ESP32 controller.

[13:06:52] [ GCODE Parser ] GCODE File (from gcode editor) sent to backend
[13:07:06] [ JOB COMPLETE ] Job completed in 00h00m
[13:07:55] [ disconnect ] PORT INFO: Port closed
[13:07:57] [ connect ] PORT INFO: Port is now open: COM5 - Attempting to detect Controller...
[13:07:57] [ connect ] Attempting to detect Controller (1): (Autoreset)
[13:07:57] [ connect ] Attempting to detect Controller (2): (Ctrl+X)
[13:07:58] [ connect ] Attempting to detect Controller (3): (others)
[13:07:58] [ connect ] Attempting to detect Controller (4): (DTR Enable)
[13:07:59] [ connect ] ERROR!: No Response from Controller - See https://docs.openbuilds.com/doku.php?id=docs:blackbox:faq-usb-connection-failed for troubleshooting information. Closing port COM5
[13:07:59] [ disconnect ] PORT INFO: Port closed 

### Controller Board

esp32 dev

### Machine Description

Gantry CNC with standard 6600 stepper drivers 

### Input Circuits

```yaml

```

### Configuration file

```yaml
board: None
name: Garage CNC 1sqm
stepping:
  engine: RMT
  idle_ms: 255
  pulse_us: 10
  dir_delay_us: 0
  disable_delay_us: 0
  segments: 12
spi:
  miso_pin: NO_PIN
  mosi_pin: NO_PIN
  sck_pin: NO_PIN
sdcard:
  cs_pin: NO_PIN
  card_detect_pin: NO_PIN
  frequency_hz: 8000000
kinematics: {}
axes:
  shared_stepper_disable_pin: gpio.26:pd
  shared_stepper_reset_pin: NO_PIN
  homing_runs: 2
  x:
    steps_per_mm: 320
    max_rate_mm_per_min: 3000
    acceleration_mm_per_sec2: 300
    max_travel_mm: 855
    soft_limits: true
    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: gpio.33
      hard_limits: true
      pulloff_mm: 1
      standard_stepper:
        step_pin: gpio.27:pd
        direction_pin: gpio.14:pd
        disable_pin: NO_PIN
    homing:
      cycle: 2
      positive_direction: false
      allow_single_axis: true
      mpos_mm: 0.000
      seek_mm_per_min: 1400
      feed_mm_per_min: 50
  y:
    steps_per_mm: 320
    max_rate_mm_per_min: 3000
    acceleration_mm_per_sec2: 300
    max_travel_mm: 845
    soft_limits: true
    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: gpio.34
      hard_limits: false
      pulloff_mm: 1
      standard_stepper:
        step_pin: gpio.17:pd
        direction_pin: gpio.16:pd
        disable_pin: NO_PIN
    homing:
      cycle: 2
      positive_direction: false
      allow_single_axis: true
      mpos_mm: 0.000
      seek_mm_per_min: 1400
      feed_mm_per_min: 50
  z:
    steps_per_mm: 1257.142
    max_rate_mm_per_min: 1500
    acceleration_mm_per_sec2: 200
    max_travel_mm: 85
    soft_limits: true
    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: gpio.35
      hard_limits: true
      pulloff_mm: 1
      standard_stepper:
        step_pin: gpio.4:pd
        direction_pin: gpio.15:pd
        disable_pin: NO_PIN
    homing:
      cycle: 1
      positive_direction: true
      allow_single_axis: true
      mpos_mm: 0.000
      seek_mm_per_min: 600
      feed_mm_per_min: 50
control:
  safety_door_pin: NO_PIN
  reset_pin: NO_PIN
  feed_hold_pin: gpio.32:low
  cycle_start_pin: gpio.25:low:pu
  macro0_pin: NO_PIN
  macro1_pin: NO_PIN
  macro2_pin: NO_PIN
  macro3_pin: NO_PIN
  fault_pin: NO_PIN
  estop_pin: NO_PIN
coolant:
  flood_pin: NO_PIN
  mist_pin: NO_PIN
  delay_ms: 0
probe:
  pin: gpio.5:pu:low
  toolsetter_pin: NO_PIN
  check_mode_start: true
  hard_stop: false
macros: {}
start:
  must_home: true
  deactivate_parking: false
  check_limits: true
parking:
  enable: false
  axis: Z
  target_mpos_mm: -5
  rate_mm_per_min: 800
  pullout_distance_mm: 5
  pullout_rate_mm_per_min: 250
user_outputs:
  analog0_pin: NO_PIN
  analog1_pin: NO_PIN
  analog2_pin: NO_PIN
  analog3_pin: NO_PIN
  analog0_hz: 5000
  analog1_hz: 5000
  analog2_hz: 5000
  analog3_hz: 5000
  digital0_pin: NO_PIN
  digital1_pin: NO_PIN
  digital2_pin: NO_PIN
  digital3_pin: NO_PIN
  digital4_pin: NO_PIN
  digital5_pin: NO_PIN
  digital6_pin: NO_PIN
  digital7_pin: NO_PIN
user_inputs:
  analog0_pin: NO_PIN
  analog1_pin: NO_PIN
  analog2_pin: NO_PIN
  analog3_pin: NO_PIN
  digital0_pin: NO_PIN
  digital1_pin: NO_PIN
  digital2_pin: NO_PIN
  digital3_pin: NO_PIN
  digital4_pin: NO_PIN
  digital5_pin: NO_PIN
  digital6_pin: NO_PIN
  digital7_pin: NO_PIN
arc_tolerance_mm: 0.002
junction_deviation_mm: 0.01
verbose_errors: true
report_inches: false
enable_parking_override_control: false
use_line_numbers: false
planner_blocks: 16
10V:
  forward_pin: gpio.18:pd:low
  output_pin: gpio.19:pd
  pwm_hz: 5000
  speed_map: 0=0.000% 1000=0.000% 24000=100.000%
```

### Startup Messages

```text
not applicable
```

### User Interface Software

openbuilds and webui

### What happened?

as per above

bdring, 2025-02-13:
I don't use the open builds senders. I am not sure what this means.

```
[13:06:52] [ GCODE Parser ] GCODE File (from gcode editor) sent to backend
[13:07:06] [ JOB COMPLETE ] Job completed in 00h00m
[13:07:55] [ disconnect ] PORT INFO: Port closed
[13:07:57] [ connect ] PORT INFO: Port is now open: COM5 - Attempting to detect Controller...
```

It is not clear if the sender is closing the port or it is detecting a port closure. 

- Try a different sender
- Send a picture of your controller.

Russellfam, 2025-02-13:
so the close command is me disconnecting as I had done sent several other instructions like moves etc but no feedback from the controller. When I disconnect and try reconnect then I get those port issues,

MitchBradley, 2025-02-13:
Please do not assume that the startup messages are not applicable.  They contain, in addition to other information that is useful to us, the version number of the firmware you are using.

Russellfam, 2025-02-17:
> Please do not assume that the startup messages are not applicable. They contain, in addition to other information that is useful to us, the version number of the firmware you are using.

100%

$G
$SS
[MSG:INFO: FluidNC v3.9.4 https://github.com/bdring/FluidNC]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.7-dirty]
[MSG:INFO: Local filesystem type is littlefs]
[MSG:INFO: Configuration file:config.yaml]
[MSG:INFO: Machine Garage CNC 1sqm]
[MSG:INFO: Board None]
[MSG:INFO: Stepping:RMT Pulse:10us Dsbl Delay:0us Dir Delay:0us Idle Delay:255ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Shared stepper disable gpio.26:pd]
[MSG:INFO: Axis X (0.000,855.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:gpio.27:pd Dir:gpio.14:pd Disable:NO_PIN]
[MSG:INFO:  X All Limit gpio.33]
[MSG:INFO: Axis Y (0.000,845.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:gpio.17:pd Dir:gpio.16:pd Disable:NO_PIN]
[MSG:INFO:  Y All Limit gpio.34]
[MSG:INFO: Axis Z (-85.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:gpio.4:pd Dir:gpio.15:pd Disable:NO_PIN]
[MSG:INFO:  Z All Limit gpio.35]
[MSG:INFO: feed_hold_pin gpio.32:low]
[MSG:INFO: cycle_start_pin gpio.25:low:pu]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: Connecting to STA SSID:Lunar]
[MSG:INFO: Connecting.]
[MSG:INFO: Connecting..]
[MSG:INFO: Connected - IP is 192.168.0.101]
[MSG:INFO: WiFi on]
[MSG:INFO: Start mDNS with hostname:http://fluidnc.local/]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]
[MSG:INFO: 10V Spindle Ena:NO_PIN Out:gpio.19:pd Dir:NO_PIN Fwd:gpio.18:low:pd Rev:NO_PIN Freq:5000Hz Period:8191]
[MSG:INFO: Probe gpio.5:low:pu]
ok
<Alarm|MPos:0.000,0.000,0.000|FS:0,0|

Russellfam, 2025-02-18:
So some updates.I've tried a different sender so it's not a sender issue. I waiting for a new usb cable so I will hopefully be able to update that. Obviously a win would be able to use wifi but sending gcode at a high rate is not an option

MitchBradley, 2025-02-19:
You could try doing an "air cut" with the spindle turned off and no bit.  If the crash does not happen in that case, the problem is EMI from the spindle.

Russellfam, 2025-02-19:
> You could try doing an "air cut" with the spindle turned off and no bit. If the crash does not happen in that case, the problem is EMI from the spindle.

I'll try that. The concern is that once it has happened I have to do multiple reboots of the esp before I can get it up and running. I'm also going to try reflash the esp and reload my config