bdring/FluidNC #1452: Problem: Siemsn V20 VFD not working

ramack, 2025-02-14:
### Wiki Search Terms

Siemens V20

### Controller Board

6x CNC Controller

### Machine Description

3 axis CNC using stepperonline NEMA 23 closed loop motors with CL57T-V41 drivers, with 1.5kW china spindle on a Siemens V20 VFD

### Input Circuits

```yaml

```

### Configuration file

```yaml
board: 6 Pack CNC by Bart Dring
name: JaRa CNC
meta: 2025-02-14 RMk

control:
  safety_door_pin: NO_PIN
  reset_pin: NO_PIN
  feed_hold_pin: gpio.2:low:pu

i2so:
  bck_pin: gpio.22
  data_pin: gpio.21
  ws_pin: gpio.17

spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

sdcard:
  card_detect_pin: NO_PIN
  cs_pin: gpio.5

start:
  must_home: false
        
SiemensV20:
  uart:
    txd_pin: gpio.15
    rxd_pin: gpio.16
    rts_pin: gpio.14
    baud: 9600
    mode: 8E1
  modbus_id: 1
  tool_num: 0
  spinup_ms: 500
  spindown_ms: 500
  speed_map: 0=0% 1=12.5% 3000=37.5% 9000=37.5% 24000=100%
#  off_on_alarm: true

stepping:
  engine: I2S_STREAM
#we never disable the output pins automatically, so set to 255
  idle_ms: 255
# the CL57T can handle stepping frequencies up to 200 kHz, so in theory 2.5us pulses should be possible.
# other sources claim 500kHz, so we could go with 1us pulses
# 5 limits to 100kHz
  pulse_us: 6
  dir_delay_us: 6
  disable_delay_us: 0
  segments: 6

axes:
  shared_stepper_disable_pin: NO_PIN
  homing_runs: 2
  x:
# microstep mode 5: 1000 steps/round with 5mm/round -> 200 steps/mm  
    steps_per_mm: 200.0
    max_rate_mm_per_min: 1000.000
    acceleration_mm_per_sec2: 25.000
    max_travel_mm: 430.000
    soft_limits: true
    homing:
      cycle: 2
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 500.000
      settle_ms: 250
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.26:low:pu
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1.000
      standard_stepper:
        step_pin: I2SO.2
        direction_pin: I2SO.1
        disable_pin: I2SO.0

  y:
    steps_per_mm: 200.000
    max_rate_mm_per_min: 1000.000
    acceleration_mm_per_sec2: 25.000
    max_travel_mm: 850.000
    soft_limits: true
    homing:
      cycle: 2
      positive_direction: true
      mpos_mm: 560.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 500.000
      settle_ms: 250
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.33:low:pu
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1.000
      standard_stepper:
        step_pin: I2SO.5
        direction_pin: I2SO.4
        disable_pin: I2SO.7
    
  z:
    steps_per_mm: 200.000
    max_rate_mm_per_min: 500.000
    acceleration_mm_per_sec2: 25.000
    max_travel_mm: 380.000
    soft_limits: true
    homing:
      cycle: 1
      positive_direction: true
      mpos_mm: 140.000
      feed_mm_per_min: 50.000
      seek_mm_per_min: 250.000
      settle_ms: 250
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: gpio.32:low:pu
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1.000
      standard_stepper:
        step_pin: I2SO.10
        direction_pin: I2SO.9
        disable_pin: I2SO.8
```

### Startup Messages

```text
[MSG:INFO: FluidNC v3.9.3 https://github.com/bdring/FluidNC]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.7-dirty]
[MSG:INFO: Local filesystem type is littlefs]
[MSG:INFO: Configuration file:JaRa_CNC.yaml]
[MSG:INFO: Machine JaRa CNC]
[MSG:INFO: Board 6 Pack CNC by Bart Dring]
[MSG:INFO: I2SO BCK:gpio.22 WS:gpio.17 DATA:gpio.21]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
[MSG:INFO: SD Card cs_pin:gpio.5 detect:NO_PIN freq:8000000]
[MSG:INFO: Stepping:I2S_STREAM Pulse:6us Dsbl Delay:0us Dir Delay:6us Idle Delay:255ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Axis X (0.000,430.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:I2SO.2 Dir:I2SO.1 Disable:I2SO.0]
[MSG:INFO:  X Neg Limit gpio.26:low:pu]
[MSG:INFO: Axis Y (-290.000,560.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:I2SO.5 Dir:I2SO.4 Disable:I2SO.7]
[MSG:INFO:  Y Neg Limit gpio.33:low:pu]
[MSG:INFO: Axis Z (-240.000,140.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:I2SO.10 Dir:I2SO.9 Disable:I2SO.8]
[MSG:INFO:  Z Pos Limit gpio.32:low:pu]
[MSG:INFO: feed_hold_pin gpio.2:low:pu]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: Connecting to STA SSID:PalimPalim]
[MSG:INFO: Connecting.]
[MSG:INFO: Connecting..]
[MSG:INFO: Connecting...]
[MSG:INFO: Connected - IP is 192.168.0.77]
[MSG:INFO: WiFi on]
[MSG:INFO: Start mDNS with hostname:http://fluidnc.local/]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]
[MSG:INFO: UART1 Tx:gpio.15 Rx:gpio.16 RTS:gpio.14 Baud:9600]
[MSG:INFO: SiemensV20 Spindle Tx:gpio.15 Rx:gpio.16 RTS:gpio.14 Baud:9600]
```

### User Interface Software

WebUI

### What happened?

I configured the motor parameters in the VFD and can start the spindle in manual operation mode (after pressing M+OK). Also I 
set the parameters in the VFD as mentioned in #457 and activated connection macro CN11.

I then tried to start the spindle by
```M3 S10000```
but it doesn't start. Instead after a while it prints:
```
[MSG:ERR: SiemensV20 spindle did not reach device units 166. Reported value is 0]
ok
[MSG:INFO: ALARM: Spindle Control]
ALARM:10
ALARM:10
```
I am sure the RS485 is connected properly, as the scope clearly shows the transmitted data and (with slightly different voltage levels) the response from the VFD.

Any idea for the issue or how I could dig into it deeper to find a solution is warmly welcome.

### Other Information

Big THANK YOU for all the work here.
FluidNC rocks.
YOU rock!

MitchBradley, 2025-02-15:
Ask the author of #457, who submitted the V20 code. The fluidnc developers do not have one and this have no experience with it.

ramack, 2025-02-15:
Yeah sure. @rootcnc can advise?

RootCNC, 2025-02-15:
Hey Ramack, when you commanded to the turn spindle on, Did you send a speed command with it? If not. I would expect this error, as the spindle starts up to it's minimum speed. Which could be different to the minimum speed Fluidnc is expecting.

Please try and sending a spindle command with a relative high speed. For example, "M3 S10000"

ramack, 2025-02-15:
Thanks for thw tipp, I tried this, as also S10000 alone, followed by M3. Even started the spindle at th VFD in manual mode and issued the command then, but had the same:
```
[MSG:INFO: SiemensV20 Spindle Tx:gpio.15 Rx:gpio.16 RTS:gpio.14 Baud:9600]
ok
<Idle|MPos:0.000,0.000,0.000|FS:0,0|Ov:100,100,100>
$H
$H
[MSG:Homed:Z]
[MSG:Homed:XY]
ok
<Idle|MPos:0.000,560.000,140.000|FS:0,0>
M3 S10000
[MSG:ERR: SiemensV20 spindle did not reach device units 166. Reported value is 0]
ok
[MSG:INFO: ALARM: Spindle Control]
ALARM:10
ALARM:10

```

RootCNC, 2025-02-15:
I think if you're able to manually control the spindle with the BOP (builtin operators panel) then you might have some parameters wrong in the configuration of the VFD.

If you have a look in the V20spindle.cpp file. You'll see a list of vfd parameters you need to set.

ramack, 2025-02-16:
Thanks for pointing those out! I tried double checking them several times without any luck. - For now we have connected the 0-10V control such that we are at least able to commission our CNC. I'd still prefer to go for the RS485 connection, so I hope to find the time and motivation to dig deeper in it another day.

ramack, 2025-02-18:
BTW @RootCNC isn't the V20 also based on Modbus, so why don't we use the generic modbus configuration for the Siemens VFD?

ramack, 2025-02-18:
I will surely test this, thanks, but need to get some issues in the orthogonality of the axis fixed first and will probably not be able to do it this weekend.

RootCNC, 2025-02-18:
@ramack the Generic module was not a thing why at the time that module has been committed - but has been working solidly since that commit for me.

I know of a couple of users who have been using this V20 VFD type with no issues, so it is somewhat odd you're struggling to get working.

MitchBradley, 2025-02-18:
@RootCNC does that mean that you have tested the V20 spindle with the generic module, or am I misinterpreting?  Is the V20 generic config correct?

MitchBradley, 2025-02-18:
Long-term, I hope to remove as many of the hardcoded vfd spindles as possible, to save code space and complexity.

RootCNC, 2025-02-18:
Mitch, I have not tested the Generic configuration but it looks correct. I have a feeling @ramack might need to perform a reset of the drive and ensure the registers are set as per the settings listed in the file https://github.com/bdring/FluidNC/blob/main/FluidNC/src/Spindles/VFD/SiemensV20Protocol.cpp

ramack, 2025-02-18:
fairly possible, even though I double checked several times and got it easily working with the AIN. With the generic modbus driver I have some possibilities to output communication details for debugging, right?

RootCNC, 2025-02-18:
You definitely swapped to CN11?

ramack, 2025-03-08:
Finally it looks like I can dig into this again. Is the generic ModbusVFD support in 3.9.5-pre1 enough to give it a try, or would I need to compile the head?

MitchBradley, 2025-03-08:
We are doing some merging at the moment, so give us some time to complete that.

ramack, 2025-03-14:
Here we go. First I was disappointed a bit, that I didn't find any guide how to compile the code from scratch, but after seeing build-release.py installing remoteio it was easy to flash _FluidNC 3.9.5-pre1 (main-18aff2e2-dirty)_.

So I tried configuration of the generic modbus spindle driver with:

```
ModbusVFD:
  debug: 3
  uart_num: 1
  modbus_id: 1
  model: SiemensV20
  cw_cmd: 06 00 63 0c 7f > echo
  ccw_cmd: 06 00 63 04 7f > echo
  off_cmd: 06 00 63 0c 7e > echo
  set_rpm_cmd: 06 00 64 rpm  > echo
#  max_rpm_cmd: 03 10 82 00 01 > 03 02 rpm*60/100
#  min_rpm_cmd: 03 10 80 00 01 > 03 02 rpm*60/100
  get_rpm_cmd: 03 00 6e 00 01 > 03 02 rpm
  speed_map: 0=0.000% 1000=0.000% 24000=100.000%
  speed_map: 0=0% 1=12.5% 3000=37.5% 9000=37.5% 24000=100%
  off_on_alarm: true
```

The **max_rpm_cmd** caused an error and it looks, like the wiki should be changed to **get_max_rpm_cmd** (same for min).

The log shows 
```
[MSG:INFO: RS485 Tx:  01 03 40 21]
[MSG:INFO: RS485 Rx: ]
[MSG:INFO: RS485 No response]
```
while there the V20 doesn't respond at all and I even wonder what this command is about.

So I switched back to the V20 driver and activated logging there also with 
```
SiemensV20:
  debug: 3
  uart:
    txd_pin: gpio.15
    rxd_pin: gpio.16
    rts_pin: gpio.14
    baud: 9600
    mode: 8E1
  modbus_id: 1
  tool_num: 0
  spinup_ms: 500
  spindown_ms: 500
  speed_map: 0=0% 1=12.5% 3000=37.5% 9000=37.5% 24000=100%
  off_on_alarm: true
```

There I get the following:
```
[MSG:INFO: RS485 Tx:  01 03 00 6E 00 01 E5 D7]
[MSG:INFO: RS485 Rx:  01 03 02 00 00 B8 44]
[MSG:INFO: RS485 Tx:  01 03 00 6E 00 01 E5 D7]
[MSG:DBG: SiemensV20: setState:3 SpindleSpeed:10000]
[MSG:INFO: RS485 Rx:  01 03 02 00 00 B8 44]
[MSG:DBG: Syncing speed. Requested: 166 current:0]
[MSG:INFO: RS485 Tx:  01 03 00 6E 00 01 E5 D7]
[MSG:INFO: RS485 Rx:  01 03 02 00 00 B8 44]
```

So communication is working and reading the current speed (as zero) works, but adjusting the spindle speed from the UI logs state and speed changes, but doesn't transmit anything that would make the VFD ajdust anything and after some clicks a buffer overflows.

Any ideas?

ramack, 2025-03-14:
With the SiemensV20 driver from @RootCNC I get the following while running the spindle from the BOP:

```
[MSG:INFO: RS485 Tx:  01 03 00 6E 00 01 E5 D7]
[MSG:INFO: RS485 Rx:  01 03 02 14 9B F6 EF]
[MSG:INFO: RS485 Tx:  01 03 00 6E 00 01 E5 D7]
[MSG:INFO: RS485 Rx:  01 03 02 14 9B F6 EF]
[MSG:INFO: RS485 Tx:  01 03 00 6E 00 01 E5 D7]
[MSG:INFO: RS485 Rx:  01 03 02 14 9B F6 EF]
[MSG:INFO: RS485 Tx:  01 03 00 6E 00 01 E5 D7]
[MSG:INFO: RS485 Rx:  01 03 02 14 9B F6 EF]
[MSG:INFO: RS485 Tx:  01 03 00 6E 00 01 E5 D7]
[MSG:INFO: RS485 Rx:  01 03 02 14 9B F6 EF]
[MSG:INFO: RS485 Tx:  01 03 00 6E 00 01 E5 D7]
[MSG:INFO: RS485 Rx:  01 03 02 14 9B F6 EF]
```
so the speed is read correctly. Only the other commands are not transmitted...

RootCNC, 2025-03-15:
It's been a little while since I digested the programming manual for the V20. I seem to recall there is a register that tells the vfd what commands It can respond too. Its a mixture of speed and enable and direction commands. If you're using CN10 (IIRC) The BOP should not respond to button pressed. So it sounds like the vfd is miss configured.

ramack, 2025-03-15:
Thanks for your comment. The BOP only react when I put it explicitly in manual mode, normally it doesn't react, and I am in CN11. As there is no command in the Tx log that would indicate any command (expect reading the current speed) I am quite sure that it's not (only) the VFD configuration.

With the SiemensV20 spindle I at least get some communication established, with the generic ModbusVFD FluidNC only emmits too short frames (which are not in the configuration)...

MitchBradley, 2025-03-15:
Feel free to fix any errors that you find.  I am busy working on the IO expander and don't have time right now to deal with a VFD that I don't have access to.

ramack, 2025-03-16:
OMG I messed up the parity setting when configuring the generic VFD driver. We clearly need:
`mode: 8E1`
for the Siemens V20.

MitchBradley, 2025-03-16:
If you type "compile" in the wiki search box, this page comes up http://wiki.fluidnc.com/en/installation#compiling-reference-only-you-do-not-need-to-compile.

ramack, 2025-05-02:
I think we can close this. In the meantime I upgraded to [v3.9.6](https://github.com/bdring/FluidNC/releases/tag/v3.9.6) and with the SiemensV20 driver it works. (Did not yet test with the generic driver.)

I once had an "overtemperature" warning in the V20 but the temperature calculation was not working properly due to the motor parameters, so it would be quite cool to monitor the parameters which could cause trouble and display them somewhere in FluidNC and trigger a warning before the VFD stops the spindle or even try cooling it.

But this is not essential for sure, it just happened in the early approaches and it would have been cool to see it before in FluidNC.

Also it would be cool to see live data of the power (+voltage and current?) of the spindle in a "spindle" view. Anyhow, for now I don't even write a feature request ticket for it, maybe I will try to implement something to read more "interesting" status registers.

closed 2025-05-02 by ramack

ramack, 2025-05-02:
fair enough. I didn't expect you to work on it ;-) I mainly wrote down the random thoughts I had, so far I didn't even see a place where such details could be shown (expect maybe the console but it might not be the right place). In the long term it could be good to switch to the generic VFD driver but then such additional data readings are getting even more challenging.

So just in case you can propose some way to show additional data from the VFD (in the WebUI and FluidDial) feel free to give me a pointer.