bdring/FluidNC #1453: Problem: gpio.14 (integrated esp32) does not change state. Plasma branch f/w.

jusbo20, 2025-02-16:
### Wiki Search Terms

Created Discord post with same-ish heading. Was told to submit github issue by Mr. Dring
"Plasma" was then used as a search term on the wiki to recall specific plasma page.

### Controller Board

6x CNC Controller board v1.2 integrated ESP32, external stepper drivers with base config of xyyz yaml official.
Wiring is shown on picture attached.
[image]
[image]

### Machine Description

Plasma firmware branch on 6x CNC Controller Board v1.2. Using DMM to test output of gpio.14.
The firmware was a drop? Artifact? I don't recall the verbiage used....I did not compile it myself. The file was for windows.

### Input Circuits

```yaml
N/A
```

### Configuration file

```yaml
board: 6 Pack
name: 6 Pack External XYYZ
meta: 2023-01-29 B. Dring
stepping:
  engine: I2S_STREAM
  idle_ms: 250
  pulse_us: 4
  dir_delay_us: 1
  disable_delay_us: 0
axes:
  shared_stepper_disable_pin: NO_PIN
  x:
    steps_per_mm: 199.64
    max_rate_mm_per_min: 5000
    acceleration_mm_per_sec2: 100
    max_travel_mm: 300
    soft_limits: false
    homing:
      cycle: 2
      positive_direction: true
      mpos_mm: 0
      feed_mm_per_min: 100
      seek_mm_per_min: 200
      settle_ms: 500
      seek_scaler: 1.1
      feed_scaler: 1.1
    motor0:
      limit_neg_pin: gpio.2:pu:low
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1
      standard_stepper:
        step_pin: i2so.10
        direction_pin: i2so.9
        disable_pin: NO_PIN
  y:
    steps_per_mm: 199.64
    max_rate_mm_per_min: 5000
    acceleration_mm_per_sec2: 100
    max_travel_mm: 300
    soft_limits: false
    homing:
      cycle: 3
      positive_direction: false
      mpos_mm: 0
      feed_mm_per_min: 100
      seek_mm_per_min: 200
      settle_ms: 500
      seek_scaler: 1.1
      feed_scaler: 1.1
    motor0:
      limit_neg_pin: gpio.26:pu:low
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1
      standard_stepper:
        step_pin: i2so.2
        direction_pin: i2so.1
        disable_pin: NO_PIN
    motor1:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1
      standard_stepper:
        step_pin: i2so.5
        direction_pin: i2so.4
        disable_pin: NO_PIN
  z:
    steps_per_mm: 197.78
    max_rate_mm_per_min: 5000
    acceleration_mm_per_sec2: 100
    max_travel_mm: 150
    soft_limits: false
    homing:
      cycle: 1
      positive_direction: true
      mpos_mm: 150
      feed_mm_per_min: 100
      seek_mm_per_min: 800
      settle_ms: 500
      seek_scaler: 1.1
      feed_scaler: 1.1
    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1
      standard_stepper:
        step_pin: i2so.13
        direction_pin: i2so.12:low
        disable_pin: NO_PIN
probe:
  pin: gpio.33:pu:low
  check_mode_start: true
  toolsetter_pin: NO_PIN
PlasmaSpindle:
  enable_pin: gpio.14
  arc_ok_pin: gpio.36:low
  arc_wait_ms: 1200
  tool_num: 0
  off_on_alarm: true
i2so:
  bck_pin: gpio.22
  data_pin: gpio.21
  ws_pin: gpio.17
spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18
sdcard:
  card_detect_pin: NO_PIN
  cs_pin: gpio.5
start:
  must_home: false
```

### Startup Messages

```text
Grbl 3.0 [FluidNC v3.0.x (HEAD-ef4b1c4) (wifi) '$' for help]
$Startup/Show
[MSG:INFO: FluidNC v3.0.x (HEAD-ef4b1c4) https://github.com/bdring/FluidNC]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.7-dirty]
[MSG:INFO: Local filesystem type is littlefs]
[MSG:INFO: Configuration file:config.yaml]
[MSG:INFO: Machine 6 Pack External XYYZ]
[MSG:INFO: Board 6 Pack]
[MSG:INFO: I2SO BCK:gpio.22 WS:gpio.17 DATA:gpio.21Min Pulse:2us]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
[MSG:INFO: SD Card cs_pin:gpio.5 detect:NO_PIN freq:8000000]
[MSG:INFO: Stepping:I2S_STREAM Pulse:4us Dsbl Delay:0us Dir Delay:1us Idle Delay:250ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Axis X (-300.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:I2SO.10 Dir:I2SO.9 Disable:NO_PIN]
[MSG:INFO:  X Neg Limit gpio.2:low:pu]
[MSG:INFO: Axis Y (0.000,300.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:I2SO.2 Dir:I2SO.1 Disable:NO_PIN]
[MSG:INFO:  Y Neg Limit gpio.26:low:pu]
[MSG:INFO:   Motor1]
[MSG:INFO:     standard_stepper Step:I2SO.5 Dir:I2SO.4 Disable:NO_PIN]
[MSG:INFO: Axis Z (0.000,150.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:I2SO.13 Dir:I2SO.12:low Disable:NO_PIN]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: Connecting to STA SSID:Unifi]
[MSG:INFO: Connecting.]
[MSG:INFO: Connecting..]
[MSG:INFO: Connected - IP is 192.168.99.88]
[MSG:INFO: WiFi on]
[MSG:INFO: Start mDNS with hostname:http://6pack.local/]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]
[MSG:INFO: ArcOK gpio.36:low]
[MSG:INFO: PlasmaSpindle Ena:gpio.14 Arc OK:gpio.36:low]
[MSG:INFO: Probe gpio.33:low:pu]
ok
```

### User Interface Software

WebUi via usb and or wifi

### What happened?

Before hooking up to my plasma machine I am trying to verify that gpio.14 is either

1. Putting out voltage
2. Dry open/closure 

No voltage readings between gpio.14&GND. Continuity is closed no matter what command (M3, M3s1, M3S1000, M5) is inputted into the command box of the webUI.

### GCode File

No GCode file available issued single commands 1 at a time. They are M3, M3s1, M3S1000, M5

### Other Information

I have provided pictures of my wiring and my enclosure of my actual wiring accurately depicted on the piece of paper.

bdring, 2025-02-16:
Please show the console responses, when you send M3.

bdring, 2025-02-16:
I have a setup that works. If you send M3 without getting an arc ok signal you get this, with a 1200 ms delay after M3.

```
Grbl 3.9 [FluidNC v3.9.4 (PlasmSpindle-ae36fe45) (wifi) '$' for help]
M3
[MSG:ERR: PlasmaSpindle failed to get arc OK signal]
ok
[MSG:INFO: ALARM: Spindle Control]
ALARM:10
```          

Here I activate the arc_ok signal right after the M3.

```
M3
ok                                                                                                                
``` 

Here I activate the arc_ok signal right after the M3 and then release the arc_ok

```
M3
ok
[MSG:INFO: ALARM: Abort Cycle]
ALARM:3
```

In all cases the LED for gpio.14 lights correctly.

Please send your console data.

jusbo20, 2025-02-16:
Mr. Dring,
I noticed on your startup messages you have "PlasmaSpindle" as your
firmware....mine says
FluidNC v3.0.x (HEAD-ef4b1c4) https://github.com/bdring/FluidNC]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.7-dirty]
When I gave the M3 commands, it always says comes back OK.I may need to
flash this board again to get it in line with yours before we continue
troubleshooting.....which I appreciate, sincerely.

bdring, 2025-02-16:
- Did you compile from the PlasmSpindle branch?
- Did you use Git to pull the code?

jusbo20, 2025-02-16:
I used the artifact, or drop.....(I forget the exact verbiage)I did *not*
compile it. I downloaded it from github web interface.

jusbo20, 2025-02-16:
But yes I did change to the plasma branch.....sorry forgot to add that to
the last correspondence

bdring, 2025-02-16:
- How are you triggering the arc_ok signal during these tests?  It might be trigger all the time. 
- Is the LED for io.14 going on?

Depending how you compile, the rev info string can be different. I have attached my bin file. I think you can load that via the WebUI.

[firmware.zip](https://github.com/user-attachments/files/18816648/firmware.zip)

jusbo20, 2025-02-17:
I reflashed using the WebUI and your bin file you sent. It completed
just fine. For what it's worth the last time didn't go smooth...it
hung when restarting the controller. Here is the output when I used
the Web interface of the controller and issued "M3"....I copied a
couple lines before just for context.
As probably expected the pin 14 LED did not light.

[MSG:INFO: Telnet started on port 23]
[MSG:INFO: ArcOK gpio.36:low]
[MSG:INFO: PlasmaSpindle Ena:gpio.14 Arc OK:gpio.36:low]
[MSG:INFO: Probe gpio.33:low:pu]
ok
<Idle|MPos:0.000,0.000,0.000|FS:0,0>
M3
[MSG:ERR: PlasmaSpindle failed to get arc OK signal]
ok
[MSG:INFO: ALARM: Spindle Control]
ALARM:10
ALARM:10

jusbo20, 2025-02-17:
I have attached the page that is in my manual (Everlast 62i plasma) that
gives a description of the pinout. You may or may not find it helpful, but
I thought I would share.

bdring, 2025-02-17:
I have a comment in the code that we probably want to check the state of the arc_ok before turning on the plasma. I think I will add code for that.

If you try to turn on the plasma and you already have an arc_ok, it means something is wrong with your system. It should send an alarm and not enable the plasma.

[image]

jusbo20, 2025-02-17:
Not necessarily. The ARC OK is.....(explained in the perspective of if the
plasma was a human) Ok, I'm on i'm ready to do my job. (M3 issued, turnon)
Ok, I'm on, I've fired my torch,my arc is good....lets send the ARC OK
signal (closes contacts) to let the motors move me.

The way I see this is arc ok needs to come after m3 (fire) but BEFORE
moving. Is that what you mean?

bdring, 2025-02-17:
Yes, but the new code would do an error check before sending the enable signal.

If the arc_ok is on ***before enable***, it would assume there is a error in the system and refuse to enable.

jusbo20, 2025-02-21:
I am just now getting to play with my table again. I am running the code
for a straight line using sheet cam to generate and OpenBuilds Control to
send.
G20
G0X3.0078Y4.3731
G0Z0.1181
M3
G04 P0.8
F15.0
G1X6.9448Y8.3101
M5
G4 P0.8
M5 M30

I get an alarm.....Here is the log:
[17:04:30] [ GCODE Parser ] GCODE File (from gcode editor) sent to backend

[17:04:31] [ G20 ] ok

[17:04:31] [ $G ] [GC:G0 G54 G17 G20 G90 G94 M5 M9 T0 F0 S0]

[17:04:31] [ $G ] ok

[17:04:31] [ G0X3.0078Y4.3731 ] ok

[17:04:31] [ G0Z0.1181 ] ok

[17:04:35] [ M3 ] [MSG:ERR: PlasmaSpindle failed to get arc OK signal]

[17:04:35] [ M3 ] ok

[17:04:35] [ $G ] [MSG:INFO: ALARM: Spindle Control]

[17:04:35] [ ALARM ] ALARM: 10 - EStop asserted. Clear and reset [ $G ]

[17:04:35] [ ] ALARM: 10 - EStop asserted. Clear and reset [ $G ]

[17:04:35] [ $G ] ALARM:10

[17:04:35] [ $G ] [GC:G0 G54 G17 G20 G90 G94 M3 M9 T0 F0 S0]

[17:04:35] [ $G ] ok

[17:04:35] [ ] error: 9 - G-code commands are locked out during alarm or
jog state. [ G04 P0.8 ]<hr>This error may just be a symptom of an earlier
event:<br> ALARM: 10 - EStop asserted. Clear and reset

[17:04:35] [ ERROR ] error: 9 - G-code commands are locked out during alarm
or jog state. [ G04 P0.8 ]

This error may just be a symptom of an earlier event:
ALARM: 10 - EStop asserted. Clear and reset
[17:04:35] [ G04 P0.8 ] error:9

[17:04:35] [ ] [MSG:ERR: GCode cannot be executed in lock or alarm state]

[17:04:44] [ [clear alarm] ] Operator clicked Clear Alarm: Cleared Lockout
and Emptied Queue

[17:04:44] [ ] Grbl 3.9 [FluidNC v3.9.4 (PlasmSpindle-ae36fe45) (wifi) '$'
for help]

[17:04:44] [ ] [MSG:INFO: Caution: Unlocked]

[17:04:44] [ JOB COMPLETE ] Job completed in 00h00m

[17:04:44] [ ] ok

[17:04:46] [ [clear alarm] ] Operator clicked Clear Alarm: Cleared Lockout
and Emptied Queue

[17:04:48] [ ] Grbl 3.9 [FluidNC v3.9.4 (PlasmSpindle-ae36fe45) (wifi) '$'
for help]

[17:04:48] [ ] ok

bdring, 2025-02-22:
What is the arc ok signal connected to?

Malibrans, 2025-06-10:
Hey guys, I was reading your post and wanted to take the liberty of commenting on my own case, which is a bit similar. I’m running a 6x CNC Controller for the first time on a test bench and I’m having a problem with the plasma start relay. This may sound strange, but when I issue the M3 command, the relay doesn’t activate, and when arc_wait_ms runs out, the alarm goes off. But I discovered that if I issue M3 and then simulate the ARC ok signal, the relay activates immediately, which confuses me. Because to obtain the ARC ok signal, the plasma start relay needs to be activated first, or am I wrong? On the other hand, if I first simulate the ARC ok signal and then issue the M3 command, the machine understands that something is wrong and generates an alarm. Thanks in advance !

jusbo20, 2025-06-11:
Yes I had the same issue. You have to jump the input (whatever you made the Arc-OK input to be)to make the plasma fire. As it stands the last time I fooled with it, the program wants to put "the cart before the horse" but then the horse runs into the back of the cart and it catches fire. I couldn't keep messing with it so I got a different controller. Take care!