bdring/FluidNC #1456: Feature: Dynamically Adjusting Acceleration in CNC Motion (G0 vs. G1/G2/G3)

mikeoverbay, 2025-02-23:
### Machine Context

Any machine.

### Feature Description

I’ve been working on dynamically adjusting motor acceleration based on whether a CNC machine is executing rapid (G0) or linear (G1, G2, G3) moves. My goal is to optimize motion by changing acceleration time (t) in the equation:
a = (v - u) / t
where:
u = initial velocity
v = target velocity
t = time to reach target velocity
Why This Matters
Line moves (G1, G2, G3) can handle much faster acceleration because the feed rate is typically 1/2 to 1/10 of rapid speeds—sometimes greatly less when cutting alloys. Since the mass is already in motion, the force required for acceleration is minimal.
I've done the the math using chatGPT 03-mini-high. The theory looks solid to me based on the charts.

[image]

/*
    Copyright 2013-2020 Will Winder

    This file is part of Universal Gcode Sender (UGS).

    UGS is free software: you can redistribute it and/or modify
    it under the terms of the GNU General Public License as published by
    the Free Software Foundation, either version 3 of the License, or
    (at your option) any later version.

    UGS is distributed in the hope that it will be useful,
    but WITHOUT ANY WARRANTY; without even the implied warranty of
    MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
    GNU General Public License for more details.

    You should have received a copy of the GNU General Public License
    along with UGS.  If not, see <http://www.gnu.org/licenses/>.
*/
package com.willwinder.universalgcodesender.gcode;

import com.willwinder.universalgcodesender.gcode.processors.CommandProcessor;
import com.willwinder.universalgcodesender.gcode.processors.CommandProcessorList;
import com.willwinder.universalgcodesender.gcode.processors.Stats;
import com.willwinder.universalgcodesender.gcode.util.Code;
import com.willwinder.universalgcodesender.gcode.util.GcodeParserException;
import com.willwinder.universalgcodesender.gcode.util.GcodeParserUtils;
import com.willwinder.universalgcodesender.types.PointSegment;

import java.util.ArrayList;
import java.util.Collection;
import java.util.List;
import java.util.regex.Matcher;
import java.util.regex.Pattern;

Some code changes to GcodeParser.java.
It is untested and created with chatGPT 03-mini-high

/**
 * Object to parse G-code one command at a time in a way that can be used by any
 * other class which needs to know about the current state at a given command.
 *
 * This object can be extended by adding any number of ICommandProcessor objects
 * that are applied to each command in the order they were inserted.
 *
 * <p>Dynamic Acceleration Control (FluidNC):
 * When a G0 or G1 move is detected, this modified version injects three commands.
 * The values sent in $120, $121, and $122 depend on whether the move is rapid or feed:
 * <br>
 * &nbsp;&nbsp;&nbsp;• For G0 moves, the fixed (stored) values are used.
 * <br>
 * &nbsp;&nbsp;&nbsp;• For G1 moves, the stored values are multiplied by a dynamic tFactor.
 * <br>
[... 309 lines cut ...]
            }
            
            cmd = "$122=" + originalAccel;
            meta = GcodeParserUtils.processCommand(cmd, line, state, true);
            if (meta != null) {
                for (GcodeMeta m : meta) {
                    if (m.state != null) {
                        state = m.state;
                        statsProcessor.processCommand(cmd, state);
                    }
                }
            }
            
            originalSettingsStored = false;
        }
    }
}

### Other Approaches

none

### How I Can Help

I can't

breiler, 2025-02-23:
Your request in other words: "Please split up the settings for acceleration on rapid moves (G0) and cutting moves (G1, G2, G3)". 

That ChatGPT nonsense does not help anyone.

mikeoverbay, 2025-02-23:
> Your request in other words: "Please split up the settings for acceleration on rapid moves (G0) and cutting moves (G1, G2, G3)".
> 
> That ChatGPT nonsense does not help anyone.

Did you bother to look at the charts or just deicide to bully me?
It's actually pretty fascinating. 
If you know your mass, you can calculate optimal accel for give speeds based on current velocity. 
I have been studying this for a couple of days to explore the variables and if the impact is worth the effort.
My issue is I don't know java for shit. Give me C im good to go. I was hoping for input. Not snide remarks.

MitchBradley, 2025-02-23:
Stepper motors are not constant-force devices.  Their torque decreases with speed.

mikeoverbay, 2025-02-23:
Yes. I am aware of this. I looked at curves of nema 34 and nema 23 motors.
I don't think it will effect changing the accel setting. It would be useful to looking the variance of motor bands and sizes.
If we could find a good average, we could plug that in to the equations to stop the risk of skipping?
Introduce Q as torque. O3 is amazing at doing hard math.
I'll think about this.
When I get my controller, I'll hook it up to my gantry and see if I can get fluidNC to build.

MitchBradley, 2025-02-23:
Realistically, we FluidNC developers are probably not going to do anything about this.  People have enough trouble tuning with fixed acceleration.  We are unlikely to add the extra configuration complexity, incurring the support costs of documenting it and helping people who don't understand it, in exchange for the modest performance improvements that might result.  The motion planning is ridiculously complicated already; making it more complicated is not on our radar unless the gains would be compelling.  You might think they are; my best guess is that they are not., across the range of machines and use cases that we support.

Feel free to continue you investigation, with the understanding that, in order for us to deploy your changes, the bar for proving the value is very high.  Furthermore, since you are focused on changing UGS, which represents a fraction of the many UIs that are used with FluidNC, any changes to UGS will apply to a minority of FluidNC users.  With that in mind, perhaps it would be better to move this discussion to a different forum, perhaps one specific to UGS.

As a further consideration, your idea of sending $ commands for reconfiguration in the middle of GCode sequences is problematic.  It is not a common situation and we do not support it.  It might happen to work, but we cannot guarantee that it will always do what you expect.  Some configuration things cannot be changed on-the-fly, and when they do work, they can have timing implications that interact badly with the execution of GCode.  Configuration is intended to be done infrequently, typically once at startup.

breiler, 2025-02-24:
As one of the developers on UGS, I will probably not spend time on this experiment. But the approach that you have come up with (changing the settings while streaming the gcode) can easily be done with a script or a search and replace it in the gcode file.

mikeoverbay, 2025-02-24:
Thank you!
This is what I wanted to hear. 
I will do some experimenting on my own. 
If I see that it does what I am hoping, I will get back to folks.
Thank for all the hard work!
It's a fantastic bit of coding!

closed 2025-02-24 by mikeoverbay

mikeoverbay, 2025-02-24:
> As one of the developers on UGS, I will probably not spend time on this experiment. But the approach that you have come up with (changing the settings while streaming the gcode) can easily be done with a script or a search and replace it in the gcode file.

Thank you for the work. If I do see some usefulness for this idea, I will hack a script plugin together, if it can be done that way. 
Ideally, it would be better just for the motion planner to handle it. I think It's worth studying so I'll continue. 
Thank you for the hard work!

bdring, 2025-02-24:
A higher acceleration increases jerk. I would think you want a lower jerk when cutting. Jerk causes stress in machines than can affect cut quality.

Larger machines use s-curve acceleration with a jerk parameter. Some day we may implement that, but it would require a lot of work, so it is not a priority at this time.

BTW: Those curves don't make any sense to me. NewtonChat said: F=MA. There is no t.

mikeoverbay, 2025-02-25:
T is the time constant to get to speed. 
All I know is that uping accel, it runs smoother during YZ contouring but rapid, it wants to move and stop to fast from to low of speed or a dead stop. Once its in motion , it can handle much faster accel as long as the deferential is too great.
This may not be a issue with the much faster planning the ESP32 affords. I haven't test on that chip.