bdring/FluidNC #1465: Feature: Auto resume for pen plotter after power loss

Himanshu2213, 2025-03-18:
### Machine Context

I am using config file for eggbot.h for my pen plotter.

### Feature Description

Can we have an Auto resume feature after power loss like one in the 3d printers. I made a pen plotter and sometimes it takes 15-20 hours to complete a sketch and if unfortunately there is a power loss whole sketch made is wasted and it starts from the begining after that, i am searching for a solution to this problem as it helps a lot and saves a lot of time . Please suggest some solution to this problem. Thank you.

### Other Approaches

I tried providing continuous power supply but it won't work in some rare cases.

### How I Can Help

we can derive this feature from 3d printers.

bdring, 2025-03-18:
We don't have a good way of doing that at this time. We had a long discussion on Discord and decided it is probably best to use a UPS. Even if the UPS is short lived you might be able to safely save the status and shut down. 

https://discord.com/channels/780079161460916227/1341581512457584641

If you know exactly how 3D printers do it, please save us the research time.

MitchBradley, 2025-03-18:
"It won't work in some rare cases" - that is true of pretty much every solution.

Any solution will involve the addition of hardware to provide advance warning of impending power loss while maintaining power long enough for controlled shutdown.  The development of such hardware will require time, not counting the software work to make effective use of it.  Even if there are existing products that could be used, the selection, procurement, testing, integration, documentation, and support of them will take time, especially since many people will want to use a different one that they already purchased from AliExpress or whatever.

My best estimate is that, considering everything that would have to be done this feature would require a minimum of 150 hours of developer time before it is stable and working well enough to be relied on, not counting the hardware cost or the ongoing support cost.  Developer time is worth at least $70/hour and arguably more, so the market price of this development would be a minimum of $10,000.  Are you prepared to fund that, or to lead a coalition of people to collectively fund it?

mikeoverbay, 2025-04-12:
> We don't have a good way of doing that at this time. We had a long discussion on Discord and decided it is probably best to use a UPS. Even if the UPS is short lived you might be able to safely save [...]

I know how the do it..

The store the current XYZ on every single move. This is why having power lost restore can affected the quality of prints. pausing to write to EEPROM.
You simply need to store these positions, have a flag for power lost (reset when starts) and to enable it and sender software that can check power loss. If it has, get the XYZ and find that starting location.
This is a fairly simple thing to do but you need warzs to use it.
I wrote code to break up huge CNC code. My machine only had 48k of mem. Big old machining center.
You have to parse all the importing information like active work coord (G54..G55..) Feeds and speeds and find a good clearance for the tool. Pre send this info before staring where it left off.

MitchBradley, 2025-04-12:
ESP32 systems typically do not have EEPROM.  Non-volatile storage is emulated using a partition of the FLASH that is used for the program code and the localfs file system.  Writing to that FLASH is *very* slow compared to other types of accesses, and cannot coexist with real-time motion execution which requires interrupts.  Storing positions to non-volatile storage after every move is just not feasible on an ESP32 without extra hardware.  If you were to add a real EEPROM or a battery-backed NVRAM, it would be doable, but that would require either a special controller board or mods to existing boards that would be different for every board.

closed 2025-09-30 by MitchBradley