bdring/FluidNC #1466: Problem: please provide procedure for compiling FluidNC source code.

Bharathk-Gowda, 2025-03-26:
### Wiki Search Terms

@ramack @MateoAero @craiglink @bdring @brianredbeard @nils @ryba-xek 
I need to add backlash compensation feature in this source code,
but I don't know what IDE to use for compiling and debug, please help me with this starting issue, i see wiki compilation part, no much information is provided, please help me.
Once I get to know this procedure I will proceed with backlash compensation feature addition. 

### Controller Board

6-pack

### Machine Description

XYYZB configuration machine.

### Input Circuits

```yaml

```

### Configuration file

```yaml
XYYZB configuration machine.
```

### Startup Messages

```text
emty
```

### What happened?

compilation problem.

bdring, 2025-03-26:
Search our wiki for [compiling](http://wiki.fluidnc.com/en/installation#compiling-reference-only-you-do-not-need-to-compile).

MitchBradley, 2025-03-26:
You should probably also search Discord, the issues list, and the pull requests for "backlash" to understand the extensive existing discussion around this feature.

bdring, 2025-03-26:
http://wiki.fluidnc.com/en/development/open_issues#backlash-compensation