bdring/FluidNC #1469: Problem:  Unexpected Cross-Axis Movement During Homing on CoreXY Frame

bibidentuhanoi, 2025-03-31:
### Wiki Search Terms

Corexy

### Controller Board

MKS DLC32

### Machine Description

Corexy 

[image]

### Input Circuits

```yaml

```

### Configuration file

```yaml
board: MKS-DLC32 V2.1
name: Plottergeist
meta: (21.11.2022) by thedevleon

kinematics:
  corexy:

stepping:
  engine: I2S_STATIC
  idle_ms: 0
  pulse_us: 4
  dir_delay_us: 1
  disable_delay_us: 0

start:
  must_home: false
  deactivate_parking: true
  check_limits: false
  

axes:
  shared_stepper_disable_pin: I2SO.0
  
  x:
    steps_per_mm: 80.000
    max_rate_mm_per_min: 6000.000
    acceleration_mm_per_sec2: 5000
    max_travel_mm: 230.000
    soft_limits: true
    homing:
      cycle: 2
      positive_direction: false
      mpos_mm: 0
      feed_mm_per_min: 500
      seek_mm_per_min: 500
      settle_ms: 2
      seek_scaler: 1.100
      feed_scaler: 1.100
    motor0:
      limit_all_pin: gpio.36:low
      hard_limits: true
      pulloff_mm: 3
      stepstick:
        step_pin: I2SO.1
        direction_pin: I2SO.2:low

  y:
    steps_per_mm: 80.000 
    max_rate_mm_per_min: 6000.000
    acceleration_mm_per_sec2: 5000
    max_travel_mm: 240.000
    soft_limits: true
    homing:
      cycle: 2
      positive_direction: false
      mpos_mm: 0
      feed_mm_per_min: 500.000
      seek_mm_per_min: 500.000
      settle_ms: 2
      seek_scaler: 1.1
      feed_scaler: 1.1
    motor0:
      limit_all_pin: gpio.35:low
      hard_limits: true
      pulloff_mm: 3
      stepstick:
        step_pin: I2SO.5
        direction_pin: I2SO.6:low

  z:
    steps_per_mm: 10
    max_rate_mm_per_min: 2000
    acceleration_mm_per_sec2: 2000
    max_travel_mm: 10
    homing:
      cycle: 1
      mpos_mm: 0
      positive_direction: true
    motor0:
      rc_servo:
        pwm_hz: 50
        output_pin: gpio.22
        min_pulse_us: 1000
        max_pulse_us: 2000

i2so:
  bck_pin: gpio.16
  data_pin: gpio.21
  ws_pin: gpio.17

spi:
  miso_pin: gpio.12
  mosi_pin: gpio.13
  sck_pin: gpio.14
```

### Startup Messages

```text
[MSG:INFO: FluidNC v3.9.5 https://github.com/bdring/FluidNC]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.7-dirty]
[MSG:INFO: Local filesystem type is spiffs]
[MSG:INFO: Configuration file:config.yaml]
[MSG:INFO: Machine Plottergeist]
[MSG:INFO: Board MKS-DLC32 V2.1]
[MSG:INFO: I2SO BCK:gpio.16 WS:gpio.17 DATA:gpio.21Min Pulse:2us]
[MSG:INFO: SPI SCK:gpio.14 MOSI:gpio.13 MISO:gpio.12]
[MSG:INFO: Stepping:I2S_STATIC Pulse:4us Dsbl Delay:0us Dir Delay:1us Idle Delay:0ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Shared stepper disable I2SO.0]
[MSG:INFO: Axis X (0.000,230.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.1 Dir:I2SO.2:low Disable:NO_PIN]
[MSG:INFO:  X All Limit gpio.36:low]
[MSG:INFO: Axis Y (0.000,240.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.5 Dir:I2SO.6:low Disable:NO_PIN]
[MSG:INFO:  Y All Limit gpio.35:low]
[MSG:INFO: Axis Z (-10.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     rc_servo Pin:gpio.22 Pulse Len(1000,2000 period:1048575)]
[MSG:INFO:     Update timer for rc_servo at 20 ms]
[MSG:INFO: Kinematic system: CoreXY]
[MSG:INFO: Connecting to STA SSID:*****]
[MSG:INFO: Connecting.]
[MSG:INFO: Connecting..]
[MSG:INFO: Connected - IP is *****]
[MSG:INFO: WiFi on]
[MSG:INFO: Start mDNS with hostname:http://fluidnc.local/]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]
ok
```

### User Interface Software

Webui

### What happened?

I’m using a CoreXY setup with limit switches. During homing, when the Y axis is homed, the X axis moves unexpectedly (and vice versa). This only happens during the homing pull-off; normal movements and limit switches work fine.I expect the only the homing axis should move, with no cross-axis movement.

### Other Information

$HX
<Home|MPos:9.538,-280.725,0.000|FS:707,0|WCO:0.000,0.000,0.000>
<Home|MPos:6.837,-280.725,0.000|FS:707,0|Ov:100,100,100>
<Home|MPos:4.125,-280.725,0.000|FS:707,0>
<Home|MPos:1.425,-280.725,0.000|FS:707,0>
<Home|MPos:0.050,-280.725,0.000|FS:707,0|Pn:X>
<Home|MPos:1.413,-280.725,0.000|FS:707,0>
<Home|MPos:2.038,-280.725,0.000|FS:707,0>
<Home|MPos:0.675,-280.725,0.000|FS:707,0|Pn:X>
<Home|MPos:2.500,-280.725,0.000|FS:707,0>
[MSG:Homed:X]
ok
<Idle|MPos:0.000,-280.725,0.000|FS:0,0>
<Idle|MPos:0.000,-280.725,0.000|FS:0,0|WCO:0.000,0.000,0.000>

bdring, 2025-03-31:
Try changing the homing cycles to

- Z: `cycle: 1`
- X  `cycle: 2`
- Y  `cycle: 3`

...basically change the cycle for Y in your config to 3.

bibidentuhanoi, 2025-03-31:
Thanks for your reply, but the problem is still there—the toolholder slowly drifts toward the X axis limit switch while homing Y.

$HY
[MSG:Homed:Y]
ok
<Idle|MPos:0.000,0.000,0.000|FS:0,0|Pn:X>
$HY
[MSG:Homed:Y]
ok
<Idle|MPos:0.000,0.000,0.000|FS:0,0|Pn:X>
$HX
[MSG:ERR: Ambiguous limit switch touching. Manually clear all switches]
[MSG:INFO: ALARM: Ambiguous Switch]
ALARM:12
ALARM:12
ok
<Alarm|MPos:0.000,0.000,0.000|FS:0,0|Pn:X>

bdring, 2025-03-31:
Are you sure the microstepping is the same on both motors?

bibidentuhanoi, 2025-03-31:
I forgot to mention that this issue only occurs when I set the kinematics to CoreXY. In midTbot mode, the problem of drifting toward the limit switch disappears. However, with midTbot (using the same steps-per-mm), the Y axis doesn’t move far enough to reach the intended goal distance.

bdring, 2025-03-31:
That is another clue that suggests a microstepping issue

Try removing the kinematics temporarily and jogging short equal amounts for X and Y. Watch the rotation amount for each motor. Ignore the actual motion of the machine.

bibidentuhanoi, 2025-03-31:
Based on my measurements and the OLED display on the controller board, each 10mm move results in approximately 90 degrees of motor rotation.

bdring, 2025-03-31:
without kinematics, do some long jogs at 45 degree angles. that should be straight lines on a corexy machine.

bibidentuhanoi, 2025-03-31:
Without kinematics, it should be in Cartesian mode, right? In that case, with a long jog (100mm), both axes move when I command a single-axis move.

bibidentuhanoi, 2025-03-31:
oh sorry I commanded G0 X60 Y60, and yes, it moves in a straight line along the X axis.

bdring, 2025-03-31:
I don't have any more things to try. There are a lot of corexy users and I don't hear any issues.

A lot of people have given up on those MKS motors. Primarily for reliability issues. They lock up sometimes.

MitchBradley, 2025-04-02:
You might try adding :low to the step pins in the config file.

Vorobeyyyyyy, 2025-04-10:
Iam facing same issue for now.  Right after touching X endstop i moves X back (as expected) and also Y forward (around 0.5mm)

my configuration:
`board: None
name: Default (Test Drive no I/O)
stepping:
  engine: RMT (using GPIO)
  idle_ms: 255
  pulse_us: 4
  dir_delay_us: 0
  disable_delay_us: 0
  segments: 12
uart1:
  txd_pin: gpio.22
  rxd_pin: gpio.23
  baud: 115200
  mode: 8N1
kinematics:
  corexy:
axes:
  shared_stepper_disable_pin: gpio.21
  shared_stepper_reset_pin: NO_PIN
  homing_runs: 2
  x:
    max_rate_mm_per_min: 12000
    acceleration_mm_per_sec2: 5000
    motor0:
      pulloff_mm: 4
      limit_neg_pin: gpio.36
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      tmc_2209:
        run_amps: 1.2
        step_pin: gpio.32
        direction_pin: gpio.33
        uart_num: 1
        addr: 0
        use_enable: false
        run_mode: StealthChop
        homing_mode: StealthChop
    homing:
      cycle: 2
      positive_direction: false
      seek_mm_per_min: 3000
      feed_mm_per_min: 600
      feed_scaler: 20
      settle_ms: 0
  y:
    max_rate_mm_per_min: 12000
    acceleration_mm_per_sec2: 5000
    motor0:
      pulloff_mm: 4
      limit_neg_pin: gpio.39
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      tmc_2209:
        run_amps: 1.2
        step_pin: gpio.25
        direction_pin: gpio.26
        uart_num: 1
        addr: 1
        run_mode: StealthChop
        homing_mode: StealthChop
    homing:
      cycle: 3
      positive_direction: false
      seek_mm_per_min: 3000
      feed_mm_per_min: 600
      feed_scaler: 20
      settle_ms: 0
coolant:
  flood_pin: NO_PIN
  mist_pin: NO_PIN
  delay_ms: 0
macros: {}
start:
  must_home: false
  deactivate_parking: false
  check_limits: false
arc_tolerance_mm: 0.002
junction_deviation_mm: 0.01
verbose_errors: true
report_inches: false
enable_parking_override_control: false
use_line_numbers: false
planner_blocks: 16
`

Vorobeyyyyyy, 2025-04-12:
Ive created PR to fix this. Iam new in contributing on github so dont know how to link properly
https://github.com/bdring/FluidNC/pull/1478

PR #1478 (2025-04-16): Fixes #1469 CoreXY releaseMotors
Fixes #1469
Fix coreXY releaseMotors function so both motors should be released for proper movement along X and Y axes
closed 2025-04-16 by MitchBradley