bdring/FluidNC #1470: 6x Axis Controller

eminfirtina, 2025-04-02:
### Wiki Search Terms

6x controller burned

### Controller Board

6x Axis Controller 

### Machine Description

I wanted to test the controller with one stepper first but I didnt even run it. 

### Input Circuits

```yaml

```

### Configuration file

```yaml
Couldn't come that point
```

### Startup Messages

```text
Nousb
```

### What happened?

Newly purchased 6x Axis Controller usb connection does not working at all. First I connected to 24V power supply then I plugged usb to the pc. It was kind of weird. Then it lost the connection in a couple of minutes. My voltage and everything is stable. Still it has 5V led light on. It seems normal but I changed the cable, I tried with a different computer nothing. It looks like usb chip is dead. Also I checked there is no any hotspot wifi. 

### Other Information

 order #246355. There is 24vdc and 0vdc and usb connected only.

eminfirtina, 2025-04-04:
Any assistance or guidance?

breiler, 2025-04-04:
What do you mean by this? What was wierd?
>  First I connected to 24V power supply then I plugged usb to the pc. It was kind of weird.

How were you connected, through what?
> Then it lost the connection in a couple of minutes.

Which cable?
> It seems normal but I changed the cable

What makes you think that?
> It looks like usb chip is dead.

One step at a time
> Also I checked there is no any hotspot wifi.

eminfirtina, 2025-04-04:
[image]

- The pc didnt load the drivers automatically. It had an exclamation mark on it.

- Usb was connected to the pc sir. So basically the USB cable was connected to the controller and the computer. 

- After the pc didn't see the controller from device manager, I tried another USB cable. Later I tried 4 different cables and a different pc. 

- I think my USB cables are in good condition. The reason I said usb chip is bc there is no connection with the pc. I hold the reset button didn't observe any changes, I hold the boot button again no difference. 

I also apologise for my poor explanation. I am a mechanical engineer who is trying to learn more electronic side recently. Thanks for assistance

MitchBradley, 2025-04-04:
Try installing the CP210x drivers from https://www.silabs.com/developer-tools/usb-to-uart-bridge-vcp-drivers?tab=downloads

bdring, 2025-04-04:
You need main power. Be careful with the polarity

eminfirtina, 2025-04-04:
The drivers installed. I used https://us.openbuilds.com/24v-meanwell-power-supply-bundle/ as ps. It only gives power to the controller. I can see 5V light as usual. I cant see any connection to the computer. I have a multimeter. I can check some voltage on the board with your guidelines. 
Thanks

MitchBradley, 2025-04-04:
Show a screenshot of DeviceManager with the COM ports.

[image]

eminfirtina, 2025-04-04:
[image]

bdring, 2025-04-04:
Where did you buy it? What was your order number?

eminfirtina, 2025-04-04:
Elecrow  Order number #246355

bdring, 2025-04-04:
You can contact them for a replacement.

You might try a different PC as a last check. All boards are tested and programmed at the factory via the USB.

closed 2026-02-06 by eminfirtina