bdring/FluidNC #1472: Feature: sending gcode files from HD

AntonioFromBrazil, 2025-04-03:
### Machine Context

would apply to any kind of machines

### Feature Description

Hello MitchBradley, congratulations on the excellent work. What do you think about implementing in the user interface a function to send gcode files from hd in addition to sending gtcode files from SD?

### Other Approaches

N/A

### How I Can Help

Unfortunatelly I am not a good developer...

MitchBradley, 2025-04-03:
If "HD" means "hard disk", there is no way to connect one to an ESP32.  Otherwise, I don't know what you mean.

bdring, 2025-04-03:
Use a gcode sender.

http://wiki.fluidnc.com/en/support/gcode_senders

MitchBradley, 2025-04-03:
Just to be completely clear - installer.fluidnc.com is for initial setup, not for running files.  To run a file, you must use either WebUI, which can only run files that are on the SD card, or a GCode Sender program as linked above.  The FluidNC firmware itself has no way to access files from a PC.

closed 2025-04-03 by MitchBradley

AntonioFromBrazil, 2025-04-03:
Yah yah... It is completely clear. Thanks