bdring/FluidNC #1474: Problem: Limit alarm triggered during limits test

daxliniere, 2025-04-07:
### Wiki Search Terms

(I wouldn't have the foggiest what to look for.)

### Controller Board

FNC 6X

### Machine Description

3-axis gantry
3x MKS Servo57D motor controllers

### Input Circuits

```yaml
5 limit switches (X-X+ Y-Y+ Z+) with common (6 pins).
But see below (not related
```

### Configuration file

```yaml
(Yes, XYZ are wired XZY. Cable length inside case with new controller.)

board: 6x
name: 6x Default
meta: Settings v1.0.0 (2025-01-14) Dax Liniere

stepping:
  engine: I2S_STREAM
  idle_ms: 254
  pulse_us: 4
  dir_delay_us: 1
  disable_delay_us: 0

axes:
  shared_stepper_disable_pin: NO_PIN
  x:
#motor1
    steps_per_mm: 800.000
    max_rate_mm_per_min: 3000.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 301.000
    soft_limits: true
    homing:
      cycle: 2
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 20.000
      seek_mm_per_min: 600.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.35
      limit_pos_pin: gpio.32
      limit_all_pin: NO_PIN
      hard_limits: true
      pulloff_mm: 1.000
      standard_stepper:
        step_pin: I2SO.2
        direction_pin: I2SO.1:low
        disable_pin: I2SO.000

  y:
#motor3
    steps_per_mm: 800.000
    max_rate_mm_per_min: 3000.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 186.000
    soft_limits: true
    homing:
      cycle: 2
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 20.000
      seek_mm_per_min: 600.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.33
      limit_pos_pin: gpio.26
      limit_all_pin: NO_PIN
      hard_limits: true
      pulloff_mm: 1.000
      standard_stepper:
        step_pin: I2SO.10
        direction_pin: I2SO.9:low
        disable_pin: I2SO.8

  z:
# motor2
    steps_per_mm: 800.000
    max_rate_mm_per_min: 1000.000
    acceleration_mm_per_sec2: 50.000
    max_travel_mm: 100.000
# was 46.000
    soft_limits: true
    homing:
      cycle: 1
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 20.000
      seek_mm_per_min: 200.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: gpio.2
      limit_all_pin: NO_PIN
      hard_limits: true
      pulloff_mm: 1.000
      standard_stepper:
        step_pin: I2SO.5
        direction_pin: I2SO.4:low
        disable_pin: I2SO.7

i2so:
  bck_pin: gpio.22
  data_pin: gpio.21
  ws_pin: gpio.17

spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

sdcard:
  card_detect_pin: NO_PIN
  cs_pin: gpio.5

10V:
 forward_pin: gpio.14
 reverse_pin: NO_PIN
 pwm_hz: 5000
 output_pin: gpio.13
 enable_pin: NO_PIN
 direction_pin: NO_PIN
 disable_with_s0: false
 s0_with_disable: true
 spinup_ms: 0
 spindown_ms: 0
 tool_num: 0
 speed_map: 0=0.000% 1000=0.000% 24000=100.000%
#update later with speed maps inc minimum 6000rpm
 off_on_alarm: false

probe:
  pin: gpio.34:low
  check_mode_start: true
#  toolsetter_pin: gpio.36:low

control:
  safety_door_pin: gpio.39:low
  #needs external PU resistor
  cycle_start_pin: NO_PIN
  feed_hold_pin: NO_PIN
  reset_pin: NO_PIN
  macro0_pin: gpio.36:low
  #needs external PU resistor
  macro1_pin: NO_PIN
  macro2_pin: NO_PIN
  macro3_pin: NO_PIN

coolant:
  flood_pin: gpio.12:low
# continuous air
  mist_pin:  gpio.4:low
# pulsed air 
  delay_ms: 0
  
parking:
  enable: true
  axis: Z
  pullout_distance_mm: 5.000
  pullout_rate_mm_per_min: 250.000
  target_mpos_mm: 0.000
  rate_mm_per_min: 800.000

#user_outputs:
#  analog0_pin: NO_PIN
#  analog1_pin: NO_PIN
#  analog2_pin: NO_PIN
#  analog3_pin: NO_PIN
#  analog0_hz: 5000
#  analog1_hz: 5000
#  analog2_hz: 5000
#  analog3_hz: 5000
#  digital0_pin: NO_PIN
#  digital1_pin: NO_PIN
#  digital2_pin: NO_PIN
#  digital3_pin: NO_PIN

start:
  must_home: false
  
macros:
  startup_line0:
  startup_line1:
  macro0: $H
#unlock and home XYZ
  macro1: G38.2 G91 Z-46 F100&G0 Z1&G38.2 G91 Z-2 F50&G10 L20 P1 Z4.985&G10 L20 P2 Z4.985&G10 L20 P3 Z4.985&G10 L20 P4 Z4.985&G10 L20 P5 Z4.985&G10 L20 P6 Z4.985&G53 G0 Z0&G90
#z-probe
  macro2:
  macro3: $SD/Run=lasertest.gcode

# #begin PWM
# pwm:
  # pwm_hz: 5000
  # direction_pin: NO_PIN
  # output_pin: gpio.13
  # enable_pin: gpio.14
  # disable_with_s0: false
  # s0_with_disable: true
  # spinup_ms: 0
  # spindown_ms: 0
  # tool_num: 0
  # speed_map: 0=0.000% 10000=100.000%
  # off_on_alarm: false

# #begin Laser
# Laser:
  # pwm_hz: 5000
  # output_pin: gpio.4
  # enable_pin: gpio.12
  # disable_with_s0: false
  # s0_with_disable: true
  # tool_num: 1
  # speed_map: 0=0.000% 255=100.000%
  # off_on_alarm: true
```

### Startup Messages

```text
[MSG:INFO: FluidNC v3.9.6 https://github.com/bdring/FluidNC]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.7-dirty]
[MSG:INFO: Local filesystem type is littlefs]
[MSG:INFO: Configuration file:config.yaml]
[MSG:INFO: Machine 6x Default]
[MSG:INFO: Board 6x]
[MSG:INFO: I2SO BCK:gpio.22 WS:gpio.17 DATA:gpio.21Min Pulse:2us]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
[MSG:INFO: SD Card cs_pin:gpio.5 detect:NO_PIN freq:8000000]
[MSG:INFO: Stepping:I2S_STREAM Pulse:4us Dsbl Delay:0us Dir Delay:1us Idle Delay:254ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Axis X (0.000,301.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:I2SO.2 Dir:I2SO.1:low Disable:I2SO.0]
[MSG:INFO:  Neg Limit gpio.35]
[MSG:INFO:  Pos Limit gpio.32]
[MSG:INFO: Axis Y (0.000,186.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:I2SO.10 Dir:I2SO.9:low Disable:I2SO.8]
[MSG:INFO:  Neg Limit gpio.33]
[MSG:INFO:  Pos Limit gpio.26]
[MSG:INFO: Axis Z (0.000,100.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:I2SO.5 Dir:I2SO.4:low Disable:I2SO.7]
[MSG:INFO:  Pos Limit gpio.2]
[MSG:INFO: safety_door_pin gpio.39:low]
[MSG:INFO: macro0_pin gpio.36:low]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: STA SSID is not set]
[MSG:INFO: AP SSID FluidNC IP 192.168.0.1 mask 255.255.255.0 channel 1]
[MSG:INFO: AP started]
[MSG:INFO: WiFi on]
[MSG:INFO: Captive Portal Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]
[MSG:INFO: 10V Spindle Ena:NO_PIN Out:gpio.13 Dir:NO_PIN Fwd:gpio.14 Rev:NO_PIN Freq:5000Hz Period:8191]
[MSG:INFO: Flood coolant gpio.12:low]
[MSG:INFO: Mist coolant gpio.4:low]
[MSG:INFO: Probe gpio.34:low]

Grbl 3.9 [FluidNC v3.9.6 (wifi) '$' for help]
```

### User Interface Software

UGS 2.1.13

### What happened?

I installed new 6X controller, updated to 3.9.6 firmware (UGS to 2.1.13). Attempted homing; X homed, Y homed, Z not responding (another issue). All axes jiggle slightly when another axis is moving (crosstalk, perhaps, but definitely another issue).
Powered down and attempted to debug jitteriness and Z axis issue, powered up, went to home and X threw alarm #8 (homing fail, couldn't clear limit switch. Initially I misread this as alarm #9, so I reconnected with FT and sent $limits.
During that test, I got the error you see in the screenshot:

### GCode File

N/A

### Other Information

[image]

breiler, 2025-04-07:
I have noticed that on some controllers it will always start with the switches reporting as activated. For me it is because they are configured as normally open and left floating. If I go through and trigger them one by one they will behave correctly.

MitchBradley, 2025-04-07:
To expand on what Bart said - $limits does not disable the hard limit alarm, so  it is best to turn off hard limits when using $limits for testing.  When $message/level=debug, you get messages about limit switch activations and deactivations, rendering $limits unnecessary.

[image]

Regarding the jitter - try inverting the step pulses to see if that makes a difference.

bdring, 2025-04-07:
Does the jitter add up to any net motion? Does it just jitter or slowly drift?

Most servos jitter as they feel changes in load.

token47, 2025-04-07:
I do use hard limits turned off all the time, it works well. I use TMC drivers and stallguard for homing and they also fire randomly during normal usage when the load is high enough, therefore the hard limits cannot be trusted. I have never seen any downsides on this, except that some times the logs get spammed with limit switches being activated messages (which are ignored, just annoying).

MitchBradley, 2025-04-07:
> For me it is because they are configured as normally open and left floating.

Yet another reason to prefer external pullups!  The other reason is that internal pullups tend to be weak with poorly-controlled values, often ranging from 50K to 100K.  That results in inconsistent timing.  The reason for poor internal pullups is that the silicon processes that make good, fast, small transistors do not result in good resistors with reasonable die areas.

daxliniere, 2025-04-07:
Thanks gents.
While looking for schematics for the input sections on the wiki, I discovered that 2 of the limit switches (2 & 26) require ```:pu```. Now that this is enabled, all 5 limit switches are working (though the 2 limit switches it affected were not ones _directly_ involved in the homing error).

> Test the switches manually. Do you get any false or missing activations?
Yep, always testing manually with **$limits** and no, all good (as long as it's being held during a redraw cycle).

> I would turn off hard limits while you are debugging the machine.
>$limits does not disable the hard limit alarm
Ahh, I never would have guessed that in a million years, which is why I reported it as a bug. Seems like hard limit alarms should be ignored when **$limits** is active since you can't (and shouldn't) issue motion commands during limit switch testing anyway. Don't you think?

>Add a reset button if you are worried about crashes.
I have door switch on an 'e-stop' style (mushroom) button and the main power switch is only 12" from that.

 
> What type of switches are you using? Send some pictures of your machine and wiring.
Microswitches like both of these (without lever arms)

[image]

> Turn on $message/level=debug
Can do - what am I looking for with this? If it's just limit switch testing, I'm satisfied everything is good there now.

> Start with the switch issue.
Done.

daxliniere, 2025-04-07:
> Does the jitter add up to any net motion? Does it just jitter or slowly drift?
> 
> Most servos jitter as they feel changes in load.

Axes that are not moving are jittering by random amounts. I had assumed crosstalk, so I replaced the 12-core cable I had with 3x 4-core shielded cables and I'm getting the same thing.

My next step will be to invert the pulses as per Mitch's suggestion (once I work out how to do that.)

MitchBradley, 2025-04-07:
Crosstalk between stepper motor cables is usually not a problem.  The impedance is too low, on the order of a few ohms.  You can get crosstalk from stepper cables to high impedance (10K or more) switch cables.

To invert the pulses, just add :low to the pin definition.

MitchBradley, 2025-04-07:
$message/level=debug is a better way to test switches than $limits.  It shows all the events, not just current state sampled infrequently.  I am disinclined to add more complexity to $limits, since there is a better way of testing.  Maybe if I had infinite time I would do it, or maybe I would eliminate $limits entirely.

daxliniere, 2025-04-11:
> Crosstalk between stepper motor cables is usually not a problem. The impedance is too low, on the order of a few ohms. You can get crosstalk from stepper cables to high impedance (10K or more) switch cables.

Okay, that's good to know, thanks.l

> To invert the pulses, just add :low to the pin definition.

Is that just for the step pin or dir and disable?

daxliniere, 2025-04-11:
Also, any idea why all 3 enable (disable?) LEDs change simultaneously even though I have defined individual pins and have **shared_stepper_disable_pin: NO_PIN**?

daxliniere, 2025-04-11:
@MitchBradley To simplify my wiring setup, would there be any problem if I used the shield of the 4-core cable as the ground for the motor? The motor positive supply is already a separate wire running parallel to the 4-core signal cable. (I currently have the motor negative supply wire dangling in a temporary setup.)

Would _this_ introduce jitter/crosstalk?

bdring, 2025-04-11:
## Stepper Enables

Most machines should use `idle_ms: 255` to keep the motors on. They could drift after homing if they disable. Change your config.

All enables work at the same time regardless of whether you use a shared or separate pins. The only advantage of separate enables is that you can manually disable one motor with $MD=X

bdring, 2025-04-11:
Are you seeing any step pin LED activity associated with the jittering motors?

daxliniere, 2025-04-11:
> Most machines should use `idle_ms: 255` to keep the motors on. They could drift after homing if they disable. Change your config.

Will do, thanks.

> All enables work at the same time regardless of whether you use a shared or separate pins.

Why? Surely if you are moving in a single axis you would get a cleaner result if non-moving axes had their motors held in position?

daxliniere, 2025-04-11:
Maybe it would be a good time to get you to review my config.yaml for this 6X.

daxliniere, 2025-04-11:
Enables or Disables? Motors are always marked as 'En', but 6X PCB and config.yaml both say disables.

bdring, 2025-04-11:
I2S pins do not have a defined startup. We set them as soon as we can

bdring, 2025-04-11:
Do you know the signal pulse timing? I did not see it in the manual.

Some have really long delays especially enable.

MitchBradley, 2025-04-11:
Very few people use MKS Servo 57, so we do not have much experience with what problems can occur.  Mine is sitting in a box, having been tested briefly.  I have no long-term production experience with it.

daxliniere, 2025-04-12:
> I2S pins do not have a defined startup. We set them as soon as we can

Okay, phew! I was worried that something was wrong since they seemed to be somewhat random each power-on.

daxliniere, 2025-04-12:
> Very few people use MKS Servo 57, so we do not have much experience with what problems can occur. Mine is sitting in a box, having been tested briefly. I have no long-term production experience with it.

Then I am more than happy to help getting them to a known configuration.

daxliniere, 2025-04-15:
Hey @bdring and @MitchBradley, I looked at the 6X schematic and it seems that the I2SO ICs are set up for common cathode, not common anode. How will this work with the Servo57's opto inputs being common anode? Any idea why it was working at all?

bdring, 2025-04-15:
The I2SO chips can drive and sink current. They are not "common" anything. Each can can be connected to either side of an LED.

If you need a common (+) on the servo side, then source it from another location on the controller. You can invert the pulse pin if the active state does not match what the servo expects.

[image]

daxliniere, 2025-04-15:
Thanks Bart. So I would disconnect the servos' common pins from 6X's motor COM terminals and connect altogether to 6X's H1 +5v pin, yes?

bdring, 2025-04-15:
I don't think I ever called them a COM terminal. It is labeled Gnd.

My reading of the diagram would put + on the common terminal. I assume 5v is OK.

We don't support RS485 for motion control. It is too slow for stepping and too hard to coordinate and time for planner segments.

MitchBradley, 2025-04-15:
From https://github.com/makerbase-motor/MKS-SERVO57D/blob/master/Hardware/MKS%20SERVO57D%20V1.0_001%20schematic.pdf

[image]

Looks like 5V on COM will work.

daxliniere, 2025-04-15:
> I don't think I ever called them a COM terminal.

You are absolutely correct, of course. Though this got me thinking about a revision to the PCB.

I know you're busy with the new expander PCBs, but I wonder if it would be worthwhile to change the 6 motor GND terminals to COM and add a 3-pin header and jumper to allow COM to be set to either GND or +5v?

MitchBradley, 2025-04-15:
The label COM exists on the Servo57, of which you are the only known user.  Most external drivers use per-input labels like PUL+,PUL-, DIR+, DIR-, ENA+, ENA-.  So it does not seem like relabeling it as COM would be generally beneficial.

daxliniere, 2025-04-15:
Oh, I don't mean just relabeling it, I was suggesting making the next revision 6X have selectable +5v or GND for the COM terminals (currently GND terminals).

daxliniere, 2025-04-15:
As an aside, I received some LED pushbutton samples recently which have 'no polarity' - in other words, they have inverse-wired a pair of LEDs so input polarity becomes irrelevant and it got me wondering why there aren't any optocouplers like that.

MitchBradley, 2025-04-15:
LTV-814S

daxliniere, 2025-04-23:
> I2S pins do not have a defined startup. We set them as soon as we can

Some of the pins on mine have continuity to ground when the controller is off. Is that correct??

bdring, 2025-04-23:
Yes, they are either low impedance to ground or 5v.

daxliniere, 2025-04-23:
Alright. Cool.
However, I am not getting any output from them. Specifically, I just tested Motor1 output (both Step and Dir have continuity to GND), I then piggy-backed that servo (X-axis) to the working Z-axis (Motor2 output) and it works perfectly.
I will next try connecting it to another Motor output.

Is there a program I could run which would aid in testing? Like something with constant motion on all 6 axes?

daxliniere, 2025-04-23:
ARRGGHHHHH, I'm so fed up.
Now look:

daxliniere, 2025-04-23:
Right. Now that I've calmed down, I have determined the Z-axis stepper controller (SERVO57D), the only axis that was working correctly, now has some problem and seems to be overloading the PSU and causing it to drop out (the cycling you see in that video).
X and Y axes power is connected and there is PSU is fine. 6X seems fine, too, but will have to test tomorrow when my patience has regenerated.

daxliniere, 2025-04-23:
I wanted to try one more thing before I went home. axis:motor# X:2 Y:4 Z:5

X axis (where Z-axis was previously connected) works fine.
Y-axis does fuck all, not even 6X's LEDs change. Sent Y5 and Y-5.
Z-axis motor not connected, but 6X LEDs also do not change.

Changed config to assign Z to motor6 - still no LED action.
I'm going home.

bdring, 2025-04-23:
The flashing LEDs could have been due to overloading the 5V power supply. If it is overloaded it shuts down. It is rated for at least 3A, so the overload was likely a short.

If it was shorted through the stepper motor driver terminals the I2SO chip was likely damaged. They are only rated for 25mA per pin and 75mA total for each chip. Typically when they are damaged, the whole chip goes bad.

We should set up a time to carefully assess the controller. I am free tomorrow afternoon US central time.

daxliniere, 2025-04-23:
Thanks Bart, sounds great. I'll PM you through Discord.