bdring/FluidNC #1476: Problem:  Configuration issue

Starejpes, 2025-04-11:
### Wiki Search Terms

Hi,
I am using FluidNC with BTT Rodent and CoreXY kinematics.
I set homing section following FluidNC Wiki as follows (for X and Y axis): 
    homing:
      cycle: 2
      positive_direction: false
      mpos_mm: 50.000
      feed_mm_per_min: 200.000
      seek_mm_per_min: 2000.000
      settle_ms: 500
      seek_scaler: 1.1
      feed_scaler: 1.1
The issue was that motors was rumbling on X and Y axis when moving to left side of sketch. I solved the issue by changing mpos_mm: -50.000. If there is more clever solution please let me know.
Kind regards

### Controller Board

BTT Rodent V1.1

### Machine Description

CoreXY Laser cutter

### Input Circuits

```yaml

```

### Configuration file

```yaml
board: BTT Rodent V1.0
name: BTT CNC
meta: (04.12.2024) by BTT
kinematics:
  Corexy:
stepping:
  engine: I2S_STREAM
  idle_ms: 255
  pulse_us: 4
  dir_delay_us: 1
  disable_delay_us: 0
axes:
  shared_stepper_disable_pin: NO_PIN
  x:
    steps_per_mm: 160.000
    max_rate_mm_per_min: 5000.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 500.000
    soft_limits: false
    homing:
      cycle: 2
      positive_direction: false
      mpos_mm: -10.000
      feed_mm_per_min: 200.000
      seek_mm_per_min: 2000.000
      settle_ms: 500
      seek_scaler: 1.1
      feed_scaler: 1.1
    motor0:
      limit_neg_pin: gpio.35
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 5
      tmc_5160:
        step_pin: I2SO.2
        direction_pin: I2SO.1:low
        disable_pin: I2SO.0
        cs_pin: gpio.5
        spi_index: 1
        r_sense_ohms: 0.022
        run_amps: 1.5
        hold_amps: 0.5
        microsteps: 32
        toff_disable: 0
        toff_stealthchop: 5
        use_enable: false
        run_mode: CoolStep
        homing_mode: CoolStep
        stallguard: 16
        stallguard_debug: false
        toff_coolstep: 3
        tpfd: 4
  y:
    steps_per_mm: 160.000
    max_rate_mm_per_min: 5000.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 300.000
    soft_limits: false
    homing:
      cycle: 1
      positive_direction: false
      mpos_mm: -10.000
      feed_mm_per_min: 200.000
      seek_mm_per_min: 2000.000
      settle_ms: 500
      seek_scaler: 1.1
      feed_scaler: 1.1
    motor0:
      limit_neg_pin: gpio.34
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 5
      tmc_5160:
        step_pin: I2SO.5
        direction_pin: I2SO.4
        disable_pin: I2SO.7
        cs_pin: NO_PIN
        spi_index: 2
        r_sense_ohms: 0.022
        run_amps: 1.5
        hold_amps: 0.5
        microsteps: 32
        toff_disable: 0
        toff_stealthchop: 5
        use_enable: false
        run_mode: CoolStep
        homing_mode: CoolStep
        stallguard: 16
        stallguard_debug: false
        toff_coolstep: 3
        tpfd: 4
  z:
    steps_per_mm: 800
    max_rate_mm_per_min: 5000
    acceleration_mm_per_sec2: 100
    max_travel_mm: 300
    soft_limits: false
#    homing:
#      cycle: 3
#      positive_direction: true
#      mpos_mm: 150
#      feed_mm_per_min: 100
#      seek_mm_per_min: 800
#      settle_ms: 500
#      seek_scaler: 1.1
#      feed_scaler: 1.1
    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1
      tmc_5160:
        step_pin: I2SO.10
        direction_pin: I2SO.9
        disable_pin: I2SO.8
        cs_pin: NO_PIN
        spi_index: 3
        r_sense_ohms: 0.022
        run_amps: 0.5
        hold_amps: 0.5
        microsteps: 8
        toff_disable: 0
        toff_stealthchop: 5
        use_enable: false
        run_mode: CoolStep
        homing_mode: CoolStep
        stallguard: 16
        stallguard_debug: false
        toff_coolstep: 3
        tpfd: 4
  a:
    steps_per_mm: 157.75
    max_rate_mm_per_min: 18000
    acceleration_mm_per_sec2: 1500
    max_travel_mm: 325
    soft_limits: true
#    homing:
#      cycle: 4
#      positive_direction: true
#      mpos_mm: 150
#      feed_mm_per_min: 100
#      seek_mm_per_min: 800
#      settle_ms: 500
#      seek_scaler: 1.1
#      feed_scaler: 1.1
    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1
      tmc_5160:
        step_pin: I2SO.13
        direction_pin: I2SO.12
        disable_pin: I2SO.15
        cs_pin: NO_PIN
        spi_index: 4
        r_sense_ohms: 0.022
        run_amps: 0.5
        hold_amps: 0.5
        microsteps: 16
        toff_disable: 0
        toff_stealthchop: 5
        use_enable: false
        run_mode: CoolStep
        homing_mode: CoolStep
        stallguard: 16
        stallguard_debug: false
        toff_coolstep: 3
        tpfd: 4
  homing_runs: 2
i2so:
  bck_pin: gpio.22
  data_pin: gpio.21
  ws_pin: gpio.17
spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18
sdcard:
  cs_pin: gpio.0
  card_detect_pin: NO_PIN
  frequency_hz: 8000000
#uart1:
#  txd_pin: gpio.15
#  rxd_pin: gpio.16
#  rts_pin: gpio.14
#  baud: 9600
#  mode: 8N1
#Huanyang:
#  uart_num: 1
#  modbus_id: 1
#  tool_num: 0
#  speed_map: 0=0% 0=25% 6000=25% 24000=100%
#  off_on_alarm: false
i2c0:
  sda_pin: gpio.27
  scl_pin: gpio.26
status_outputs:
  report_interval_ms: 500
  idle_pin: I2SO.11:low
  run_pin: I2SO.14:low
  alarm_pin: I2SO.3:low
Laser:
  pwm_hz: 2000
  output_pin: gpio.15:high
  enable_pin: gpio.14:high
  disable_with_S0: false
  s0_with_disable: true
  tool_num: 0
  speed_map: 0=0.000% 1000=100.000%
  off_on_alarm: true
start:
  must_home: false
```

### Startup Messages

```text
OK
```

### User Interface Software

WebUI, GRBL, LightBurn

### What happened?

Homing was OK.
Motors was rumbling on X and Y axis when trying to cut far left and down.

### GCode File

Any

MitchBradley, 2025-04-11:
Lightburn's zero position can be very confusing, so it is hard to answer your question without knowing exactly how you set the origin in Lightburn and where the sketch is located in that coordinate system.

Starejpes, 2025-04-12:
Hi Mitch,
You are right, LightBurn is sometime confusing. But my comment on situation was observed using GRBL actually.
When I set mpos_mm: to positive value having homing to negative on X and Y GRBL shows this value as current location. Gcode Instruction to go to position X=0, Z=0 or value less than that value mpos_mm: resulted in overrun physical structure.
My intention of the comment was to add into documentation something like:
"Having end-switches on negative X, Y axes will induce to set mpos_mm: value to negative as well".
Maybe it will save other users couple of hours investigating what is wrong with their project.
Cheers Starejpes

MitchBradley, 2025-04-12:
> When I set mpos_mm: to positive value having homing to negative on X and Y GRBL shows this value as current location

What exact interaction with GRBL "showed the value as current location"?  Was it a response to the "?" status report inquiry?  Was $10 AKA $Report/Status set to report WPos or MPos?

The reason that this is important is because GCode always operates in work coordinates, except for the rare case when there is a G53 on the line.  If the status report says that the MPos coordinate is 50, and you do not know the work coordinate offset for the currently-selected coordinate system (G54 .. G59), then when you say 'G0 X0 Y0`, it might go anywhere, because 50 in machine coordinates could be anything at all in work coordinates, depending on where you were the last time that you zeroed the work coordinate system.

The typical workflow is to home the machine, thus establishing the machine coordinates for the purpose of soft limits, then jog (incremental moves) to an appropriate position relative to the stock, and then zero the work coordinates.  The work coordinate offset (per work coordinate system) is persistent across reboots, so the next time that you home the machine, the work zero will be in the same position as before.

closed 2025-08-17 by bdring