bdring/FluidNC #1481: Problem:

HerbertHenri, 2025-04-17:
### Wiki Search Terms

Axis drifting/moving towards positive while running a G-code

### Controller Board

6x_CNC_Controller, I bought from elecrow few months ago.

### Machine Description

XYYZ setup, all X and Y axis with SFU1604, and Z axis with SFU1204 ballscrews, and linear rails on all axis.
The machine is DIY but quite solid, I double checked for square, and backlash on Y and X axis are abou 0.01mm (what I could measure with my gauge).

So think mechanically speaking the machine should be fine...

### Input Circuits

```yaml

```

### Configuration file

```yaml
board: 6x CNC Controller
name: 6x XYYZ PWM
stepping:
  engine: I2S_STREAM
  idle_ms: 255
  pulse_us: 4
  dir_delay_us: 1
  disable_delay_us: 0

axes:
  shared_stepper_disable_pin: NO_PIN
  x:
    steps_per_mm: 640.675
    max_rate_mm_per_min: 1950.000
    acceleration_mm_per_sec2: 200.000
    max_travel_mm: 400.000
    soft_limits: false
    homing:
      cycle: 2
      positive_direction: false
      mpos_mm: 0
      feed_mm_per_min: 100.000
      seek_mm_per_min: 200.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.2:low:pu
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1.000
      standard_stepper:
        step_pin: I2SO.2
        direction_pin: I2SO.1
        disable_pin: I2SO.0

  y:
    steps_per_mm: 639.830
    max_rate_mm_per_min: 1900.000
    acceleration_mm_per_sec2: 200.000
    max_travel_mm: 400.000
    soft_limits: false
    homing:
      cycle: 2
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 200.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.26:low:pu
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1.000
      standard_stepper:
        step_pin: I2SO.5
        direction_pin: I2SO.4:low
        disable_pin: I2SO.7

    motor1:
      #limit_neg_pin: gpio.33:low
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1.000
      standard_stepper:
        step_pin: I2SO.10
        direction_pin: I2SO.9:low
        disable_pin: I2SO.8

  z:
    steps_per_mm: 807.265
    max_rate_mm_per_min: 350.000
    acceleration_mm_per_sec2: 70.000
    max_travel_mm: 200.000
    soft_limits: false
    homing:
      cycle: 1
      positive_direction: true
      mpos_mm: 0.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 800.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.32:low
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 3.000
      standard_stepper:
        step_pin: I2SO.13
        direction_pin: I2SO.12
        disable_pin: I2SO.15

i2so:
  bck_pin: gpio.22
  data_pin: gpio.21
  ws_pin: gpio.17

spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

sdcard:
  card_detect_pin: NO_PIN
  cs_pin: gpio.5

probe:
  pin: gpio.39:low
  toolsetter_pin: gpio.36:low

start:
  must_home: false
```

### Startup Messages

```text
[MSG:INFO: FluidNC v3.9.4 https://github.com/bdring/FluidNC]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.7-dirty]
[MSG:INFO: Local filesystem type is littlefs]
[MSG:INFO: Configuration file:config.yaml]
[MSG:INFO: Machine 6x XYYZ PWM]
[MSG:INFO: Board 6x CNC Controller]
[MSG:INFO: I2SO BCK:gpio.22 WS:gpio.17 DATA:gpio.21Min Pulse:2us]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
[MSG:INFO: SD Card cs_pin:gpio.5 detect:NO_PIN freq:8000000]
[MSG:INFO: Stepping:I2S_STREAM Pulse:4us Dsbl Delay:0us Dir Delay:1us Idle Delay:255ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Axis X (0.000,400.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:I2SO.2 Dir:I2SO.1 Disable:I2SO.0]
[MSG:INFO:  X Neg Limit gpio.2:low:pu]
[MSG:INFO: Axis Y (0.000,400.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:I2SO.5 Dir:I2SO.4:low Disable:I2SO.7]
[MSG:INFO:  Y Neg Limit gpio.26:low:pu]
[MSG:INFO:   Motor1]
[MSG:INFO:     standard_stepper Step:I2SO.10 Dir:I2SO.9:low Disable:I2SO.8]
[MSG:INFO: Axis Z (-200.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:I2SO.13 Dir:I2SO.12 Disable:I2SO.15]
[MSG:INFO:  Z Neg Limit gpio.32:low]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: Connecting to STA SSID:Wi-fi 2.4.ghz]
[MSG:INFO: Connecting.]
[MSG:INFO: Connecting..]
[MSG:INFO: Connecting...]
[MSG:INFO: Connected - IP is 192.168.100.248]
[MSG:INFO: WiFi on]
[MSG:INFO: Start mDNS with hostname:http://FluidNC.local/]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]
[MSG:INFO: Probe gpio.39:low]
[MSG:INFO: Toolsetter gpio.36:low]
ok
```

### User Interface Software

CNCjs 1.10.5

### What happened?

I noticed whatever g-code  I run, all axis seems to move towards positive directions ( XY diagonally) and Z axis going up.

I tought maybe it was the machine or something, maybe the g-code, I generated the g-code on Autodesk Fusion CAM, then I tried using Aspire 12, and all the times that happens.

Last time was a Aspire g-code, I was carving a simple slot, and the slot went down like a stair, the left/top side of it goes like a ramp, and the Z axis goes up slowly, when I used MACH3, on the same machine it never really happened, wich makes me think it's something with the new board/ gcode sender (CNCjs) / fluidnc/ my configs/ maybe wiring?,  I really dont know what could it be, but i'm almost 100% sure it shouldn't be a mechanical issue... the machine is quite solid and the G-code also should be fine?

I haven't noticed yet, because most of the times i did cuts, were 1 -3 passes at most, so this time I was machining some aluminium, and noticed that happening, tried the same code, same everything on a piece of MDF (thinking it could be the metal), same thing happened, all 3 axis 'drifting away'.

Anyway, thanks in advance for the help!

### GCode File

[slot.txt](https://github.com/user-attachments/files/19786028/slot.txt)

the g-code was acctually as 'slot.gcode' i just changed to be able to upload... it's a simple slot, but with very shallow passes since it was aluminum.

### Other Information

Bart asked me info about my steppers, i'm running some chinese nema23 ones, exactly like the ones below

[image]

Nema23 Stepper motor 4.4A 2.3N.M - Model j-5718HB4401, with some DM556 drivers, on all axis they're the same, and 2, 48V PSU, I also don't think it's a PSU problem, or motors itself, since when I was with MACH3, nothing like that ever happened... but idk, any help is appreciated!

HerbertHenri, 2025-04-17:
Drivers are setup to 3200 microsteps, just read on the wiki anything over 100 is too much? I don't really know about that, I just did what a friend recommended, and like mentioned, used to work fine with MACH3 so Idk...

MitchBradley, 2025-04-17:
Try adding :low to the end of the step_pin definitions, like this: step_pin: I2SO.2:low

MitchBradley, 2025-04-17:
If the pulses are inverted compared to what the driver expects, it is possible that the driver might be inferring steps during some situations that should be treated as not being steps.  I am not sure that this is what is happening, but I want to eliminate the possibility.

HerbertHenri, 2025-04-17:
But if that was the case, wouldn't the steppers move in the other direction? 🤔

bdring, 2025-04-17:
It would not go the wrong direction, but could misread the pulse. It acts on the wrong side of the pulse. It is unlikely the issue, but worth checking.

Do you see any LED activity on axes that shot not be moving? 

I can test your config, but I will not be home until next week.

HerbertHenri, 2025-04-17:
I'll be home in a few minutes and i'll try the pin settings and will see if any led lights up.

bdring, 2025-04-17:
Try looking for led activity in both active modes.

It is really hard to see fast off blinks.

HerbertHenri, 2025-04-17:
So I did the :low thing, nothing changed, and the leds apparently are working as they should.

I mentioned my microstepping is at 3200, is that ok? too high? Is there any chance thats the problem?

bdring, 2025-04-17:
That is excessive. It could be a problem at high speeds.

Most people use 80 to 100 steps per mm.

HerbertHenri, 2025-04-17:
But could that be the cause of the problem? I ran the same g-code just now, the Z axis went like 5-6mm up, and the drift kept happening (about 1mm to the side...)

My steps per mm are ate 640~ on xy and 800 ish at Z, should then i decrease the microstepping? Wich one you'd recommend that will still be the most accurate as possible?

HerbertHenri, 2025-04-17:
> Try looking for led activity in both active modes.
> 
> It is really hard to see fast off blinks.

I was watching that, they blinked when the axis moved, no blinking was done when the axis wasnt moving (as far as i could see), both Y axis blinked togheter, i think the board might be fine... maybe my config.yaml? Idk what else to try....  and XY drift is about a 1mm on that specific g-code, and Z was about 5-6mm (going up)

bdring, 2025-04-17:
I don't know. I'll stop speculating. I will test next week.

HerbertHenri, 2025-04-17:
Ok, in the meanwhile i'll decrease the microstepping and try again. I'll be waiting on a possible solution!

HerbertHenri, 2025-04-17:
[image]

Just did these 3 slot cuts right now....

the top right one, looks okay, but it's not, the other 2 is clearly visible how much the machine is moving, the bottom right , i just did with a Fusion 360 g-code, and after reducing my microstepping from 3200 -> 800, it drifted even more, even tho the machine is moving as intended....

mikeoverbay, 2025-04-17:
did you make sure your coupliing isn't slipping?
Maybe mark it with a sharpie and see if the screw or motor shaft slips?

HerbertHenri, 2025-04-17:
> did you make sure your coupliing isn't slipping? Maybe mark it with a sharpie and see if the screw or motor shaft slips?

Did not check that... (as far as i know it shouldn't be... but doesn't hurt to check!), brb in a few.

But like the drift is in both X and Y, i run 2 Y axis, and they dont miss align.... wouldn't it need to all 4 couplings be slipping? coz  i'm experiencing drift in all 3 axis... XY going top/right, and Z going up

HerbertHenri, 2025-04-17:
> did you make sure your coupliing isn't slipping? Maybe mark it with a sharpie and see if the screw or motor shaft slips?

nothing slipping, mechanically speaking everything is fine

MitchBradley, 2025-04-17:
With external drivers, 3200 is not "microstepping", but instead "pulses/rev".  Nearly all motors are 200 full steps/rev, 3200 pulses/rev divided by 200 full steps/rev is 16 pulses/full step , so the microstepping level is 16.  That is a very typical microstepping level that, by itself, is not likely to be a problem.

[image]

HerbertHenri, 2025-04-17:
> With external drivers, 3200 is not "microstepping", but instead "pulses/rev". Nearly all motors are 200 full steps/rev, 3200 pulses/rev divided by 200 full steps/rev is 16 pulses/full step , so the [...]

Yes! that what i meant by 3200!

but i changed to 800 now, to see if anything changes (surprisingly the machine feels like its running smoother...) but the drift keeps happening, Z axis goes up like 8mm after that specific g-code, and XY moves like 2-3mm (towards the positive)...

I re-checkd all couplings, screws, squarenenss, everything mechanical-wise is ok...

I was googling like crazy, and could it be wiring? some sort of interference? ( i read that spindles can cause that... but i use a makita router? idk), bad connections? (idk, i made sure everything is tight and well ajusted on the wiring... I really have no idea...

Maybe my config.yaml is messed? I picked up a xyyz example and edited what i needed only... 

Any kind of suggesting i'm willing to try haha

MitchBradley, 2025-04-17:
I am trying to create a test environment that is close to your setup.

Meanwhile, here are two things to try:

1. Run the program with the router turned off - an "air cut" without a bit.  That will eliminate the possibility of interference from the spindle.
2. change the steps_per_mm values to integers to see if there could be a rounding error in the math.  Obviously that will change the dimensions a little, but we can see if that affects the drift.

HerbertHenri, 2025-04-17:
> I am trying to create a test environment that is close to your setup.
>
> Meanwhile, here are two things to try:
>
> 1. Run the program with the router turned off - an "air cut" without a bit.  That will [...]

Did both, still drifting :/, and Z axis still going up (or not going down properly?! idk, the thing is when i 0 the z axis againg theres a somewhat consistend 8mm increase.

HerbertHenri, 2025-04-18:
One thing i noticed, the drifting with the fusion 360 code is way bigger than the aspire gcode, idk why

MitchBradley, 2025-04-18:
I am using your config file - as given above in the initial posting - on an old 6-pack board with external driver adapters.  I have one motor connected to X.  The motor is an IHSV57 integrated driver/stepper combo set to 3200 pulses/rev.  The motor is not connected to any mechanical system but I have a tape "flag" on the shaft so I can see the rotational position.

When I run your program, the X motor first rotates a little more than 1 turn, which is consistent with the move from X0 to X5.5 (5.5 mm * 640 steps/mm  / 3200 steps/rev= 1.1 rev).  It then shuttles back and for about 0.4 revs - again consistent with the movement from X4.4 to X6.6.  At the end of the program, it returns to the original starting point at X=0.

I am going to try with Y

MitchBradley, 2025-04-18:
With the motor on the first Y connector, the behavior is as expected.  Several initial revolutions to get to Y49.6, then back and forth shuttling along the length of the slot, then a final return to the starting position.

Please post detailed photos of the wiring between you controller board and the stepper drivers.

HerbertHenri, 2025-04-18:
[image] 

[image]

[image]

[image]

I tried my best to show how it's wired,,, but all drivers are wired the same

HerbertHenri, 2025-04-18:
> With the motor on the first Y connector, the behavior is as expected. Several initial revolutions to get to Y49.6, then back and forth shuttling along the length of the slot, then a final return to [...]

The g-code is probably fine... 
Is there any chance CNCjs is messing up? I dunno...

MitchBradley, 2025-04-18:
Do your drivers have a switch to set the input sensitivity?

[image]

MitchBradley, 2025-04-18:
I suppose it could be possible that CNCjs is sending something wrong.  Please try with FluidTerm, sending "$sd/run=slot.gc".

HerbertHenri, 2025-04-18:
> Do your drivers have a switch to set the input sensitivity?
> 
> [image]

nope, it does not, my drive is this one https://oceantech-automation.com.br/produtos/driver-dm556/, exactly this one, it's pretty standard i guess

HerbertHenri, 2025-04-18:
> $sd/run=slot.gc

[MSG:ERR: sdmmc_card_init failed code 0x108]

i got that, should i have a sd card? cncjs is running on a raspberry pi 3b, with an ssd

MitchBradley, 2025-04-18:
Well that certainly makes things more complicated.  I have been running my tests with WebUI and FluidTerm using an SD card.

I have to go now.  I will try with CNCjs on a PC.  I don't have time to set it up on a Pi.

HerbertHenri, 2025-04-18:
> Well that certainly makes things more complicated. I have been running my tests with WebUI and FluidTerm using an SD card.
> 
> I have to go now. I will try with CNCjs on a PC. I don't have time to set it up on a Pi.

how do i send a gcode from the web ui? I could test it and see if it's cncjs messing up

MitchBradley, 2025-04-18:
WebUI is not a sender.  It lets you run code from the SD card but it does not send from files on the host.

HerbertHenri, 2025-04-18:
> WebUI is not a sender. It lets you run code from the SD card but it does not send from files on the host.

I'll be trying with another sender, and see if anything changes... and i'll report soon

HerbertHenri, 2025-04-18:
Could it be a connection problem due to the thing being transmited via wi-fi? (altough, the cncjs on the Pi, is connect tru wire... idk)

MitchBradley, 2025-04-18:
I tried with CNCjs 1.10.4 on Windows - the Electron App version.  It behaved the same as with WebUI - the motors returned to the starting position.

One thing you could try is to disconnect the ENA wires, so the drivers are always enabled.  See if that changes anything.  Then disconnect the Y and Z drivers to see if the drift happens when only moving in X.

HerbertHenri, 2025-04-19:
> I tried with CNCjs 1.10.4 on Windows - the Electron App version. It behaved the same as with WebUI - the motors returned to the starting position.
>
> One thing you could try is to disconnect the ENA [...]

So i disconnect all the ENA wires? positive and negative? on all axes or just one of the axis? I'll be trying it in a moment and see is anything changes, i'm almost at the verge of losing my mind...

MitchBradley, 2025-04-19:
An electrical circuit requires both wires to work, so it suffices to disconnect either the positive wire, the negative wire, or both.  I would try disconnecting ENA on all axes.  ENA is really "disable", so when it is not connected, the driver will always be enabled.

I do not expect this to help, but since I cannot directly debug your setup, and I cannot reproduce your problem with the hardware I have, I am fishing for things to try.

I am going to get a 6x board next week; maybe I will be able to reproduce the problem with it.

> i'm almost at the verge of losing my mind...

Me too, and yours is not the only problem I am trying to solve on hardware I do not have.

HerbertHenri, 2025-04-19:
> An electrical circuit requires both wires to work, so it suffices to disconnect either the positive wire, the negative wire, or both. I would try disconnecting ENA on all axes. ENA is really [...]

haha one last effort, I said mechanically wise, the machine is ok... but if it wasn't, what could possibly cause this kind of drift?

I'll disconnect the motors and run the program like you did, and see if they go back to the same position... Idk, could be something i missed.... (probably not, but i'm willing to try anything now...)

HerbertHenri, 2025-04-19:
Also, I mentioned that i use a makita router, is there anyway i can measure if it's causing any kind of interference?

MitchBradley, 2025-04-19:
Sure, try an air cut with the router turned off

HerbertHenri, 2025-04-20:
I did both tests, nothing changed, and apparently it's moving XY consistently 2.5mm towards positive.... I know the step downs are little (0.1mm) but XY shouldn't drift anyway (tried that on an aircut, with router off and plugged off...) 

So now i'm nearly 100% it's eletronics or my config.yaml may be fuckd... IDK...

MitchBradley, 2025-04-20:
I created this page http://wiki.fluidnc.com/en/support/setup/drift with everything I could think of that might cause axis drift.

There is not much more I can do until I get a 6x board next week.

bdring, 2025-04-22:
I have a 6x controller running with your original config file. 

I have a logic analyzer attached to the Y, Y and Z step pins. I set it up to trigger on any step of a selected axis. I have not seen any extra steps yet. I am running long moves lasting as long as 10 minutes per move at full speed.

I have tested the setup with moves on the instrumented axes and it exactly records the correct amount of steps.

Do you get drift if you just move back and forth on X?

[image]

MitchBradley, 2025-04-22:
Please tell us exactly how you created that gcode file, CAD and CAM.  It has many sequences of very short segments, all in the same direction, like:
```
G1Y50.687
G1Y50.901
G1Y51.116
G1Y51.330
...
G1Y58.407
G1Y58.621
G1Y58.640
```
Most CAD/CAM setups would just generate a single move `G1Y58.640`.  The short moves ought to work, but we are wondering how this strange gcode came about.

bdring, 2025-04-22:
My logic analyzer has a stepper motor decoder. It counts up or down based on the direction signal. Due to the analyzer memory issues, I can only capture a few of the loops and return to x0y0, but it never shows any offset.

The counter is the total step count.

[image]

MitchBradley, 2025-04-22:
Can you configure your post to not generate G2 arc moves, instead using line segments to approximate the arcs?  I want to see if there is something going wrong with arc computation in FluidNC.

MitchBradley, 2025-04-22:
I have instrumented the arc computation code and found nothing of interest.

Another thing to try - disconnect all the Z driver wires at the controller and run the program to see if X and Y still drift.

HerbertHenri, 2025-04-23:
> I have a 6x controller running with your original config file.
>
> I have a logic analyzer attached to the Y, Y and Z step pins. I set it up to trigger on any step of a selected axis. I have not seen [...]

Not that i've noticed, i can run some tests tomorrow morning (23pm rn), and see if something happens, but i guess not... the drift is happening in both X and Y, and some how Z is also going up,

HerbertHenri, 2025-04-23:
> Please tell us exactly how you created that gcode file, CAD and CAM. It has many sequences of very short segments, all in the same direction, like:
>
> ```
> G1Y50.687
> G1Y50.901
> G1Y51.116
> G1Y51.330
> ... [...]

This one was generated on Aspire, i just selected the profile, and told it to cut (inside), and stepdowns of 0.1mm, I did pretty much the same on Fusion 360, a profile pass, 0.1mm, and in fusion the drift looks even bigger!

HerbertHenri, 2025-04-23:
> Can you configure your post to not generate G2 arc moves, instead using line segments to approximate the arcs? I want to see if there is something going wrong with arc computation in FluidNC.

Not sure, but i can try...

HerbertHenri, 2025-04-23:
> driver wires at the controller and run the program to see if X and Y still drift.

Ok, i'll test everything tomorrow morning.

Also i just bought one easy servo, to use on the X axis, as soon as it arrives i'll install it and see if anything changes, could be a issue with the drivers or the motors... idk.

MitchBradley, 2025-04-23:
Please attach the gcode from Fusion360 and a list of the post settings.  Was it the Grbl post?  Also show the dimensioned sketch from which the pocket was extruded.

MitchBradley, 2025-04-23:
After the program is finished, does the machine return to the starting point, or to a point that is offset from the starting point by the cumulative drift?

Do the DROs in the sender report the expected ending location X0 Y0 Z29.75 afterwards?

bdring, 2025-04-23:
Try changing you `dir_delay_us` to 6 and the `disable_delay_us` to 6

```yaml
stepping:
  engine: I2S_STREAM
  idle_ms: 255
  pulse_us: 4
  dir_delay_us: 1
  disable_delay_us: 0
```

[image]

HerbertHenri, 2025-04-23:
> Was the profile imported into Aspire or created in Aspire? If imported, from where?
> 
> The gcode is quite odd. It is valid gcode, so FluidNC should be able to handle it. It is just another clue to help solve this. It may also be unrelated.
> 
> The problem could be the controller, the firmware or your drivers/wiring. We only have access to controllers and firmware. Thanks for being patient and trying our suggestions.

I used a dxf exported from fusion 360, so the profile was imported.

the fusion gcode was generated from the piece i had modeled.

[SLOT-360_1_T4.txt](https://github.com/user-attachments/files/19869065/SLOT-360_1_T4.txt)
(the file was acctualy .nc but github wont allow me to upload, so i saved as txt)

Also it's me who have to thank you guys for taking the time to help me! I'll be going down to the shop after lunch and i'll try everything and report here. I really enjoyed FluidNc interface and features i DO NOT want to go back to mach3

HerbertHenri, 2025-04-23:
> Please attach the gcode from Fusion360 and a list of the post settings. Was it the Grbl post? Also show the dimensioned sketch from which the pocket was extruded.

[image] 

that was the piece, and the slot i was testing (at first i tried cutting the whole piece, but it was drifing, so i just focused on the slot...)

[image]
Only setting i've changed is the ones that are in blue... everything was done in mm

HerbertHenri, 2025-04-23:
> Can you configure your post to not generate G2 arc moves, instead using line segments to approximate the arcs? I want to see if there is something going wrong with arc computation in FluidNC.

not really sure how to do that =/ 
I'll begin testing now, back in a few

HerbertHenri, 2025-04-23:
> If you start the program with the machine at X0, Y0, after the program is finished, does the machine return to the same X0 Y0 location, or to a point that is offset from the starting point by the cumulative drift?
> 
> Do the DROs in the sender report the expected ending location X0 Y0 Z29.75 afterwards?
> 
> Also please try the "disconnect Z" test that I mentioned earlier.

the fusion g-code, it just lifts up Z axis, where it stops, but if i send it back to X0 Y0, it has moved, thats how i measured, i set up a gauge at the 0.0 position, sent the code, made it go back, and in both x and y, the drift was about 2.5 -3mm, the z axis i noticed the drift because when i sent it back to 0Z, it went down but stoped around 8mm high from the acctual 0

bdring, 2025-04-23:
- Can you post the dxf file?
- What vectric post processor are you using?
- What is the motor current setting on the drivers? Do the drivers or motors get hot when running or in idle?

[33 more replies]

closed 2025-08-17 by bdring