bdring/FluidNC #1483: Problem: Real-time Status Report shows that spindle is still running after M5 command

hmantz, 2025-04-27:
### Wiki Search Terms

Status Report Spindle Speed M5

### Controller Board

PiBot FluidNC GRBL Laser CNC Controller V4.7 B but it can be reproduced in any generic ESP32 board

### Machine Description

I didn't connect it to my machine. I checked the PWM spindle output with my oscilloscope.

### Input Circuits

```yaml

```

### Configuration file

```yaml
board: PiBotV48A
name: Factory_Test
stepping:
  engine: I2S_STREAM
  idle_ms: 254
  pulse_us: 4
  dir_delay_us: 1
  disable_delay_us: 0
axes:
  shared_stepper_disable_pin: NO_PIN
  x:
    steps_per_mm: 800
    max_rate_mm_per_min: 5000
    acceleration_mm_per_sec2: 100
    max_travel_mm: 300
    soft_limits: false
    homing:
      cycle: 2
      positive_direction: false
      mpos_mm: 150
      feed_mm_per_min: 100
      seek_mm_per_min: 200
      settle_ms: 500
      seek_scaler: 1.1
      feed_scaler: 1.1
    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: gpio.2:pu:low
      hard_limits: false
      pulloff_mm: 1
      standard_stepper:
        step_pin: i2so.2
        direction_pin: i2so.1
        disable_pin: i2so.0
  y:
    steps_per_mm: 800
    max_rate_mm_per_min: 5000
    acceleration_mm_per_sec2: 100
    max_travel_mm: 300
    soft_limits: false
    homing:
      cycle: 2
      positive_direction: true
      mpos_mm: 150
      feed_mm_per_min: 100
      seek_mm_per_min: 200
      settle_ms: 500
      seek_scaler: 1.1
      feed_scaler: 1.1
    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: gpio.26:pu:low
      hard_limits: false
      pulloff_mm: 1
      standard_stepper:
        step_pin: i2so.5
        direction_pin: i2so.4
        disable_pin: i2so.7
  z:
    steps_per_mm: 800
    max_rate_mm_per_min: 5000
    acceleration_mm_per_sec2: 100
    max_travel_mm: 300
    soft_limits: false
    homing:
      cycle: 1
      positive_direction: true
      mpos_mm: 150
      feed_mm_per_min: 100
      seek_mm_per_min: 800
      settle_ms: 500
      seek_scaler: 1.1
      feed_scaler: 1.1
    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: gpio.33:pu:low
      hard_limits: false
      pulloff_mm: 1
      standard_stepper:
        step_pin: i2so.10
        direction_pin: i2so.9
        disable_pin: i2so.8
i2so:
  bck_pin: gpio.22
  data_pin: gpio.21
  ws_pin: gpio.17
spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18
probe:
  pin: gpio.32:pu:low
  toolsetter_pin: NO_PIN
start:
  must_home: false
pwm:
  pwm_hz: 50
  output_pin: gpio.16
  disable_with_s0: true
  s0_with_disable: true
  spinup_ms: 0
  spindown_ms: 0
  tool_num: 0
  speed_map: 0=5% 5000=10%
  off_on_alarm: true
uart1:
  txd_pin: gpio.27
  rxd_pin: gpio.25
  rts_pin: NO_PIN
  cts_pin: NO_PIN
  baud: 921600
  mode: 8N1
uart_channel1:
  report_interval_ms: 250
  uart_num: 1
control:
  estop_pin: gpio.35:low
  safety_door_pin: NO_PIN
  reset_pin: NO_PIN
  feed_hold_pin: NO_PIN
  cycle_start_pin: NO_PIN
  macro0_pin: NO_PIN
  macro1_pin: NO_PIN
  macro2_pin: NO_PIN
  macro3_pin: NO_PIN
  fault_pin: NO_PIN
```

### Startup Messages

```text
[MSG:INFO: FluidNC v3.9.6 https://github.com/bdring/FluidNC]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.7-dirty]
[MSG:INFO: Local filesystem type is spiffs]
[MSG:INFO: Configuration file:config1.yaml]
[MSG:INFO: Machine Factory_Test]
[MSG:INFO: Board PiBotV48A]
[MSG:INFO: UART1 Tx:gpio.27 Rx:gpio.25 RTS:NO_PIN Baud:921600]
[MSG:INFO: uart_channel1 created at report interval: 250]
[MSG:INFO: I2SO BCK:gpio.22 WS:gpio.17 DATA:gpio.21Min Pulse:2us]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
[MSG:INFO: Stepping:I2S_STREAM Pulse:4us Dsbl Delay:0us Dir Delay:1us Idle Delay:254ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Axis X (150.000,450.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:I2SO.2 Dir:I2SO.1 Disable:I2SO.0]
[MSG:INFO:  All Limit gpio.2:low:pu]
[MSG:INFO: Axis Y (-150.000,150.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:I2SO.5 Dir:I2SO.4 Disable:I2SO.7]
[MSG:INFO:  All Limit gpio.26:low:pu]
[MSG:INFO: Axis Z (-150.000,150.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:I2SO.10 Dir:I2SO.9 Disable:I2SO.8]
[MSG:INFO:  All Limit gpio.33:low:pu]
[MSG:INFO: estop_pin gpio.35:low]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: Connecting to STA SSID:##########]
[MSG:INFO: Connecting.]
[MSG:INFO: Connecting..]
[MSG:INFO: Connecting...]
[MSG:INFO: Connected - IP is ##########]
[MSG:INFO: WiFi on]
[MSG:INFO: Start mDNS with hostname:http://fluidnc.local/]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]
[MSG:INFO: PWM Spindle Ena:NO_PIN Out:gpio.16 Dir:NO_PIN Freq:50Hz Period:1048575]
[MSG:INFO: Probe gpio.32:low:pu]

Grbl 3.9 [FluidNC v3.9.6 (wifi) '$' for help]
ok
[VER:3.9 FluidNC v3.9.6:]
[OPT:PHS]
[MSG:Machine: Factory_Test]
[MSG:Mode=STA:SSID=HMsrv.24G:Status=Connected:##########:MAC=############]
ok
```

### User Interface Software

Universal Gcode Sender 20250108 but the issue also exists on the fluidnc.local web interface

### What happened?

I start the system from fresh. I issue S5000 command on the terminal. Real-time Status Report is:

<Idle|MPos:0.000,0.000,0.000|FS:0,0>

I then issue M3. Real-time Status Report changes to:

<Idle|MPos:0.000,0.000,0.000|FS:0,5000>

If I then issue M5 the motor will stop but the Real-time Status Report remains the same:

<Idle|MPos:0.000,0.000,0.000|FS:0,5000>

All the sequence follows:

<Idle|MPos:0.000,0.000,0.000|FS:0,0>
<Idle|MPos:0.000,0.000,0.000|FS:0,0|WCO:0.000,0.000,0.000>
>>> s5000
ok
<Idle|MPos:0.000,0.000,0.000|FS:0,0|Ov:100,100,100>
<Idle|MPos:0.000,0.000,0.000|FS:0,0>
<Idle|MPos:0.000,0.000,0.000|FS:0,0|WCO:0.000,0.000,0.000>
>>> m3
ok
<Idle|MPos:0.000,0.000,0.000|FS:0,5000>
<Idle|MPos:0.000,0.000,0.000|FS:0,5000|WCO:0.000,0.000,0.000>
<Idle|MPos:0.000,0.000,0.000|FS:0,5000|Ov:100,100,100|A:S>
<Idle|MPos:0.000,0.000,0.000|FS:0,5000>
>>> m5
ok
<Idle|MPos:0.000,0.000,0.000|FS:0,5000>
<Idle|MPos:0.000,0.000,0.000|FS:0,5000>
<Idle|MPos:0.000,0.000,0.000|FS:0,5000>

I went back all the way back to v3.1.3 and the issue is still there. Grbl_Esp32 does not have this issue, neither of course GRBL v1.1. They both change the Real-time Status Report to FS0,0 after M5.

### GCode File

S5000
M3
M5

### Other Information

This issue causes me a great  difficulty because I use the Real-time Status Report information to control a custom PID controller running on a separate ESP32-S3. At present I have to manually edit every .NC file and add S0 before the M5 command. I can modify Fusion 360 grbl post-processor to do this automatically but I prefer to solve the issue within FluidNC source code instead.

bdring, 2025-04-27:
Are you talking about the FS: vslue?

Feed and speed are both modal values. They stay at the last value set. The is per the gcode standard.

M3 M4 and M5 are also modal. See the $G report for that.

bdring, 2025-04-27:
Use auto reporting to get the $g report

hmantz, 2025-04-27:
> Are you talking about the FS: vslue?

Yes, exactly!

> Feed and speed are both modal values. They stay at the last value set. This is the gcode standard.

With all due respect, I think you may be mistaken. We are not talking about what the operator sets, but what FluidNC _reports back_. If you check the GRBL v1.1 Wiki Interface section

[https://github.com/gnea/grbl/wiki/Grbl-v1.1-Interface#real-time-status-reports](url)

it states clearly that Real-time Status Reports "Contains **real-time data** of Grbl’s state, position, and other data" and further down that "The second value _[of the FS: field]_ is the **current spindle speed** in RPM"

> M3 M4 and M5 are also modal. See the $G report for that.

$G also reports that the spindle is running when it is clearly stopped:

```
>>> s5000
ok
>>> m3
ok
>>> m5
ok
>>> $g
[GC:G0 G54 G17 G21 G90 G94 M5 M9 T0 F0 S5000]
ok
```

I am not saying that this is a bug as it seems to be intended behaviour, however, it clearly deviates from the GRBL v1.1 standard. It is interesting to note that all GRBL senders I tried state that the spindle is still running after M5.

Is it too difficult to change this? I am willing to help if you point me at the right direction. I had a quick look at the source files on Platformio and was overwhelmed... :)

hmantz, 2025-04-27:
P.S. I would assume that setting "s0_with_disable: true" in the PWM: section of the config file would do exactly what I want. However, it seems that this setting doesn't do anything.

bdring, 2025-04-27:
@hmantz is correct.

Feed is same. If motion ends or G0 is set. F still keeps it's last value

MitchBradley, 2025-04-27:
Possibly fixed by #1484.  Needs testing with a lot of different spindle types.  s0_with_disable must be true.

hmantz, 2025-04-27:
> Possibly fixed by [#1484](https://github.com/bdring/FluidNC/pull/1484). Needs testing with a lot of different spindle types. s0_with_disable must be true.

It works fine with my setup! :)

hmantz, 2025-05-07:
Fixed by [https://github.com/bdring/FluidNC/pull/1484](url)

closed 2025-05-07 by hmantz