bdring/FluidNC #1485: command (M47)

IgorUTch, 2025-04-28:
### Wiki Search Terms

G-code

### Controller Board

ESP32

### Machine Description

Cutting machine with automatic feeding of workpieces

### Input Circuits

```yaml

```

### Configuration file

```yaml
board: ESP32
name: FluidNC
stepping:
  engine: RMT
  idle_ms: 250
  pulse_us: 4
  dir_delay_us: 0
  disable_delay_us: 0
  segments: 12
spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18
sdcard:
  cs_pin: gpio.5
  card_detect_pin: NO_PIN
  frequency_hz: 8000000
kinematics: {}
axes:
  shared_stepper_disable_pin: gpio.13:pd
  shared_stepper_reset_pin: NO_PIN
  homing_runs: 2
  x:
    steps_per_mm: 80
    max_rate_mm_per_min: 5000
    acceleration_mm_per_sec2: 20
    max_travel_mm: 3000
    soft_limits: false
    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1
      standard_stepper:
        step_pin: gpio.12
        direction_pin: gpio.14
        disable_pin: NO_PIN
  y:
    steps_per_mm: 80
    max_rate_mm_per_min: 5000
    acceleration_mm_per_sec2: 20
    max_travel_mm: 3000
    soft_limits: false
    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1
      standard_stepper:
        step_pin: gpio.26
        direction_pin: gpio.15
        disable_pin: NO_PIN
coolant:
  flood_pin: NO_PIN
  mist_pin: NO_PIN
  delay_ms: 0
macros:   
start:
  must_home: false
  deactivate_parking: true
  check_limits: false
parking:
  enable: false
  axis: Z
  target_mpos_mm: -5
  rate_mm_per_min: 800
  pullout_distance_mm: 5
  pullout_rate_mm_per_min: 250
user_outputs:
  analog0_pin: NO_PIN
  analog1_pin: NO_PIN
  analog2_pin: NO_PIN
  analog3_pin: NO_PIN
  analog0_hz: 5000
  analog1_hz: 5000
  analog2_hz: 5000
  analog3_hz: 5000
  digital0_pin: NO_PIN
  digital1_pin: NO_PIN
  digital2_pin: NO_PIN
  digital3_pin: NO_PIN
  digital4_pin: NO_PIN
  digital5_pin: NO_PIN
  digital6_pin: NO_PIN
  digital7_pin: NO_PIN
user_inputs:
  analog0_pin: NO_PIN
  analog1_pin: NO_PIN
  analog2_pin: NO_PIN
  analog3_pin: NO_PIN
  digital0_pin: NO_PIN
  digital1_pin: NO_PIN
  digital2_pin: NO_PIN
  digital3_pin: NO_PIN
  digital4_pin: NO_PIN
  digital5_pin: NO_PIN
  digital6_pin: NO_PIN
  digital7_pin: NO_PIN
arc_tolerance_mm: 0.002
junction_deviation_mm: 0.01
verbose_errors: true
report_inches: false
enable_parking_override_control: false
use_line_numbers: false
planner_blocks: 16
PWM:
  output_pin: gpio.2
  enable_pin: gpio.22
  s0_with_disable: true
```

### Startup Messages

```text
It works great.
```

### What happened?

Good day. In my projects, I came across the need to use the G-code for cyclic program restart (M47). Is it possible to implement it?

MitchBradley, 2025-04-28:
You can do it this way http://wiki.fluidnc.com/en/features/gcode_parameters_expressions#looping .  It only works if you are running the program from an SD card.  If you are using a GCode sender, the sender would have to implement the repeat logic.

IgorUTch, 2025-04-28:
Thank you very much, I will try.

> 
>MitchBradley left a comment  (bdring/FluidNC#1485)
>You can do it this way  http://wiki.fluidnc.com/en/features/gcode_parameters_expressions#looping . It only works if you are running the program from an SD card. If you are using a GCode sender, the sender would have to implement the repeat logic.
 
 
Best regards
***@***.***

betuss2345, 2025-05-08:
Hello,

I have an MKS DLC32 MAX V1.0 board equipped with an ESP32-S3 module. I'm aiming to control my CNC machine, including the probe functionality, using either FluidNC or Grbl_ESP32 firmware.

However, it appears that ESP32-S3 is not currently supported by either firmware. The FluidNC developers suggested reaching out to MKS directly for support.

Therefore, I have two questions:

Is there any firmware available (even experimental) that supports the ESP32-S3 version of the MKS DLC32 MAX V1.0, including CNC and probing features?

If not, is it feasible to replace the ESP32-S3 module with a standard ESP32-WROOM module to make the board compatible with FluidNC or Grbl_ESP32?

Thank you very much for your assistance!

breiler, 2025-05-08:
@betuss2345 What does this question have to do with this issue? Why are you reposting the same question when you already got the answer?

IgorUTch, 2025-05-08:
I'm sorry, but I didn't do it.
--
Отправлено из  Mail для Android четверг,

closed 2025-05-08 by MitchBradley