bdring/FluidNC #1488: Problem: after_homing macro not executed after homing with external pin

ramack, 2025-05-02:
### Wiki Search Terms

homing_button_pin homing macro

### Controller Board

6x CNC Controller

### Machine Description

3 axis CNC using stepperonline NEMA 23 closed loop motors with CL57T-V41 drivers, with 1.5kW china spindle on a Siemens V20 VFD

### Input Circuits

```yaml
button connecting input pin to ground on push
```

### Configuration file

```yaml
board: 6 Pack CNC by Bart Dring
name: JaRa CNC
meta: 2025-05-02 RMk RS485 + FLuidDial

macros:
  after_homing: G53G00X165y520Z130

control:
# STOP Taste
  safety_door_pin: gpio.35:low
# Start Taste:
  cycle_start_pin: gpio.34:low
#Home Taste:
  homing_button_pin: gpio.39:low
# FU + Motortreiber WARN
  fault_pin: gpio.2:low:pu

  reset_pin: NO_PIN
#  fault_pin: gpio.36
#  feed_hold_pin: gpio.2:low:pu

status_outputs:
# LED Start
  hold_pin: I2SO.23
# LED Home:
  run_pin: I2SO.21

i2so:
  bck_pin: gpio.22
  data_pin: gpio.21
  ws_pin: gpio.17

spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

sdcard:
  card_detect_pin: NO_PIN
  cs_pin: gpio.5

start:
  must_home: true

parking:
  enable: true
  axis: Z
  pullout_distance_mm: 20.000
  pullout_rate_mm_per_min: 250.000
  target_mpos_mm: 120.000
  rate_mm_per_min: 800.000

# UART for FluidDial
uart2:
  txd_pin: gpio.27
  rxd_pin: gpio.25
  rts_pin: NO_PIN
  cts_pin: NO_PIN
  baud: 1000000
  mode: 8N1

uart_channel1:
  report_interval_ms: 200
  uart_num: 2
  message_level: debug

#UART for RS485
#uart1:
#  txd_pin: gpio.15
#  rxd_pin: gpio.16
#  rts_pin: gpio.14
#  cts_pin: NO_PIN
#  baud: 9600
#  mode: 8E1

#use correct UART!
#ModbusVFD:
# see https://support.industry.siemens.com/cs/mdm/109824500?c=160470548491&dl=nl&lc=en-US
#  debug: 3
#  uart_num: 1
#  modbus_id: 1
#  model: SiemensV20
#  ccw_cmd: 06 00 63 0c 7f > echo
#  cw_cmd: 06 00 63 04 7f > echo # maybe ccw and cw are flipped
#  off_cmd: 06 00 63 0c 7e > echo
#  set_rpm_cmd: 06 00 64 rpm  > echo
#  get_rpm_cmd: 03 00 6e 00 01 > 03 02 rpm

#  get_max_rpm_cmd: 03 00 0F 00 01 > 03 02 maxrpm*6/10 # this uses the base RPM and works
#  get_max_rpm_cmd: 03 10 82 00 01 > 03 02 rpm*60/100
#  get_min_rpm_cmd: 03 10 80 00 01 > 03 02 rpm*60/100
#  get_max_rpm_cmd: 03 01 69 00 01 > 03 02 maxrpm*6/10
#  get_min_rpm_cmd: 03 01 68 00 01 > 03 02 minrpm*6/10

#  speed_map: 0=0% 1=12.5% 3000=37.5% 9000=37.5% 24000=100%
#  off_on_alarm: true

SiemensV20:
  debug: 3
  uart:
    txd_pin: gpio.15
    rxd_pin: gpio.16
    rts_pin: gpio.14
    baud: 9600
    mode: 8E1
  modbus_id: 1
  tool_num: 0
  spinup_ms: 5000
  spindown_ms: 10000
  speed_map: 0=0% 1=12.5% 3000=37.5% 9000=37.5% 24000=100%
  off_on_alarm: true
#  poll_ms: 2000 # to be changed later, we use 2s right now only to reduce the debug output
#  disable_with_s0: true
#  s0_with_disable: true

#10V:
#  forward_pin: NO_PIN
#  reverse_pin: NO_PIN
#  pwm_hz: 15000
#  output_pin: gpio.13
#  enable_pin: gpio.15
#  direction_pin: gpio.14:low
#  disable_with_s0: false
#  s0_with_disable: true
#  spinup_ms: 1500
#  spindown_ms: 1500
#  tool_num: 0
#  speed_map: 0=0.000% 1000=0.000% 24000=100.000%
#  speed_map: 0=0% 1=12.5% 3000=37.5% 9000=37.5% 24000=100%
#  off_on_alarm: true

stepping:
  engine: I2S_STREAM
#we never disable the output pins automatically, so set to 255
  idle_ms: 255
# the CL57T can handle stepping frequencies up to 200 kHz, so in theory 2.5us pulses should be possible.
# other sources claim 500kHz, so we could go with 1us pulses
# 5 limits to 100kHz
  pulse_us: 6
  dir_delay_us: 6
  disable_delay_us: 0
  segments: 6

axes:
  shared_stepper_disable_pin: NO_PIN
  homing_runs: 2
  x:
    steps_per_mm: 1000.0
    max_rate_mm_per_min: 2000.000
    acceleration_mm_per_sec2: 50.000
    max_travel_mm: 410.000
    soft_limits: true
    homing:
      cycle: 2
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 50.000
      seek_mm_per_min: 500.000
      settle_ms: 250
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.26:low:pu
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1.000
      standard_stepper:
        step_pin: I2SO.2
        direction_pin: I2SO.1
        disable_pin: I2SO.0

  y:
    steps_per_mm: 1000.000
    max_rate_mm_per_min: 2200.000
    acceleration_mm_per_sec2: 50.000
    max_travel_mm: 555.000
    soft_limits: true
    homing:
      cycle: 2
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 50.000
      seek_mm_per_min: 500.000
      settle_ms: 250
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.33:low:pu
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1.000
      standard_stepper:
        step_pin: I2SO.5
        direction_pin: I2SO.4
        disable_pin: I2SO.7
    
  z:
    steps_per_mm: 1000.000
    max_rate_mm_per_min: 2000.000
    acceleration_mm_per_sec2: 25.000
    max_travel_mm: 380.000
    soft_limits: true
    homing:
      cycle: 1
      positive_direction: true
      mpos_mm: 140.000
      feed_mm_per_min: 50.000
      seek_mm_per_min: 400.000
      settle_ms: 250
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: gpio.32:low:pu
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1.000
      standard_stepper:
        step_pin: I2SO.10
        direction_pin: I2SO.9
        disable_pin: I2SO.8
```

### Startup Messages

```text
[MSG:INFO: uart_channel0 created]

[MSG:RST]
[MSG:INFO: FluidNC v3.9.6 https://github.com/bdring/FluidNC]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.7-dirty]
[MSG:INFO: Local filesystem type is littlefs]
[MSG:INFO: Configuration file:JaRa_CNC.yaml]
[MSG:DBG: Running after-parse tasks]
[MSG:DBG: Checking configuration]
[MSG:INFO: Machine JaRa CNC]
[MSG:INFO: Board 6 Pack CNC by Bart Dring]
[MSG:INFO: UART2 Tx:gpio.27 Rx:gpio.25 RTS:NO_PIN Baud:1000000]
[MSG:INFO: uart_channel2 created at report interval: 200]
[MSG:INFO: I2SO BCK:gpio.22 WS:gpio.17 DATA:gpio.21Min Pulse:2us]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
[MSG:INFO: SD Card cs_pin:gpio.5 detect:NO_PIN freq:8000000]
[MSG:INFO: Stepping:I2S_STREAM Pulse:6us Dsbl Delay:0us Dir Delay:6us Idle Delay:255ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Axis X (0.000,410.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:I2SO.2 Dir:I2SO.1 Disable:I2SO.0]
[MSG:INFO:  Neg Limit gpio.26:low:pu]
[MSG:INFO: Axis Y (0.000,555.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:I2SO.5 Dir:I2SO.4 Disable:I2SO.7]
[MSG:INFO:  Neg Limit gpio.33:low:pu]
[MSG:INFO: Axis Z (-240.000,140.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:I2SO.10 Dir:I2SO.9 Disable:I2SO.8]
[MSG:INFO:  Pos Limit gpio.32:low:pu]
[MSG:INFO: safety_door_pin gpio.35:low]
[MSG:INFO: cycle_start_pin gpio.34:low]
[MSG:INFO: fault_pin gpio.2:low:pu]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: Connecting to STA SSID:PalimPalim]
[MSG:INFO: Connecting.]
[MSG:INFO: Connecting..]
[MSG:INFO: Connected - IP is 192.168.0.77]
[MSG:INFO: WiFi on]
[MSG:INFO: Start mDNS with hostname:http://fluidnc.local/]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]
[MSG:INFO: Status outputs Interval:500 Idle:NO_PIN Cycle:I2SO.21 Hold:I2SO.23 Alarm:NO_PINDoor:NO_PIN]
[MSG:INFO: UART1 Tx:gpio.15 Rx:gpio.16 RTS:gpio.14 Baud:9600]
[MSG:INFO: SiemensV20 Spindle Tx:gpio.15 Rx:gpio.16 RTS:gpio.14 Baud:9600]
[MSG:INFO: RS485 Tx:  01 03 00 6E 00 01 E5 D7]
[MSG:DBG: SiemensV20: setState:5 SpindleSpeed:0]
[MSG:DBG: Synced speed. Requested:0 current:0]
[MSG:DBG: safety_door_pin 0]
[MSG:DBG: cycle_start_pin 0]
[MSG:DBG:  Neg Limit 0]
[MSG:DBG:  Pos Limit 0]
[MSG:DBG: SiemensV20: setState:5 SpindleSpeed:0]
[MSG:DBG: Synced speed. Requested:0 current:0]
```

### User Interface Software

WebUI, FluidDial

### What happened?

I configured the the homing_button_pin and it works fine to perform the homing. But the after_homing macro is not executed on button push. When using the FluidDial or the WebUI to perform homing the macro is executed after it's homed.

MitchBradley, 2025-05-02:
PR #1489 might fix it.  I don't have time to set up a test at the moment.

ramack, 2025-05-03:
works like a charm! Thanks a lot!

MitchBradley, 2025-05-03:
Did you check if it still works with $H ?

ramack, 2025-05-03:
Works also fine

ramack, 2026-03-22:
I stumbled over this ticket and think it is fixed in the meantime, so I close it.

closed 2026-03-22 by ramack