bdring/FluidNC #1497: Problem: Tool setter not working.

MadeInRu, 2025-05-12:
### Wiki Search Terms

I've searched online for possible solutions and tried a few. The biggest problem for me is that I don’t know whether there’s a hardware issue or not. If you guys can advise on a solution, it would be much appreciated! Thanks a lot in advance.

### Controller Board

[6-pack](https://us.ratrig.com/bigtreetech-rat-rig-rodent-cnc-controller-tmc2160.html)

### Machine Description

https://ratrig.com/cnc-kits/mills-coming-soon/rat-rig-mill.html

Beginner bundle

### Input Circuits

```yaml
https://us.ratrig.com/bigtreetech-rat-rig-rodent-cnc-controller-tmc2160.html
```

### Configuration file

```yaml
board: BTT Rodent-Z V1.0
name: Rat Rig Mill
meta: (19.12.2024) by MCruz Rat Rig PWM Spindle Config

kinematics:
  Cartesian:

# stepping:
#   engine: RMT
#   idle_ms: 250
#   dir_delay_us: 1
#   pulse_us: 2
#   disable_delay_us: 0
stepping:
  engine: I2S_STREAM
  idle_ms: 255
  pulse_us: 4
  dir_delay_us: 1
  disable_delay_us: 0

axes:
  shared_stepper_disable_pin: NO_PIN
  x:
    steps_per_mm: 400.000
    max_rate_mm_per_min: 3000.000
    acceleration_mm_per_sec2: 300.000
    max_travel_mm: 340.000
    soft_limits: True
    homing:
      cycle: 1
      positive_direction: false
      mpos_mm: 0
      feed_mm_per_min: 400.000
      seek_mm_per_min: 800.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.500

    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: gpio.35
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 3.000
      tmc_5160:
        step_pin: I2SO.2
        direction_pin: I2SO.1
[... 257 lines cut ...]
[MSG:INFO: Captive Portal Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]
[MSG:INFO: Status outputs Interval:500 Idle:I2SO.11:low Cycle:I2SO.14:low Hold:NO_PIN Alarm:I2SO.3:lowDoor:NO_PIN]
[MSG:INFO: PWM Spindle Ena:NO_PIN Out:gpio.13 Dir:NO_PIN Freq:5000Hz Period:8191]
[MSG:INFO: Toolsetter gpio.37]
```

### What happened?

Good time of the day. I’m having trouble connecting the tool setter to my RatRig Mill. I’m using the Rodent board version 1.1. I’ve tried all possible wiring combinations with the four wires, but none seem to work.

I mistakenly posted a photo of an error message saying the text couldn’t be found, but I can now confirm that message is unrelated. There’s no reaction from the tool setter in debug mode when I press it. Meanwhile, the endstops for X, Y, and Z respond immediately.

I’ve checked the tool setter with a multimeter and identified the correct wire pairs. In my configuration, I'm only using the pin for the tool setter because the 3D probe isn’t working (or isn’t working properly), so I can’t confirm its behavior.

I’m new to this, so I’m not entirely sure how to check whether the Rodent board might be faulty. However, all endstops and motors function correctly. I’ve also noticed that if I set the tool setter pin to low, the "P" icon in the dashboard turns dark blue — as if it’s permanently activated.

### Other Information

[image]
[image]
[image]
[image]
[image]

bdring, 2025-05-12:
Are the pins you are using on the toosetter N.C or N.O. 

Do you have a link to the toolssetter where you bought it?

MadeInRu, 2025-05-12:
> Are the pins you are using on the toosetter N.C or N.O.
> 
> Do you have a link to the toolssetter where you bought it?

Hi!

It's this one and from what i see it's N.C.

https://www.aliexpress.com/item/1005002315538312.html

bdring, 2025-05-12:
Change low to high on the pin

MadeInRu, 2025-05-12:
This is all i get and i've tried all possible combinations. Could my board be faulty? But if so then the rest of it shouldn't be working either. Is there a way to test pin37 and that port in general?

[image]

MadeInRu, 2025-05-12:
Is it possible to test it if i set Z limit to pin 37 and see if the limit switch will work? So at least i would know if the whole port with pin 37 is working? Cause i have no idea.

MadeInRu, 2025-05-12:
I think there is something wrong with the board. I switched the Z limit to pin 37 and the tool setter to pin 33. The Z limit switch appears to be "working" — the light is on — but it doesn’t activate. However, the tool setter suddenly came to life! I can now activate it easily. I think the board might be defective.

MitchBradley, 2025-05-12:
I found the problem.  BTT changed the pin assignment for that signal between revision 1.0 (GPIO37) and 1.1(GPIO39). Look at pages 6 and 7 of the user manual at https://github.com/bigtreetech/Rodent/blob/master/BIGTREETECH%20Rodent%20V1.x%20User%20Manual.pdf

If you change to `toolsetter_pin: gpio.39` it will probably work.

xtrasalty, 2025-05-13:
I am in the same boat as you, v1.1 and my pin 37 is displaying the same behavior as yours.   Back of my board for reference.  

[image]

MitchBradley, 2025-05-13:
To use the Rodent Probe input with a probe that pulls the signal pin to ground, it is necessary to install both the NPN jumper and also one of the adjacent jumpers - +5V, +12V, or +VB, depending on the characteristics of the probe.  Most typically you would use +5V.

I did not find information on a "Carbide3D Bitsetter v2", but there is a "Carbide3D BitZero v2".  Its manual goes into great detail about how to plug it into Carbide3D boards, but I could not find any description of how it actually works at the level necessary to be sure how to connect it to a Rodent board.

In the future, please proofread you posts for errors like the device name and the jumper name (PNP vs NPN).  Wondering what you actually meant wastes our time.

xtrasalty, 2025-05-13:
Sorry to waste your time Mitch, I removed that for you.

bdring, 2025-05-15:
@MadeInRu Is this issue resolved?

MadeInRu, 2025-05-15:
> [@MadeInRu](https://github.com/MadeInRu) Is this issue resolved?

Hi yes thanks a lot! It is pin 39 and not 37.

closed 2025-05-15 by bdring