bdring/FluidNC #1498: Problem: gpio.02 on 6x NC not working with either homing_button_pin or as limit

joel-oo, 2025-05-13:
### Wiki Search Terms

homing_button_pin
gpio.2

### Controller Board

6x NC v 3.9.7

### Machine Description

Gantry router with external DM542T stepper drivers, dual Y motors, NC limit switches on all axis. Spindle is a Bosch router controlled via solid state relay wired directly to io.4. Cycle start & feed hold are presently NO momentary contact switches. 

[Router.numbers-Electrical.pdf](https://github.com/user-attachments/files/20188005/Router.numbers-Electrical.pdf)

[image]

### Input Circuits

```yaml

```

### Configuration file

```yaml
board: "FluidNC 6x CNC Controller"
name: "Router"
meta: "2025.05.13 05:35"

arc_tolerance_mm: 0.002
junction_deviation_mm: 0.01
report_inches: false
enable_parking_override_control: false
use_line_numbers: false
planner_blocks: 16
# not default
verbose_errors: true

start:
  must_home: true
  deactivate_parking: true
# not default
  check_limits: true

i2so:
  bck_pin: gpio.22
  data_pin: gpio.21
  ws_pin: gpio.17

stepping:
  engine: I2S_STREAM 
  idle_ms: 255
  pulse_us: 4
  dir_delay_us: 0
  disable_delay_us: 0
  segments: 12
# not default

axes:
  shared_stepper_disable_pin: NO_PIN
  shared_stepper_reset_pin: NO_PIN
  homing_runs: 2
# not default

  x:
    steps_per_mm: 80.0
    max_rate_mm_per_min: 2400.0
    acceleration_mm_per_sec2: 50.0
    max_travel_mm: 1275.0
    soft_limits: true
    homing:
      cycle: 2
      allow_single_axis: false
      positive_direction: false
      mpos_mm: 0.0
      seek_mm_per_min: 600
      feed_mm_per_min: 50
      settle_ms: 250
      seek_scaler: 1.1
      feed_scaler: 1.1
    # motor block 2
    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: gpio.34
      hard_limits: true
      pulloff_mm: 5.0
      standard_stepper:
        step_pin: I2SO.5
        direction_pin: I2SO.4:low
        disable_pin: I2SO.7

  y:
    steps_per_mm: 80.0
    max_rate_mm_per_min: 2400.0
    acceleration_mm_per_sec2: 50.0
    max_travel_mm: 1275.0
    soft_limits: true
    homing:
      cycle: 2
      allow_single_axis: false
      positive_direction: false
      mpos_mm: 0.0
      seek_mm_per_min: 600
      feed_mm_per_min: 50
      settle_ms: 250
      seek_scaler: 1.1
      feed_scaler: 1.1
    # motor block 3 - left side (X home)
    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: gpio.39
      hard_limits: true
      pulloff_mm: 15.8
      standard_stepper:
        step_pin: I2SO.10
        direction_pin: I2SO.9:low
        disable_pin: I2SO.8
    # motor block 4 - right side (X plus)
    motor1:
      limit_neg_pin: gpio.36
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: true
      pulloff_mm: 10.0
      standard_stepper:
        step_pin: I2SO.13
        direction_pin: I2SO.12:low
        disable_pin: I2SO.15

  z:
    # steps/mm is 80% of X & Y
    steps_per_mm: 100.0
    max_rate_mm_per_min: 1920.0
    acceleration_mm_per_sec2: 25.0
    max_travel_mm: 115.0
    soft_limits: true
    homing:
      cycle: 1
      allow_single_axis: false
      positive_direction: true
      mpos_mm: 0.0
      seek_mm_per_min: 480
      feed_mm_per_min: 40
      settle_ms: 250
      seek_scaler: 1.1
      feed_scaler: 1.1
    # motor block 1
    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      #35
      limit_all_pin: gpio.35
      hard_limits: true
      pulloff_mm: 5.0
      standard_stepper:
        step_pin: I2SO.2
        direction_pin: I2SO.1:low
        disable_pin: I2SO.0

control:
  safety_door_pin: NO_PIN
  reset_pin: NO_PIN
  estop_pin: NO_PIN
  # not default
  #** when wired w/ new sw remove ':low' 2 places below
  feed_hold_pin: gpio.33:low
  cycle_start_pin: gpio.32:low
  #**
  fault_pin: gpio.26
  homing_button_pin: gpio.02

relay:
  direction_pin: NO_PIN
  output_pin: gpio.4
  enable_pin: NO_PIN
  disable_with_s0: false
  s0_with_disable: true
  spinup_ms: 0
  spindown_ms: 0
  tool_num: 0
  speed_map: 0=0.0% 0=100.0% 1=100.0%
  off_on_alarm: true

spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18
sdcard:
  cs_pin: gpio.5
  card_detect_pin: NO_PIN
  frequency_hz: 8000000
```

### Startup Messages

```text
Grbl 3.9 [FluidNC v3.9.7 (bt) '$' for help]
[MSG:INFO: uart_channel0 created]

[MSG:RST]
[MSG:INFO: FluidNC v3.9.7 https://github.com/bdring/FluidNC]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.7-dirty]
[MSG:INFO: Local filesystem type is littlefs]
[MSG:INFO: Configuration file:router.yaml]
[MSG:INFO: Machine Router]
[MSG:INFO: Board FluidNC 6x CNC Controller]
[MSG:INFO: I2SO BCK:gpio.22 WS:gpio.17 DATA:gpio.21Min Pulse:2us]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
[MSG:INFO: SD Card cs_pin:gpio.5 detect:NO_PIN freq:8000000]
[MSG:INFO: Stepping:I2S_STREAM Pulse:4us Dsbl Delay:0us Dir Delay:0us Idle Delay:255ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Axis X (0.000,1275.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:I2SO.5 Dir:I2SO.4:low Disable:I2SO.7]
[MSG:INFO:  All Limit gpio.34]
[MSG:INFO: Axis Y (0.000,1275.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:I2SO.10 Dir:I2SO.9:low Disable:I2SO.8]
[MSG:INFO:  All Limit gpio.39]
[MSG:INFO:   Motor1]
[MSG:INFO:     standard_stepper Step:I2SO.13 Dir:I2SO.12:low Disable:I2SO.15]
[MSG:INFO:  Neg Limit gpio.36]
[MSG:INFO: Axis Z (-115.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:I2SO.2 Dir:I2SO.1:low Disable:I2SO.0]
[MSG:INFO:  All Limit gpio.35]
[MSG:INFO: feed_hold_pin gpio.33:low]
[MSG:INFO: cycle_start_pin gpio.32:low]
[MSG:INFO: fault_pin gpio.26]
[MSG:INFO: homing_button_pin gpio.2]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: BT Started with FluidNC]
[MSG:INFO: Relay Spindle Ena:NO_PIN Out:gpio.4 Dir:NO_PIN]

Grbl 3.9 [FluidNC v3.9.7 (bt) '$' for help]
[MSG:INFO: ALARM: Unhomed]
ALARM:14

Grbl 3.9 [FluidNC v3.9.7 (bt) '$' for help]
```

### User Interface Software

Fluid Web Installer / Chrome / MacBook Pro / USB

### What happened?

When a NC switch is connected to gipo.02 the input seems to be ignored. This happens when gpio.02 is assigned to homing_button_pin as well as limit_all_pin on the Z axis. The home_button_pin function does work when assigned to a different gpio (in this case gpio.35). Voltages at the terminals for io.2 & associated ground are 0v when the NC switch is not pressed & 4.3v when pressed. When gpio.02 is assigned to limit_all_pin for the Z axis the calibrate page shows a grey "?" for the input.

bdring, 2025-05-13:
Remove one of the switch wires from the controller side. Then check the status.

bdring, 2025-05-13:
I am sorry, but I forgot that gpio.2 needs the `pu:` attribute. It is a strapping pin and would mess with the bootloader if there was an external pullup resistor.

That is mentioned on the wiki page. I added a little more to that page and added a schematic snippet.

joel-oo, 2025-05-13:
Adding :high:pu  did it (:pu alone does not). 

I had read about io2 & :pu so I chose to avoid the issue by not using that pin which worked up until the homing feature was added, I had run out of other pins, and forgotten about the requirement for :pu.

THANKS for the help. 

Where's the appropriate forum to raise an issue with documentation or functionality?
Where either I don't understand something or the functionality doesn't seem right or the documentation seems off. Nothing that rises to the level of not working but might be worth looking onto at some point.

Happy to open it here, discord, or keep it to myself - just don't want to  create extra work for you.

bdring, 2025-05-13:
That's great to hear.

You can post it the help forum on Discord. You can also follow these instructions to edit the wiki yourself.

http://wiki.fluidnc.com/en/wiki_contributions

closed 2025-05-14 by MitchBradley