bdring/FluidNC #1499: Feature: Skipped step detection with TMC drivers

eduwes, 2025-05-15:
### Machine Context

Jackpot controller. TMC 2209's for now, but those can be changed if needed.

### Feature Description

Please add the ability to detect skipped steps and then automatically stop the currently running job. This should hopefully be possible with TMC drivers. Without closed loop you won't know how many steps or be able to correct it but you should still be able to detect that there were skipped steps and abort the job. This would be a tagged on/off feature of course.

Thanks

### Other Approaches

Closed loop steppers

### How I Can Help

Testing

MitchBradley, 2025-05-15:
Connect the TMC diag output to a GPIO and assign it as fault_pin.  Configure for stallguard mode and set the threshold pretty high.

bdring, 2025-05-15:
Read this wiki page.

http://wiki.fluidnc.com/en/support/stallguard_tuning

closed 2025-05-15 by bdring