bdring/FluidNC #1501: Problem: Safety Door Parking activated before homing

ramack, 2025-05-15:
### Wiki Search Terms

door homing

### Controller Board

6x CNC Controller

### Machine Description

4 axis CNC using stepperonline NEMA 23 closed loop motors with CL57T-V41 drivers, with 1.5kW china spindle on a Siemens V20 VFD

### Input Circuits

```yaml
homing limit switches on all axis normally open
safety door pin on a "stop switch", normally open
resume button on cycle_start_pin, normally open

homing working fine, safety door action working in normal operation as expected.
```

### Configuration file

```yaml
board: 6 Pack CNC by Bart Dring
name: JaRa CNC
meta: 2025-05-15 RMk RS485 + FLuidDial

macros:
#  after_homing: G53G00X520Y-165Z130

control:
# STOP Taste
  safety_door_pin: gpio.35:low
# Start Taste:
  cycle_start_pin: gpio.34:low
#Home Taste:
  homing_button_pin: gpio.39:low
# FU + Motortreiber WARN
  fault_pin: gpio.2:low:pu

  reset_pin: NO_PIN
#  fault_pin: gpio.36
#  feed_hold_pin: gpio.2:low:pu

status_outputs:
# LED Start
  hold_pin: I2SO.23
# LED Home:
  run_pin: I2SO.21

i2so:
  bck_pin: gpio.22
  data_pin: gpio.21
  ws_pin: gpio.17

spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

sdcard:
  card_detect_pin: NO_PIN
  cs_pin: gpio.5

start:
  must_home: true

parking:
[... 252 lines cut ...]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]
[MSG:INFO: Status outputs Interval:500 Idle:NO_PIN Cycle:I2SO.21 Hold:I2SO.23 Alarm:NO_PINDoor:NO_PIN]
[MSG:INFO: UART1 Tx:gpio.15 Rx:gpio.16 RTS:gpio.14 Baud:9600]
[MSG:INFO: SiemensV20 Spindle Tx:gpio.15 Rx:gpio.16 RTS:gpio.14 Baud:9600]
[MSG:INFO: RS485 Tx:  01 03 00 6E 00 01 E5 D7]
[MSG:DBG: SiemensV20: setState:5 SpindleSpeed:0]
[MSG:DBG: Synced speed. Requested:0 current:0]
ok
<Alarm|MPos:0.000,0.000,0.000,0.000|FS:0,0>
```

### User Interface Software

WebUI, FluidDial pendant

### What happened?

When activating the button on the safety door pin (opening the DOOR) before the machine is homed FluidNC tries to move to parking position and during the pullout (or later) it crashed our z-axis and even ignored the limit switch.

Before homing the machine should not move due to door opening, and definitely consider the limit switch to stop there.

### Other Information

maybe also for macros should be checked if the limit switches are considered

bdring, 2025-05-17:
You will have to show an exact log of what you do to cause the problem.

I did some testing. This is what happens if I activate the door before homing. It stays in alarm mode. The door pin is reporting as active. No motion occurs.

```
Grbl 3.9 [FluidNC v3.9.7 (wifi) '$' for help]
[MSG:INFO: ALARM: Unhomed]
ALARM:14
[MSG:INFO: Check door]
<Alarm|MPos:0.000,0.000,0.000|FS:0,0|Pn:D|WCO:0.000,0.000,-50.000>
```

Here is the door active on startup. No motion occurs.
```
Grbl 3.9 [FluidNC v3.9.7 (wifi) '$' for help]
[MSG:ERR: safety_door_pin is active at startup]
[MSG:INFO: ALARM: Unhomed]
ALARM:14
[MSG:INFO: ALARM: Control Pin Initially On]
ALARM:11
<Alarm|MPos:0.000,0.000,0.000|FS:0,0|Pn:D|WCO:0.000,0.000,-50.000>
```

If I clear the homing alarm with $X. The door/parking feature can be activated, but since the Z machine position is is likely wrong, problems could occur. This is true for most of the features in FluidNC.

If I enable hard limits on Z the parking will trigger an alarm if it hits a switch.

```
$axes/z/motor0/hard_limits=true
ok
[MSG:INFO: Check door]
[MSG:INFO: ALARM: Hard Limit]
ALARM:1
[MSG:ERR: Reset to continue]
```

ramack, 2025-05-21:
For reproducing we tried three cases:
A) power on, fluidnc in unhomes Alarm, open door
B) power on, fluidnc in unhomes Alarm, clear alarm in WebUI, Idle state, open door
C) change config to have `hard_limits: true` power on, fluidnc in unhomes Alarm, clear alarm in WebUI, Idle state, open door, manually activate the limit switch

with A all is fine, no movement
with B the "unexpected" overrun of the limit switch occurs
with C also all fine, movement starts and stops at hard_limit

So basically it was a configuration issue on our side with the wrong expectation that a configured limit switch would also be a safety limit. Would be cool if you could document this a bit more explicitly - searching in the wiki for "hard_limits" did not even bring up a helpful result.

For the sake of completeness here is the log for B is:
```[MSG:INFO: FluidNC v3.9.6 (Test_1489-9f00d07f-dirty) git@github.com:ramack/FluidNC.git]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.7-dirty]
[MSG:INFO: Local filesystem type is littlefs]
[MSG:INFO: Configuration file:JaRa_CNC.yaml]
[MSG:DBG: Running after-parse tasks]
[MSG:DBG: Checking configuration]
[MSG:INFO: Machine JaRa CNC]
[MSG:INFO: Board 6 Pack CNC by Bart Dring]
[MSG:INFO: UART2 Tx:gpio.27 Rx:gpio.25 RTS:NO_PIN Baud:1000000]
[MSG:INFO: uart_channel2 created at report interval: 200]
[MSG:INFO: I2SO BCK:gpio.22 WS:gpio.17 DATA:gpio.21Min Pulse:2us]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
[MSG:INFO: SD Card cs_pin:gpio.5 detect:NO_PIN freq:8000000]
[MSG:INFO: Stepping:I2S_STREAM Pulse:6us Dsbl Delay:0us Dir Delay:6us Idle Delay:255ms]
[MSG:INFO: Axis count 4]
[MSG:INFO: Axis X (0.000,555.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:I2SO.2 Dir:I2SO.1 Disable:I2SO.0]
[MSG:INFO:  Neg Limit gpio.33:low:pu]
[MSG:INFO: Axis Y (-410.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:I2SO.5 Dir:I2SO.4 Disable:I2SO.7]
[MSG:INFO:  Neg Limit gpio.26:low:pu]
[MSG:INFO: Axis Z (-240.000,140.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:I2SO.10 Dir:I2SO.9 Disable:I2SO.8]
[MSG:INFO:  Pos Limit gpio.32:low:pu]
[MSG:INFO: Axis A (-960.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:I2SO.13 Dir:I2SO.12 Disable:I2SO.15]
[MSG:INFO: safety_door_pin gpio.35:low]
[MSG:INFO: cycle_start_pin gpio.34:low]
[MSG:INFO: fault_pin gpio.2:low:pu]
[MSG:INFO: homing_button_pin gpio.39:low]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: Connecting to STA SSID:PalimPalim]
[MSG:INFO: Connecting.]
[MSG:INFO: Connecting..]
[MSG:INFO: Connected - IP is 192.168.0.77]
[MSG:INFO: WiFi on]
[MSG:INFO: Start mDNS with hostname:http://fluidnc.local/]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]
[MSG:INFO: Status outputs Interval:500 Idle:NO_PIN Cycle:I2SO.21 Hold:I2SO.23 Alarm:NO_PINDoor:NO_PIN]
[MSG:INFO: UART1 Tx:gpio.15 Rx:gpio.16 RTS:gpio.14 Baud:9600]
[MSG:INFO: SiemensV20 Spindle Tx:gpio.15 Rx:gpio.16 RTS:gpio.14 Baud:9600]
[MSG:INFO: RS485 Tx:  01 03 00 6E 00 01 E5 D7]
[MSG:DBG: SiemensV20: setState:5 SpindleSpeed:0]
[MSG:DBG: Synced speed. Requested:0 current:0]
ok
<Alarm|MPos:0.000,0.000,0.000,0.000|FS:0,0|Ov:100,100,100>
$X
[MSG:INFO: Caution: Unlocked]
ok
[MSG:DBG: safety_door_pin 1]
[MSG:INFO: Check door]
[MSG:DBG: Parking pullout]
[MSG:DBG:  Pos Limit 1]
[MSG:DBG: Limit switch tripped for Z motor 0]
[MSG:DBG:  Pos Limit 0]
[MSG:DBG:  Pos Limit 1]
[MSG:DBG: Limit switch tripped for Z motor 0]
[MSG:DBG:  Pos Limit 0]
[MSG:DBG:  Pos Limit 1]
```

MitchBradley, 2025-05-21:
Feel free to edit the wiki to say whatever you think needs to be said.

ramack, 2025-05-21:
I surely would be willing, but as wiki contributions require a discord account I am out. That's why I have chosen to be even nobler and proposed to you that you could improve the documentation - without complaining about the link (https://github.com/bdring/FluidNC/wiki/Hardware-that-Runs-FluidNC) in the issue template that points to the outdated wiki 😉.

Don't get me wrong, I really appreciate the efforts you put into fluidnc and I'm giving my best to help where I can, but right now I don't feel motivated to contribute. "It doesn't work! Help-Me!"-users are a pain, but you should also try to avoid treating everyone as troll.

closed 2025-05-21 by ramack

bdring, 2025-05-21:
I added some info to the parking page.