bdring/FluidNC #1502: Problem: How Setup Servo Motor  - StepperOnline Model A6-RS750H2A1-M17

GermainTejos, 2025-05-17:
### Wiki Search Terms

Hi, recently I bought a 6x CNC Controller from Barton Dring. I have experience using stepper motor, but now I tried to conect Servo motor and I can´t got run it with 6x CNC Controller fluidnc.

I hope anyone can help me

Link of Servomotor: https://www.omc-stepperonline.com/index.php?route=product/product/get_file&file=5100/A6-RS%20series%20servo%20drive%20manual%20(1).pdf

[image]

### Controller Board

6x CNC Controller fluidnc from Barton Dring

### Machine Description

Router CNC

### Input Circuits

```yaml

```

### Configuration file

```yaml
-
```

### Startup Messages

```text
-
```

### User Interface Software

WebUi

### What happened?

Hi, recently I bought a 6x CNC Controller from Barton Dring. I have experience using stepper motor, but now I tried to conect Servo motor and I can´t got run it with 6x CNC Controller fluidnc.

I hope anyone can help me

MitchBradley, 2025-05-17:
We can't help you if you choose not to supply all the information that we ask for in the issue template.

GermainTejos, 2025-05-17:
@MitchBradley sorry , I'm new on this forum.
I usually conect Stepper motors to 6X CNC Controller by @bdring using this schematics

Stepper driver              6X CNC Controller
PUL +                    ---> STP
DIR  +                    ---> DIR            
ENA +                    ---> ENA
PUL-/DIR-/ENA-    ---> GND

After using WebUi it works ok

but now, I can't get wiring properly servomotor to 6X CNC Controller, so I really hope if anyone can help me.

MitchBradley, 2025-05-18:
Connect STP to PULSE+, DIR to SIGN+.  PULSE- and SIGN- to GND.

GermainTejos, 2025-06-10:
@MitchBradley 
Mr could you help me, I need to setup A-axis as a Slave axis from Y-axis.

This is my configuration in YAML file

[image]

bdring, 2025-06-10:
Do you mean you want a dual motor Y axis?

GermainTejos, 2025-06-11:
@bdring your advice works! thanks a lot

Now I just need to invert direction of motor, how can I get that?

**this is my YAML file**

board: 6x
name: 6x GTC
stepping:
  engine: I2S_STREAM
  idle_ms: 254
  pulse_us: 4
  dir_delay_us: 1
  disable_delay_us: 0

axes:
  shared_stepper_disable_pin: NO_PIN
  x:
    steps_per_mm: 1600.000
    max_rate_mm_per_min: 5000.000
    acceleration_mm_per_sec2: 500.000
    max_travel_mm: 600.000
    soft_limits: false
    homing:
      cycle: 2
      positive_direction: true
      mpos_mm: 150.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 200.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.2:low:pu
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1.000
      standard_stepper:
        step_pin: I2SO.2
        direction_pin: I2SO.1
        disable_pin: I2SO.0

  y:
    steps_per_mm: 1600.000
    max_rate_mm_per_min: 5000.000
    acceleration_mm_per_sec2: 500.000
    max_travel_mm: 300.000
    soft_limits: false
    homing:
      cycle: 2
      positive_direction: true
      mpos_mm: 150.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 200.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.26:low:pu
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1.000
      standard_stepper:
        step_pin: I2SO.5
        direction_pin: I2SO.4
        disable_pin: I2SO.7

    motor1:
      limit_neg_pin: gpio.32:low
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1.000
      standard_stepper:
        step_pin: I2SO.13
        direction_pin: I2SO.12
        disable_pin: I2SO.15

  z:
    steps_per_mm: 1600.000
    max_rate_mm_per_min: 1000.000
    acceleration_mm_per_sec2: 200.000
    max_travel_mm: 300.000
    soft_limits: false
    homing:
      cycle: 1
      positive_direction: false
      mpos_mm: 150.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 800.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.33:low    
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1.000
      standard_stepper:
        step_pin: I2SO.10
        direction_pin: I2SO.9
        disable_pin: I2SO.8
        
  b:
    steps_per_mm: 808.000
    max_rate_mm_per_min: 5000.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 200.000
    soft_limits: false
    homing:
      cycle: 1
      positive_direction: false
      mpos_mm: 150.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 800.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.35:low
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 3.000
      standard_stepper:
        step_pin: I2SO.18
        direction_pin: I2SO.17
        disable_pin: I2SO.16
        
  c:      
    steps_per_mm: 808.000
    max_rate_mm_per_min: 5000.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 200.000
    soft_limits: false
    homing:
      cycle: 1
      positive_direction: false
      mpos_mm: 150.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 800.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.34:low
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 3.000
      standard_stepper:
        step_pin: I2SO.21
        direction_pin: I2SO.20
        disable_pin: I2SO.23
        
       
i2so:
  bck_pin: gpio.22
  data_pin: gpio.21
  ws_pin: gpio.17

spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

sdcard:
  card_detect_pin: NO_PIN
  cs_pin: gpio.5
  
probe:
  pin: gpio.39:low
  toolsetter_pin: gpio.36:low
  
# Using MOSFETs (Check Spindle Pin Usage
coolant:
  flood_pin: gpio.12
  mist_pin: gpio.4
  delay_ms: 0

start:
  must_home: false
  
## Begin Huanyang  
# uart1:
  # txd_pin: gpio.15
  # rxd_pin: gpio.16
  # rts_pin: gpio.14
  # baud: 9600
  # mode: 8N1

# Huanyang:
  # uart_num: 1
  # modbus_id: 1
  # tool_num: 0
  # speed_map: 0=0% 0=25% 6000=25% 24000=100%
  # off_on_alarm: false

# #begin PWM
pwm:
  pwm_hz: 5000
  direction_pin: NO_PIN
  output_pin: gpio.13
  enable_pin: gpio.14
  disable_with_s0: false
  s0_with_disable: true
  spinup_ms: 0
  spindown_ms: 0
  tool_num: 0
  speed_map: 0=0.000% 10000=100.000%
  off_on_alarm: false
  
# #begin Laser
# Laser:
  # pwm_hz: 5000
  # output_pin: gpio.4
  # enable_pin: gpio.12
  # disable_with_s0: false
  # s0_with_disable: true
  # tool_num: 1
  # speed_map: 0=0.000% 255=100.000%
  # off_on_alarm: true
  
# #begin 10V
# 10V:
  # forward_pin: gpio.15
  # reverse_pin: gpio.14
  # pwm_hz: 5000
  # output_pin: gpio.13
  # enable_pin: NO_PIN
  # direction_pin: NO_PIN
  # disable_with_s0: false
  # s0_with_disable: true
  # spinup_ms: 0
  # spindown_ms: 0  
  # tool_num: 0
  # speed_map: 0=0.000% 1000=0.000% 24000=100.000%
  # off_on_alarm: false

bdring, 2025-06-11:
http://wiki.fluidnc.com/en/support/faq#how-do-i-invert-a-pin-state