bdring/FluidNC #1503: Feature: toggle signal activating push button switch

Abdelrahman22a, 2025-05-19:
### Machine Context

I need a simulator macro that turns my prob on and off instead of me pressing a push button
with a certain delay (5-10 sec.) before any other Gcode movement
for waterjet prototype
it's going to be hooked on optocoupled relay
I need the config file and any .g files for ESP32
please help as soon as you can, it's for my graduation project 

### Feature Description

I need a simulator macro that turns my prob on and off instead of me pressing a push button
with a certain delay (5-10 sec.) before any other Gcode movement
like a 300 msec pulse

### Other Approaches

we can use any of the current Gcode like water mist or flood instead of Macros if we can achieve the pulse toggle effect and the delay afterwards 

### How I Can Help

I've made this one already
but there is error with the Macros and with [MSG:ERR: Configuration parse error on line 12: Invalid character]
so if anyone can help fix it or make it work, I'd realy appreciate it

here's  it 

name: Waterjet_ESP32_PulseRelay
board: esp32dev

stepping:
  engine: StepDir
  idle_ms: 250
  dir_delay_us: 1
  pulse_us: 5
  disable_delay_us: 0

axes:
  - axis: x
    steps_per_mm: 160
    max_rate_mm_per_min: 5000
    acceleration_mm_per_sec2: 200
    max_travel_mm: 500
    motor0:
      driver: StepDir
      step_pin: gpio22
      dir_pin: gpio21
      enable_pin: gpio23

  - axis: y
    steps_per_mm: 160
    max_rate_mm_per_min: 5000
    acceleration_mm_per_sec2: 200
    max_travel_mm: 500
    motor0:
      driver: StepDir
      step_pin: gpio19
      dir_pin: gpio18
      enable_pin: gpio17
    motor1:
      driver: StepDir
      step_pin: gpio16
      dir_pin: gpio15
      enable_pin: gpio14

endstops:
  - pin: gpio33
    axis: x
    hard_limit: true
    pullup: true
    invert: true

  - pin: gpio32
    axis: x
    hard_limit: true
    pullup: true
    invert: true

  - pin: gpio35
    axis: y
    hard_limit: true
    pullup: true
    invert: true

  - pin: gpio34
    axis: y
    hard_limit: true
    pullup: true
    invert: true

control:
  safety_door_pin: gpio36
  reset_pin: gpio39

output:
  - pin: gpio26
    start_hardware: true
    on_start: 1
    on_stop: 1
    off_timeout_sec: 5

macros:
  M100:
    actions:
      - output: gpio26, state=1  # Toggle START
      - dwell: 0.3  # Pulse duration 300ms
      - output: gpio26, state=0  # Toggle OFF
      - dwell: 6.0  # Delay 6 seconds before executing next commands

  M101:
    actions:
      - output: gpio26, state=1  # Toggle STOP
      - dwell: 0.3  # Pulse duration 300ms
      - output: gpio26, state=0  # Toggle OFF

  M200:
    actions:
      - output: gpio27, state=1  # Toggle Emergency Stop ON
      - dwell: 0.3  # Pulse duration 300ms
      - output: gpio27, state=0  # Toggle OFF

  M201:
    actions:
      - output: gpio27, state=1  # Toggle Resume ON
      - dwell: 0.3  # Pulse duration 300ms
      - output: gpio27, state=0  # Toggle OFF
      - dwell: 5.0  # Delay 5 seconds before executing next commands

start:
  must_home: false

bdring, 2025-05-19:
It would be helpful if you described what you are trying to do and why.

I don't know what those "macros" are. We do not support M100, M101, M200, M201.

Our macros are written in gcode and gcode expressions. Toggling a pin is done with M62 and M63.

Could what you are trying to do be done in a CAM postprocessor? What CAM are you using?

Abdelrahman22a, 2025-05-20:
I'm not into the CAM  phase yet
I'm still trying to figure the custom prop I have
it's water jet... it takes a 300ms toggle to start and and another one to stop by a Normal Open relay
and when it starts... it takes around 5~10 seconds to build the pressure before the pressure is suitable to perform the Gcode cutting 

so I need to assign a macros function that makes a 300msec toggle function with a 10 seconds afterwards 
and another one that makes the same thing without the 10 seconds delay 
now I'm using the config file 3axis_v3.yaml and it's working fine
if you can help on how to edit it or assign the spool to it or what ever 
and thank you very much for your assistance

bdring, 2025-05-20:
You can read all about macros here.

http://wiki.fluidnc.com/en/config/macros

here is an example

```gcode
M62 P0 ; turn on digital output 0
G4 P0.3 ; wait 300ms
M63 P0 ; turn off digital output 0
M62 P1 ; turn on digital output 1
G4 P10 ; wait 10s
```

You could put the macro in the config file, or on an SD card. 

Most people would do this via a CAM preprocessor. This would add the gcodes into all the the file that are produced. This removes the manual operations.

Abdelrahman22a, 2025-05-20:
And how can I assign the output pin for these macros?

Abdelrahman22a, 2025-05-20:
I've made these macros and I'll add them to the config file
with custome (.g) files Start_Relay.g  Stop_Relay.g Emergency_Stop.g Emergency_Resume.g
so kindly please correct me if you notice anything wrong

user_outputs:
  digital0_pin: gpio14  # Start relay output mapped to GPIO14
  digital2_pin: gpio15  # Emergency stop/resume output mapped to GPIO15

macros:
  Start_Relay:
    actions:
      - gcode: "M64 P0"  # Turn on digital output 0 (GPIO14)
      - dwell: 0.3  # Wait 300ms
      - gcode: "M65 P0"  # Turn off digital output 0 (GPIO14)
      - dwell: 10.0  # Delay 10 seconds before executing further commands

  Stop_Relay:
    actions:
      - gcode: "M64 P0"  # Turn on digital output 0 (GPIO14)
      - dwell: 0.3  # Wait 300ms
      - gcode: "M65 P0"  # Turn off digital output 0 (GPIO14)

  Emergency_Stop:
    actions:
      - gcode: "M64 P2"  # Activate emergency stop on GPIO15
      - dwell: 0.3  # Wait 300ms
      - gcode: "M65 P2"  # Deactivate emergency stop

  Emergency_Resume:
    actions:
      - gcode: "M64 P2"  # Activate resume signal on GPIO15
      - dwell: 0.3  # Wait 300ms
      - gcode: "M65 P2"  # Deactivate resume signal
      - dwell: 5.0  # Delay 5 seconds before executing next commands

Abdelrahman22a, 2025-05-21:
can you tell me please how can I define it and call it as a function?
as I'll be having couple of other similar function?

M62 P0 ; turn on digital output 0
G4 P0.3 ; wait 300ms
M63 P0 ; turn off digital output 0
M62 P1 ; turn on digital output 1
G4 P10 ; wait 10s

bdring, 2025-05-21:
You can store gcode files on an SD card or the local file system (localfs). You can run them by sending $SD/run=<myfile>.

Please search the wiki for "macro"

Abdelrahman22a, 2025-05-21:
Please, check what I'm writing, you might get an idea of what I mean

[image]

[image]

bdring, 2025-05-21:
Did an AI suggest this? It is not even close to the FluidNC file format.

Abdelrahman22a, 2025-05-21:
Yes, I'm working with AI to write it as I don't have enough knowledge 
I really appreciate the help

bdring, 2025-05-21:
The AI is clueless. What you posted is gibberish. Not even close. Read the wiki. 

Learn.  

If you have questions, please point the the exact wiki section that you don't understand.

Abdelrahman22a, 2025-05-21:
I'm working on learning how to write it correctly
yet I don't understand how to make a function that executes a certain sequence on Gcode or macrose
each block here is a function that I want to call with a macro button or a certain i/p signal
and I don't seem to find any help on the function's definition in the wiki
I really appreciate your patience with me
[image]

Abdelrahman22a, 2025-05-21:
I've solved it, thank you so much
The only issue I'm facing now is hard resetting
I need to configure it
and none of these are working

[image]

[image]

bdring, 2025-05-21:
Search the wiki for "ignored key"

Abdelrahman22a, 2025-05-22:
I've read it and the code itself doesn't have any of the reasons to ignore these keys

I've managed to make the macros work

to make the picture clear for you
I need to assign a pin on the ESP32 in a way that when it receives a certain input (either high or low) it stops everything immediately 
just as the reset button on the interface itself do stop everything and activates the alarm.... but  from a hardware signal and with an extra macro line if possible 

[image]

bdring, 2025-05-22:
Post your current config file.

Abdelrahman22a, 2025-05-22:
I need macro2 to be executed upon receiving the hardware signal
and macro3 to be executed hardware signal when that signal is removed

name: "ESP32 Dev Controller VROOM "
board: "ESP32 Dev Controller VROOM"

stepping:
  engine: RMT
  idle_ms: 250
  dir_delay_us: 1
  pulse_us: 2
  disable_delay_us: 0

axes:
  shared_stepper_disable_pin: NO_PIN
  
  y:
    steps_per_mm: 200
    max_rate_mm_per_min: 1000
    acceleration_mm_per_sec2: 100
    max_travel_mm: 1000
    homing:
      cycle: 2
      mpos_mm: 10
      positive_direction: false
    
    motor0:
      limit_all_pin: NO_PIN
      stepstick:
        direction_pin: gpio.27
        step_pin: gpio.13
    motor1:
      null_motor:

  x:
    steps_per_mm: 200
    max_rate_mm_per_min: 1000
    acceleration_mm_per_sec2: 100
    max_travel_mm: 1000
    homing:
      cycle: 2
      mpos_mm: 10
      positive_direction: false

    motor0:
      limit_all_pin: NO_PIN
      stepstick:
        direction_pin: gpio.26
        step_pin: gpio.12
    motor1:
      null_motor:

  z:
    steps_per_mm: 800
    max_rate_mm_per_min: 2000
    acceleration_mm_per_sec2: 25
    max_travel_mm: 1000
    homing:
      cycle: 1
      mpos_mm: 10
      positive_direction: true

    motor0:
      limit_all_pin: NO_PIN
      stepstick:
        direction_pin: NO_PIN
        step_pin: gpio.2
    motor1:
      null_motor:

spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18
sdcard:
  cs_pin: gpio.5
  card_detect_pin: NO_PIN

control:
  safety_door_pin: NO_PIN
  reset_pin: NO_PIN
  feed_hold_pin: NO_PIN
  cycle_start_pin: NO_PIN
  macro0_pin: NO_PIN
  macro1_pin: NO_PIN
  macro2_pin: NO_PIN
  macro3_pin: NO_PIN
  fault_pin: NO_PIN
  estop_pin: gpio.36:low
  homing_button_pin: NO_PIN

coolant:
  flood_pin: gpio.25
  mist_pin:  NO_PIN

        
probe:
  pin: gpio.32:low:pu

PWM:
  pwm_hz: 5000
  output_pin: gpio.16
  enable_pin: gpio.22
  direction_pin: NO_PIN
  disable_with_s0: false
  s0_with_disable: true
  spinup_ms: 0
  spindown_ms: 0
  tool_num: 0
  speed_map: 0=0% 10000=100%
  
  

  
user_outputs:
  digital0_pin: gpio.17  
  digital1_pin: gpio.14
  
user_inputs:
  digital2_pin: gpio.39:low

macros:
  macro0: M64 P0 G4 P0.3 M65 P0 G4 10.0  

  macro1: M64 P0 G4 P0.3 M65 P0    
  
  
  macro3: M64 P0 G4 P0.3 M65 P0     M64 P2

  macro2: M64 P0 G4 P0.3 M65 P0 G4 P5.0    M64 P2

breiler, 2025-05-22:
Can you upload it as a file instead (drag and drop it to github), the indentation is wrong in what you posted.

breiler, 2025-05-22:
Try renaming it to .txt

bdring, 2025-05-22:
I edited it to add a code section

Abdelrahman22a, 2025-05-22:
[config.txt](https://github.com/user-attachments/files/20396666/config.txt)

bdring, 2025-05-22:
Send a photo of your electronics

MitchBradley, 2025-05-22:
Two problems:
1. You need to separate the individual commands inside the macros with &, like this:
```
macros:
   macro0: M64 P0 & G4 P0.3 & M65 P0 & G4 10.0  
```
2. You have not assigned any pins to invoke the macros
```
control:
  ...
  macro0_pin: NO_PIN
```
NO_PIN needs to be replaced with some input pin like `gpio.SOME_PIN_NUMBER.low`

bdring, 2025-05-22:
Use a switch with both N.C. and N.O. contacts and connect to 2 input pins.. Currently FluidNC only supports one active state or action per input.

Good luck with your project. At this point I have given as much free support to a special project. I am going to step away from this issue.

closed 2025-05-24 by bdring