bdring/FluidNC #1507: Feature: poll sdcard cardDetect pin and take appropriate action when inserted or removed.

ellensp, 2025-06-03:
### Machine Context

Any machine that has a cardDetect pin, so it actually useful.

Currently the cardDetect pin  is passed in sdmmc_slot_config_t, but this is only used if you're using the SDMMC driver through the esp_vfs_fat_sdmmc_mount() helper, which optionally uses the CD pin to detect card presence at mount time only. 

It doesn't poll the pin. 

using $GD I was able to verify the IO for cardDetect is the IO pin I configured and is changing in response to the sdcard being in or out. But the firmware does nothing with it 

### Feature Description

On inserting a sdcard (ie cardDetect pin goes low)  the sdcard should be initialized and the web interface notified to refresh the sdcard window and status.
On removing the sdcard (ie cardDetect pin goes high) the sdcard should be set as non initialized and the web interface notified to clear  the sdcard window and update the status.

### Other Approaches

Tried telekinesis, the API didn't work.  (joke)  
No other approaches that I can think of.

### How I Can Help

Asking for  useful  features is helping to improve  FluidNC

MitchBradley, 2025-06-03:
"The web interface" is actually two versions, only one of which the FluidNC developers control.  "Any machine that has a card detect pin" excludes all of the controller boards from FluidNC developers.  To my knowledge, no controller board with a card detect comes from a vendor that supports the FluidNC project in any way, so supporting the feature would only increase our non-compensated costs in supporting products that are "free riding" on our development work and ongoing help for users.

There are many things to consider, including when it is safe to perform such a check.  Doing it in the midst of sender activity could disrupt the running of a GCode program.  There is a probably a way to do it safely, but finding all of the edge cases could take a long time, especially in light of the small percentage of boards with the capability.

My personal take is that the value of the feature across the user base is not worth the costs.  If someone else is willing to undertake the project and see it through to completion, and offer ongoing support for it, I would be happy to consider PRs to implement it.

MitchBradley, 2025-06-03:
There are other states to consider, such as "card inserted but cannot be read correctly".  That adds additional user interface complexity.