bdring/FluidNC #1508: Problem: Motor is not moving

Vikramgc, 2025-06-03:
### Wiki Search Terms

I'm new to the FluidNC 6X controller and currently trying to get a single NEMA 23 stepper motor running using a DM542T driver. I've gone through the wiki documentation and searched through many forum threads, but I haven’t found a case that matches my issue.

Here’s what’s happening:

I’m trying to control the motor through the web interface.

When I issue a movement command on the X-axis, both the STEP and DIRECTION LEDs blink 

Despite the blinking LEDs, the motor doesn’t move at all—no sound, no vibration.

I’ve double-checked my wiring and pin configuration, but I might be missing something. Has anyone experienced a similar issue or have suggestions on how to troubleshoot this?

**I also check the motor enable and disable. The motor is locked while sending $ME command And can be rotated by hand while sending $MD command.**
Also , I can feel motor being heated up.

### Controller Board

6x CNC controller

### Machine Description

Nema 23 stepper motor with DM542T driver , and 24V power supply

### Input Circuits

```yaml

```

### Configuration file

```yaml
board: 6x CNC Controller
name: 6x Default
stepping:
  engine: I2S_STREAM
  idle_ms: 255
  pulse_us: 4
  dir_delay_us: 1
  disable_delay_us: 0

axes:
  shared_stepper_disable_pin: NO_PIN
  x:
    steps_per_mm: 800.000
    max_rate_mm_per_min: 5000.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 300.000
    soft_limits: false
    homing:
      cycle: 2
      positive_direction: false
      mpos_mm: 150.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 200.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.2:low:pu
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1.000
      standard_stepper:
        step_pin: I2SO.2
        direction_pin: I2SO.1
        disable_pin: I2SO.0

  y:
    steps_per_mm: 800.000
    max_rate_mm_per_min: 5000.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 300.000
    soft_limits: false
    homing:
      cycle: 2
      positive_direction: true
      mpos_mm: 150.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 200.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.26:low:pu
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1.000
      standard_stepper:
        step_pin: I2SO.5
        direction_pin: I2SO.4
        disable_pin: I2SO.7

  z:
    steps_per_mm: 800.000
    max_rate_mm_per_min: 5000.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 300.000
    soft_limits: false
    homing:
      cycle: 1
      positive_direction: true
      mpos_mm: 150.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 800.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.33:low    
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1.000
      standard_stepper:
        step_pin: I2SO.10
        direction_pin: I2SO.9
        disable_pin: I2SO.8
        
  a:
    steps_per_mm: 53.400
    max_rate_mm_per_min: 5000.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 960.000
    soft_limits: false
    homing:
      cycle: 2
      positive_direction: false
      mpos_mm: 150.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 200.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.32:low
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 3.000
      standard_stepper:
        step_pin: I2SO.13
        direction_pin: I2SO.12
        disable_pin: I2SO.15

  b:
    steps_per_mm: 808.000
    max_rate_mm_per_min: 5000.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 200.000
    soft_limits: false
    homing:
      cycle: 1
      positive_direction: false
      mpos_mm: 150.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 800.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.35:low
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 3.000
      standard_stepper:
        step_pin: I2SO.18
        direction_pin: I2SO.17
        disable_pin: I2SO.16

  c:      
    steps_per_mm: 808.000
    max_rate_mm_per_min: 5000.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 200.000
    soft_limits: false
    homing:
      cycle: 1
      positive_direction: false
      mpos_mm: 150.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 800.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.34:low
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 3.000
      standard_stepper:
        step_pin: I2SO.21
        direction_pin: I2SO.20
        disable_pin: I2SO.23

i2so:
  bck_pin: gpio.22
  data_pin: gpio.21
  ws_pin: gpio.17

spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

sdcard:
  card_detect_pin: NO_PIN
  cs_pin: gpio.5
  
probe:
  pin: gpio.39:low
  toolsetter_pin: gpio.36:low
  
# Using MOSFETs (Check Spindle Pin Usage
coolant:
  flood_pin: gpio.12
  mist_pin: gpio.4
  delay_ms: 0

start:
  must_home: false
  
## Begin Huanyang  
# uart1:
  # txd_pin: gpio.15
  # rxd_pin: gpio.16
  # rts_pin: gpio.14
  # baud: 9600
  # mode: 8N1

# Huanyang:
  # uart_num: 1
  # modbus_id: 1
  # tool_num: 0
  # speed_map: 0=0% 0=25% 6000=25% 24000=100%
  # off_on_alarm: false

# #begin PWM
pwm:
  pwm_hz: 5000
  direction_pin: gpio.15
  output_pin: gpio.13
  enable_pin: gpio.14
  disable_with_s0: false
  s0_with_disable: true
  spinup_ms: 0
  spindown_ms: 0
  tool_num: 0
  speed_map: 0=0.000% 10000=100.000%
  off_on_alarm: false
  
# #begin Laser
# Laser:
  # pwm_hz: 5000
  # output_pin: gpio.4
  # enable_pin: gpio.12
  # disable_with_s0: false
  # s0_with_disable: true
  # tool_num: 1
  # speed_map: 0=0.000% 255=100.000%
  # off_on_alarm: true
  
# #begin 10V
# 10V:
  # forward_pin: gpio.15
  # reverse_pin: gpio.14
  # pwm_hz: 5000
  # output_pin: gpio.13
  # enable_pin: NO_PIN
  # direction_pin: NO_PIN
  # disable_with_s0: false
  # s0_with_disable: true
  # spinup_ms: 0
  # spindown_ms: 0  
  # tool_num: 0
  # speed_map: 0=0.000% 1000=0.000% 24000=100.000%
  # off_on_alarm: false
```

### Startup Messages

```text
$G
$SS
[MSG:INFO: FluidNC v3.9.7 https://github.com/bdring/FluidNC]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.7-dirty]
[MSG:INFO: Local filesystem type is littlefs]
[MSG:INFO: Configuration file:config.yaml]
[MSG:INFO: Machine 6x Default]
[MSG:INFO: Board 6x CNC Controller]
[MSG:INFO: I2SO BCK:gpio.22 WS:gpio.17 DATA:gpio.21Min Pulse:2us]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
[MSG:INFO: SD Card cs_pin:gpio.5 detect:NO_PIN freq:8000000]
[MSG:INFO: Stepping:I2S_STREAM Pulse:4us Dsbl Delay:0us Dir Delay:1us Idle Delay:255ms]
[MSG:INFO: Axis count 6]
[MSG:INFO: Axis X (150.000,450.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:I2SO.2 Dir:I2SO.1 Disable:I2SO.0]
[MSG:INFO:  Neg Limit gpio.2:low:pu]
[MSG:INFO: Axis Y (-150.000,150.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:I2SO.5 Dir:I2SO.4 Disable:I2SO.7]
[MSG:INFO:  Neg Limit gpio.26:low:pu]
[MSG:INFO: Axis Z (-150.000,150.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:I2SO.10 Dir:I2SO.9 Disable:I2SO.8]
[MSG:INFO:  Neg Limit gpio.33:low]
[MSG:INFO: Axis A (150.000,1110.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:I2SO.13 Dir:I2SO.12 Disable:I2SO.15]
[MSG:INFO:  Neg Limit gpio.32:low]
[MSG:INFO: Axis B (150.000,350.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:I2SO.18 Dir:I2SO.17 Disable:I2SO.16]
[MSG:INFO:  Neg Limit gpio.35:low]
[MSG:INFO: Axis C (150.000,350.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:I2SO.21 Dir:I2SO.20 Disable:I2SO.23]
[MSG:INFO:  Neg Limit gpio.34:low]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: STA SSID is not set]
[MSG:INFO: AP SSID FluidNC IP 192.168.0.1 mask 255.255.255.0 channel 1]
[MSG:INFO: AP started]
[MSG:INFO: WiFi on]
[MSG:INFO: Captive Portal Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]
[MSG:INFO: PWM Spindle Ena:gpio.14 Out:gpio.13 Dir:gpio.15 Freq:5000Hz Period:8191]
[MSG:INFO: Flood coolant gpio.12]
[MSG:INFO: Mist coolant gpio.4]
[MSG:INFO: Probe gpio.39:low]
[MSG:INFO: Toolsetter gpio.36:low]
ok
<Idle|MPos:690.000,0.000,0.000,0.000,0.000,0.000|FS:0,0|WCO:208.000,0.000,0.000,0.000,0.000,0.000>
```

### User Interface Software

WebUI

### What happened?

Once the WebUI page loaded , I tried to Jog by pressing the X+(10) button and expected the motor to move but its not moving at all.

bdring, 2025-06-03:
It appears everything is OK. Try sending a few commands from the console. You can use the console in the WebUI, FluidTerm or the Web Installer terminal

Send `$report/interval=200`. This will turn on auto reporting.

Send `$J=G91 G21 F1000 X10`. This should jog the the X axis 10mm in the positive direction. Then send `$J=G91 G21 F1000 X-10` to go the other way

You should see the step LED blink dimly and the motor should move 10mm. If is does not move at least see if the motor is locked.
When you change the direction, the direction LED should change on to off or the other way around.

You should also see the position change on the WebUI and when you send ?

Report what happens. If you see any alarms or warnings, let us know.

MitchBradley, 2025-06-03:
What is the setting for the DM542T's control voltage switch ?

[image]

Vikramgc, 2025-06-03:
Hello everyone, The issue is solved. My driver was damaged . I replaced the driver and its working fine.