bdring/FluidNC #1518: Problem: 6x cnc controller motor 5 and 6 do not work

LukeAV1212, 2025-06-29:
### Wiki Search Terms

n/a

### Controller Board

6x cnc controller

### Machine Description

MPCNC

### Input Circuits

```yaml
n/a
```

### Configuration file

```yaml
board: 6x CNC Controller
name: 6x Default
stepping:
  engine: I2S_STREAM
  idle_ms: 255
  pulse_us: 4
  dir_delay_us: 1
  disable_delay_us: 0

axes:
  shared_stepper_disable_pin: NO_PIN
  x:
    steps_per_mm: 100.000
    max_rate_mm_per_min: 9000.000
    acceleration_mm_per_sec2: 200.000
    max_travel_mm: 700.000
    soft_limits: false
    homing:
      cycle: 2
      positive_direction: false
      mpos_mm: 3.000
      feed_mm_per_min: 300.000
      seek_mm_per_min: 1500.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100
    #motor1
    motor0:
      limit_neg_pin: gpio.2:low:pu
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 3.000
      standard_stepper:
        step_pin: I2SO.2
        direction_pin: I2SO.1
        disable_pin: I2SO.0
    #motor2
    motor1:
      limit_neg_pin: gpio.26:low:pu
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 3.000
      standard_stepper:
        step_pin: I2SO.5
        direction_pin: I2SO.4
        disable_pin: I2SO.7

  y:
    steps_per_mm: 100.000
    max_rate_mm_per_min: 9000.000
    acceleration_mm_per_sec2: 200.000
    max_travel_mm: 300.000
    soft_limits: false
    homing:
      cycle: 2
      positive_direction: false
      mpos_mm: 3.000
      feed_mm_per_min: 300.000
      seek_mm_per_min: 1500.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100
    #motor3
    motor0:
      limit_neg_pin: gpio.33:low
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 3.000
      standard_stepper:
        step_pin: I2SO.10
        direction_pin: I2SO.9
        disable_pin: I2SO.8
    #motor4
    motor1:
      limit_neg_pin: gpio.32:low
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 3.000
      standard_stepper:
        step_pin: I2SO.13
        direction_pin: I2SO.12
        disable_pin: I2SO.15
z:
    steps_per_mm: 400.000
    max_rate_mm_per_min: 5000.000
    acceleration_mm_per_sec2: 200.000
    max_travel_mm: 150.000
    soft_limits: false
    homing:
      cycle: 1
      positive_direction: false
      mpos_mm: 3.000
      feed_mm_per_min: 300.000
      seek_mm_per_min: 1500.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100
    #motor5
    motor0:
      limit_neg_pin: gpio.35:low    
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 3.000
      standard_stepper:
        step_pin: I2SO.21
        direction_pin: I2SO.20
        disable_pin: I2SO.23
        
i2so:
  bck_pin: gpio.22
  data_pin: gpio.21
  ws_pin: gpio.17

spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

sdcard:
  card_detect_pin: NO_PIN
  cs_pin: gpio.5
  
#probe:
  #pin: gpio.39:low
  #toolsetter_pin: gpio.36:low
  
# Using MOSFETs (Check Spindle Pin Usage
#coolant:
  #flood_pin: gpio.12
  #mist_pin: gpio.4
  #delay_ms: 0

start:
  must_home: false
  
## Begin Huanyang  
uart1:
  txd_pin: gpio.15
  rxd_pin: gpio.16
  rts_pin: gpio.14
  baud: 9600
  mode: 8N1

Huanyang:
  uart_num: 1
  modbus_id: 1
  tool_num: 0
  speed_map: 0=0% 0=25% 6000=25% 24000=100%
  off_on_alarm: false

  uart2:
  txd_pin: gpio.25
  rxd_pin: gpio.27
  rts_pin: NO_PIN
  cts_pin: NO_PIN
  baud: 1000000
  mode: 8N1

uart_channel2:
  report_interval_ms: 75
  uart_num: 2
```

### Startup Messages

```text
n/a
```

### What happened?

I have the 6x cnc controller and I am trying to get it to work with an mpcnc. I have x and y running but I can't get the z axis to do anything on motor 5.
-I thought it might be my firmware but I went through it over and over but it is correct. 
-I connected one of the x axis motors to the z driver and the driver works.
-I made sure the cable connected to the z axis motor is working by using it to drive an x axis motor.
-I moved the z motor from motor 5 to motor 6 and still does not work. 
-I connected the x axis driver to the z motor and motor works.
-When I move the x and y axis, the bord lights up on the step and directions LED's but nothing happens on motor 5 and 6 when moving the       axis. 
All movements were sent through from the webui. 
Has anyone had this problem before? is it user error or faulty PCB?

bdring, 2025-06-29:
Please add the start messages

LukeAV1212, 2025-06-29:
I will have to add them tomorrow, it is late here in perth.

MitchBradley, 2025-06-29:
Also please try sending the movement commands with FluidTerm instead of WebUI, to reduce the number of variables.  You can alternate between  "G1 Z50 F500" and "G1 Z0" to move back and forth.

LukeAV1212, 2025-06-30:
Ok, so, I missed an indent and now its running. The problem is that when you look at something long enough you miss the small (big) things. Thanks for your time anyway.

bdring, 2025-06-30:
The start messages are a beautiful thing.