bdring/FluidNC #1520: Problem: IntegerDivideByZero with firmware 3.9.4 and newer when using I2S_STREAM, cannot move motors

yllarp, 2025-07-01:
### Wiki Search Terms

IntegerDivideByZero

### Controller Board

6x v1.2

### Machine Description

Double Y XYZ with DM856 drivers 
[image]
Driver connections (controller was off when I took this picture, normally the green LED are on)
[image]

### Input Circuits

```yaml

```

### Configuration file

```yaml
board: 6x
name: 6x min
stepping:
  engine: I2S_STREAM
  idle_ms: 254
  pulse_us: 10
```

### Startup Messages

```text
[MSG:INFO: uart_channel0 created]

[MSG:RST]
[MSG:INFO: FluidNC v3.9.7 https://github.com/bdring/FluidNC]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.7-dirty]
[MSG:INFO: Local filesystem type is littlefs]
[MSG:INFO: Configuration file:minimal.yaml]
[MSG:INFO: Axes: using defaults]
[MSG:INFO: Machine 6x min]
[MSG:INFO: Board 6x]
[MSG:INFO: Stepping:I2S_STREAM Pulse:10us Dsbl Delay:0us Dir Delay:0us Idle Delay:254ms]
Guru Meditation Error: Core  1 panic'ed (IntegerDivideByZero). Exception was unhandled.

Core  1 register dump:
PC      : 0x400d671c  PS      : 0x00060c30  A0      : 0x8010b782  A1      : 0x3ffb21d0  
A2      : 0xffffffff  A3      : 0x00000009  A4      : 0x01312d00  A5      : 0x00000000  
A6      : 0x3ffcca40  A7      : 0x00000000  A8      : 0x00000000  A9      : 0x00000014  
A10     : 0x00000000  A11     : 0x00000408  A12     : 0x3ffb2c88  A13     : 0x00000017  
A14     : 0x00000000  A15     : 0xffffffff  SAR     : 0x0000001a  EXCCAUSE: 0x00000006  
EXCVADDR: 0x00000000  LBEG    : 0x4008bdbc  LEND    : 0x4008bdc7  LCOUNT  : 0xffffffff  

Backtrace: 0x400d6719:0x3ffb21d0 0x4010b77f:0x3ffb2200 0x400efa4b:0x3ffb2240 0x4012c922:0x3ffb2290

ELF file SHA256: 24503d07bbb57de5

Rebooting...
[MSG:INFO: uart_channel0 created]

[MSG:RST]
[MSG:INFO: FluidNC v3.9.7 https://github.com/bdring/FluidNC]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.7-dirty]
[MSG:INFO: Local filesystem type is littlefs]
[MSG:ERR: Skipping configuration file due to panic]
[MSG:INFO: Using default configuration]
[MSG:INFO: Axes: using defaults]
[MSG:INFO: Machine Default (Test Drive)]
[MSG:INFO: Board None]
[MSG:INFO: Stepping:RMT Pulse:4us Dsbl Delay:0us Dir Delay:0us Idle Delay:255ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Axis X (-1000.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO: Axis Y (-1000.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO: Axis Z (-1000.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: Connecting to STA SSID:MYap]
[MSG:INFO: Connecting.]
[MSG:INFO: Connecting..]
[MSG:INFO: Connected - IP is 192.168.107.19]
[MSG:INFO: WiFi on]
[MSG:INFO: Start mDNS with hostname:http://fluidnc.local/]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]

Grbl 3.9 [FluidNC v3.9.7 (wifi) '$' for help]
[MSG:ERR: Configuration is invalid. Check boot messages for ERR's.]
ok
[VER:3.9 FluidNC v3.9.7:]
[OPT:PHS]
[MSG:Machine: Default (Test Drive)]
[MSG:Mode=STA:SSID=NSABuss1:Status=Connected:IP=192.168.107.19:MAC=48-E7-29-18-E4-70]
ok
```

### What happened?

Hello,
Wanted to upgrade from grbl-Mega based board to 6x but cannot get it to work.

I've encountered 2 issues:
1. Cannot use firmware newer than 3.9.3 without the IntegerDivideByZero error when my configuration file uses I2S_STREAM engine. Even the minimal example I posted errors out.
2. With firmware 3.9.3 I'm able to load the fluidnc.yaml [fluidnc.txt](https://github.com/user-attachments/files/21006264/fluidnc.txt) (renamed to txt so I could add it here) but I cannot move any motors. They are energised when I disconnect the Ena pins, but I cannot hear the loud "thump" as with old board and when I submit move command via WebUI the status gets "stuck" in JOG but motors/drivers are silent  and not moving. The $ME/$MD commands doesn't have any effect when Ena pins are connected.

As the drivers worked with my previous board I assume the standard_stepper should work and no special treatment is needed..

I can upgrade/downgrade the firmware, I can see files on SD-card and the gpio.14 LED toggles when I click flood, the pendant is connected so everything but the motors seem to work.

Do I understand correct, that when pulling the stepper pins low, the LED should go out? Mine seem to be always on, even if they are not defined in the config (except the very last one, that sometimes is off as seen on the picture above). 

Sorry for putting 2 issues into one ticket, but I have a feeling they might be related and my board could be faulty as I couldn't find anything related to the IntegerDivideByZero error and latest "working" release was 6 months ago so I imagine there would be at least some other tickets if it's common issue.

Happy to provide any additional info I missed.

Thanks,
Üllar

MitchBradley, 2025-07-01:
If what you posted is all of your config file, the /0 problem is likely to be caused by a missing i2s: section.  Try using a complete config file.

If you posted just an excerpt of your config file, please post the entire file so we can reproduce your problem.

yllarp, 2025-07-01:
Thank you, Mitch, for the prompt answer! You were right. Added 
```spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18
```
to the config and both my issues were resolved. Upgraded to latest FW and jogging runs beautifully!

Thanks again,
Üllar

closed 2025-07-01 by yllarp

MitchBradley, 2025-07-01:
I am working on improvements to the config error checking, which should prevent the /0 crash.  It will take some time before that is deployed because there are so very many cases to consider.