bdring/FluidNC #1521: Problem: 'VFD Queue Full' Message.

Monberg75, 2025-07-04:
### Wiki Search Terms

I get this message once in a while, resulting in the VFD/Spindle not spinning up. But it doesnt stop the process, resulting in a crash.

Why is there no alarm set when this message is shown?

Im using UGS for my controller.

### Controller Board

Xprov5 Controller

### Machine Description

Bulkman3D Ultimate Bee 1000x1000 with high current steppers, Xprov5 controller and VFD controlled spindle

### Input Circuits

```yaml

```

### Configuration file

```yaml
Not needed for now I think
```

### Startup Messages

```text
Not needed for now I think
```

### User Interface Software

UGS

### What happened?

Crassh after starting job due to missing spindle rotation. UGS reports VFD queue full message in console. This has happened before, but not very often, and it's been a long time since last. Main problem is that the process doesnt stop, but crashes the machine.

bdring, 2025-07-04:
We need all the information you left out.

Monberg75, 2025-07-04:
I will find it when I get home from work.. So Config file and Startup message? Gcode file isn't relevant, as it has nothing to do with the file.. It can be any file.

bdring, 2025-07-04:
Add a picture of the wiring at both ends too.

Monberg75, 2025-07-04:
What is startup messages?

Monberg75, 2025-07-04:
Here is the missing documentation.

Let me state, that I know that the link is missing the resistors to terminate the connection, but that is not the problem im trying to address.

The problem is that Fluidnc doesn't set an alarm when this message is coming, so that the process continues. Nothing else. I will get the termination done..

[config.txt](https://github.com/user-attachments/files/21061368/config.txt)

[Startup.txt](https://github.com/user-attachments/files/21061369/Startup.txt)

[image]

[image]

MitchBradley, 2025-07-04:
> What is startup messages?

[image]

MitchBradley, 2025-07-04:
We ask for complete information because problems are nearly always context-dependent.  Seeing all the information lets us form a mental picture of the full situation.  We need to see the forest and the trees.

Monberg75, 2025-07-04:
> We ask for complete information because problems are nearly always context-dependent. Seeing all the information lets us form a mental picture of the full situation. We need to see the forest and the trees.

Yes I fully understand.. Just needed to understand what you needed :)

bdring, 2025-07-04:
Try using the terminal in the web installer.

https://installer.fluidnc.com/

The start messages should look like this.

http://wiki.fluidnc.com/en/support/troubleshooting_config_files#check-the-startup-messages

MitchBradley, 2025-07-04:
We also need to know exactly what you mean by "crashes the machine".  It could mean that the bit slammed into the bed, or that things just stopped working with no messages, or that the firmware issued a bunch of software crash messages, or ...  Please be very specific and detailed about what you saw at the time of the failure.

Monberg75, 2025-07-04:
The machined tried to mill out wood with a stopped spindle.
It didnt turn on the spindle as it should. Thats basically it

This is the message in got from Joacim at ugs:

Without knowing the details about what messages you are getting from the
controller, I think you should raise this with the FluidNC developers.
It sounds serious if the controller looses the ability to control the VFD,
so the appropriate message from the controller should be an alarm and not
just a "MSG: I think we just lost control of the VFD".

Med venlig hilsen / Best regards:

*Kim Monberg*

Denmark

MitchBradley, 2025-07-05:
It is not quite as simple as alarming on queue full.  By the time that the queue fills up, the driver has already attempted to send ten other commands. The fact that they are still in the queue means that something has already gone badly wrong.  For a reliable fix we need to understand that earlier sequence.  We might need to issue an alarm way before the queue fills up.  If you send $message/level=debug, FluidNC will report more details about the interaction with the spindle.

Another thing to try is to switch to the "Generic Modbus VFD" driver with model=Huanyang.  That driver has better diagnostics than the direct Huanyang driver, especially when $message/level=debug.  You would need this in the config file, instead of the huanyang: section
```yaml
ModbusVFD:
  uart_num: 1
  modbus_id: 1
  model: Huanyang
  tool_num: 0
  speed_map: 0=0% 0=25% 6000=25% 24000=100%

MitchBradley, 2025-07-05:
Are there log messages?  With $message/level=debug and debug: 3, M3 and M5 should result in reports.  This is on a machine with ModbusVFD:/model: Huanyang but no VFD is connected 
```
INFO: ModbusVFD Spindle Tx:gpio.15 Rx:gpio.16 RTS:gpio.14 Baud:9600
INFO: RS485 Tx:  01 01 03 05 00 00 2C 4F
DBG: ModbusVFD: setState:5 SpindleSpeed:0
DBG: Synced speed. Requested:0 current:0
INFO: Flood coolant gpio.12
INFO: Mist coolant gpio.4
INFO: Probe gpio.32:low
DBG:  Pos Limit 0
DBG:  Pos Limit 0
DBG:  Pos Limit 0
DBG:  Pos Limit 0
DBG: ModbusVFD: setState:5 SpindleSpeed:0
DBG: Synced speed. Requested:0 current:0
INFO: RS485 Tx:  01 01 03 05 00 00 2C 4F
INFO: RS485 Rx: 
INFO: RS485 No response
INFO: RS485 Tx:  01 01 03 05 00 00 2C 4F
INFO: RS485 Rx: 
INFO: RS485 No response
...
> m3 s7000
ok
DBG: Synced speed. Requested:0 current:0
INFO: RS485 Tx:  01 01 03 05 00 00 2C 4F
INFO: RS485 Rx: 
INFO: RS485 No response
```

Monberg75, 2025-07-05:
This is what it shows when trying to run it. So it has connection, but doesn't spin up the spindle

[image]

Monberg75, 2025-07-05:
> The two images look identical to me. In particular, both say "ModbusVFD Spindle"

Sorry about that, wrong picture.. I have updated the post with the correct picture.

MitchBradley, 2025-07-05:
I do not see any DBG messages in the log.  Please send $message/level=debug

MitchBradley, 2025-07-05:
Yes.  Now try with the ModbusVFD config. Also try M3 S9000

Monberg75, 2025-07-05:
Oki, changing back to the ModbusVFD I get this now, so it looks like that driver is not compatible with my VFD?

[MSG:INFO: ModbusVFD Spindle Tx:gpio.4 Rx:gpio.25 RTS:NO_PIN Baud:9600]
[MSG:INFO: Mist coolant gpio.21]
[MSG:INFO: Probe gpio.22:low]
[MSG:DBG:  All Limit 0]
[MSG:DBG:  All Limit 0]
[MSG:DBG:  All Limit 0]
[MSG:DBG:  All Limit 0]
[MSG:DBG: feed_hold_pin 0]
[MSG:DBG: macro0_pin 0]
[MSG:DBG: macro1_pin 0]
[MSG:DBG: ModbusVFD: setState:5 SpindleSpeed:0]

Grbl 3.9 [FluidNC v3.9.7 (wifi) '$' for help]
[MSG:INFO: VFD RS485 Unresponsive]
[MSG:ERR: Critical VFD RS485 Unresponsive]
[MSG:INFO: ALARM: Spindle Control]
ALARM:10

MitchBradley, 2025-07-05:
Send $/modbusvfd

MitchBradley, 2025-07-05:
Make sure you have the "  model: huanyang"  and "  debug: 3" lines in the modbusvfd: section.

MitchBradley, 2025-07-05:
Were there any errors in the startup messages?

Monberg75, 2025-07-05:
I think I found the problem.. The name hyunyang is case sensitive. I changed the capital H to lower, and uploaded and reset the controller, and now it works.

Monberg75, 2025-07-05:
Yes, now it turns on and off as expected.

Monberg75, 2025-07-05:
Grbl 3.9 [FluidNC v3.9.7 (wifi) '$' for help]
[MSG:INFO: RS485 Rx:  01 01 03 05 9C 40 44 BF]
[MSG:DBG: ModbusVFD: got maxRPM 24000]
[MSG:INFO: RS485 Tx:  01 01 03 0B 00 00 4D 8C]
[MSG:INFO: RS485 Rx:  01 01 03 0B 34 15 9A 83]
[MSG:DBG: ModbusVFD: got minRPM 7999]
[MSG:INFO: RS485 Tx:  01 03 01 08 F1 8E]
[MSG:INFO: RS485 Rx:  01 03 01 00 F0 48]
m3 s8000
[MSG:DBG: ModbusVFD: setState:3 SpindleSpeed:8000]
[MSG:DBG: Synced speed. Requested:0 current:0]
ok
[MSG:INFO: RS485 Tx:  01 03 01 01 31 88]
[MSG:INFO: RS485 Rx:  01 03 01 00 F0 48]
[MSG:INFO: RS485 Tx:  01 05 02 00 00 B8 CC]
[MSG:INFO: RS485 Rx:  01 05 02 00 00 B8 CC]
m5
[MSG:DBG: ModbusVFD: setState:5 SpindleSpeed:8000]
[MSG:INFO: RS485 Tx:  01 03 01 08 F1 8E]
[MSG:INFO: RS485 Rx:  01 03 01 09 30 4E]
[MSG:DBG: Syncing speed. Requested: 0 current:-1]
[MSG:INFO: RS485 Tx:  01 04 03 01 00 00 A1 8E]
[MSG:INFO: RS485 Rx:  01 04 03 01 25 89 7A B8]
[MSG:DBG: Syncing speed. Requested: 0 current:5765]
[MSG:DBG: Synced speed. Requested:0 current:5765]
ok

MitchBradley, 2025-07-05:
Now that we have debugging enabled, perhaps we can see what is going wrong prior to the queue full situation.

Monberg75, 2025-07-05:
Ok, how to check that?
I read in the wiki, that the reason I get the message can very well be be course I didn't terminate the modbus.. I will do this later, but the main problem is that when it occurs, the process is not stopped, but the machine continues to do the job. UGS admin says it's be course the controller doesn't issue it as an alarm, but only as a message.. I think this is wrong in general?

Monberg75, 2025-07-05:
I have no idea how to produce the problem.. The problem appears very randomly, and haven't been there for months actually, but now I had it yesterday, and thats why i flagged it.

Monberg75, 2025-07-05:
Ok, so a full queue and a spindle not running when it should is not appropriate? Or am I not understand correctly?

If the spindle isn't running when it should, and the machine is moving.. That equals a crash of some sort. And should stop the machine immidiately..

MitchBradley, 2025-07-05:
I will talk through the issue with Bart when he gets back from his trip.  At some level, you are correct, but at another level, it is more complicated than it might seem, due to the way that GCode and VFD communication is synchronized.

With the ModbusVFD driver, when you ask for a state or speed changes, the driver polls the VFD until the change has occurred, blocking the GCode execution.  That does not happen with the legacy Huanyang driver.  It uses fixed (configurable) delays for spinup and spindown.

I suspect that you would not have seen the problem with the ModbusVFD driver because of its poll-until-ready behavior.

MitchBradley, 2025-07-05:
With the ModbusVFD driver, GCode execution would not proceed after an M3, M5, or S change until the spindle has reached the requested speed.  That should be testable; just disconnect the rs485 cable while the VFD is spinning up.

I think Bart is on the way back now.

I am not promising to do much more about it.  I have spent much more time on other people's VFD problems than I care to think about.

Monberg75, 2025-07-05:
I will test it and if that is the case that could be a solution until the other thing is fixed.