bdring/FluidNC #1522: Problem: work stop with [MSG:WARN:: Low memory: 14200 bytes] message

ninnux, 2025-07-07:
### Wiki Search Terms

memory, Low memory, etc

### Controller Board

jackpot from v1Engineering

### Machine Description

motor drivers are TMC2209

### Input Circuits

```yaml

```

### Configuration file

```yaml
board: Jackpot TMC2209
name: MPCNC
meta: 03-15-2024 RyanZ

planner_blocks: 32

stepping:
  engine: I2S_STATIC
  idle_ms: 255
  pulse_us: 2
  dir_delay_us: 1
  disable_delay_us: 0

uart1:
  txd_pin: gpio.0
  rxd_pin: gpio.4
  rts_pin: NO_PIN
  baud: 115200
  mode: 8N1

axes:
  shared_stepper_disable_pin: NO_PIN

  x:
    steps_per_mm: 50.000
    max_rate_mm_per_min: 9000.000
    acceleration_mm_per_sec2: 200.000
    max_travel_mm: 1220
    soft_limits: false
    homing:
      cycle: 1
      positive_direction: false
      mpos_mm: 3
      feed_mm_per_min: 300.000
      seek_mm_per_min: 1500.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100
    #X
    motor0:
      limit_neg_pin: gpio.25:high
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 4.000
      tmc_2209:
        uart_num: 1
        addr: 0
        cs_pin: NO_PIN
[... 977 lines cut ...]
G3 X-44.458 Y46.457 Z1.000 I92.674 J97.308 K0.000
G3 X-38.684 Y42.591 Z1.000 I25.740 J32.202 K0.000
G3 X-36.066 Y41.599 Z1.000 I4.309 J7.420 K0.000
G3 X-36.005 Y42.516 Z1.000 I-2.890 J0.654 K0.000
G3 X-36.824 Y46.529 Z1.000 I-21.568 J-2.316 K0.000
G3 X-41.129 Y57.111 Z1.000 I-85.111 J-28.453 K0.000
G3 X-51.019 Y75.475 Z1.000 I-259.047 J-127.667 K0.000
G3 X-67.953 Y102.543 Z1.000 I-550.856 J-325.784 K0.000
G3 X-100.261 Y148.517 Z1.000 I-887.717 J-589.492 K0.000
G2 X-98.113 Y152.496 Z1.000 I2.013 J1.483 K0.000
G1 X-83.338 Y151.700 Z1.000
G2 X-81.483 Y150.717 Z1.000 I-0.135 J-2.496 K0.000
G1 X-51.546 Y111.374 Z1.000
G0 Z15.000
G0 Z15.000
(Finish operation: Profile)
(Begin postamble)
M5
G17 G90
M2

### Other Information

I have a dev kit with 4MB and not psram.
This one: https://www.amazon.it/dp/B0DKF9NCN1?ref=ppx_yo2ov_dt_b_fed_asin_title

ninnux, 2025-07-07:
I tried to use STA-AP, STA only and last time I tried AP only, so  wifi details in my logs.
Do you suggest to use STA mode only and retry?

bdring, 2025-07-07:
Sta only when running gcode