bdring/FluidNC #1529: Problem: Can not set properly param steps_per_mm for tiny lead screw

tomasz90, 2025-07-23:
### Wiki Search Terms

steps_per_mm

### Controller Board

DLC32

### Machine Description

The ANT CNC

### Input Circuits

```yaml

```

### Configuration file

```yaml
board: MKS-DLC32 V2.1
name: K40 Laser
meta: (01.01.2022) by Skorpi
kinematics: {}
stepping:
  engine: I2S_STATIC
  idle_ms: 255
  pulse_us: 4
  dir_delay_us: 1
  disable_delay_us: 0
axes:
  shared_stepper_disable_pin: i2so.0
  x:
    steps_per_mm: 80.09
    max_rate_mm_per_min: 12000
    acceleration_mm_per_sec2: 300
    max_travel_mm: 135
    soft_limits: true
    homing:
      cycle: 1
      positive_direction: false
      mpos_mm: 0
      seek_mm_per_min: 800
      feed_mm_per_min: 200
      settle_ms: 500
      seek_scaler: 1.1
      feed_scaler: 1.1
    motor0:
      limit_neg_pin: gpio.36
      hard_limits: false
      pulloff_mm: 2
      stepstick:
        step_pin: I2SO.1
        direction_pin: I2SO.2:low
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
  y:
    steps_per_mm: 80.09
    max_rate_mm_per_min: 18000
    acceleration_mm_per_sec2: 1500
    max_travel_mm: 155
    soft_limits: true
    homing:
      cycle: 1
      positive_direction: true
      mpos_mm: 0
      seek_mm_per_min: 800
      feed_mm_per_min: 200
      settle_ms: 500
      seek_scaler: 1.1
      feed_scaler: 1.1
    motor0:
      limit_neg_pin: gpio.35
      hard_limits: false
      pulloff_mm: 2
      stepstick:
        step_pin: I2SO.5
        direction_pin: I2SO.6:low
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
  z:
    steps_per_mm: 5000
    max_rate_mm_per_min: 12000
    acceleration_mm_per_sec2: 500
    max_travel_mm: 1000
    soft_limits: true
    homing:
      cycle: 0
      positive_direction: true
      mpos_mm: 0
      seek_mm_per_min: 800
      feed_mm_per_min: 1000
      settle_ms: 500
      seek_scaler: 1.1
      feed_scaler: 1.1
    motor0:
      limit_neg_pin: gpio.34
      hard_limits: false
      pulloff_mm: 10
      stepstick:
        step_pin: I2SO.3
        direction_pin: I2SO.4
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
i2so:
  bck_pin: gpio.16
  data_pin: gpio.21
  ws_pin: gpio.17
spi:
  miso_pin: gpio.12
  mosi_pin: gpio.13
  sck_pin: gpio.14
sdcard:
  cs_pin: gpio.15
  card_detect_pin: NO_PIN
control:
  safety_door_pin: NO_PIN
  reset_pin: NO_PIN
  feed_hold_pin: NO_PIN
  cycle_start_pin: NO_PIN
  macro0_pin: gpio.33:low:pu
  macro1_pin: NO_PIN
  macro2_pin: NO_PIN
  macro3_pin: NO_PIN
macros:
  macro0: $SD/Run=lasertest.gcode
  macro1: $SD/Run=home.gcode
coolant:
  flood_pin: NO_PIN
  mist_pin: NO_PIN
  delay_ms: 0
probe:
  pin: gpio.22
  check_mode_start: true
Laser:
  pwm_hz: 5000
  output_pin: gpio.32
  enable_pin: I2SO.7
  disable_with_s0: false
  s0_with_disable: false
  tool_num: 0
  speed_map: 0=0.000% 0=12.500% 1700=100.000%
user_outputs:
  analog0_pin: NO_PIN
  analog1_pin: NO_PIN
  analog2_pin: NO_PIN
  analog3_pin: NO_PIN
  analog0_hz: 5000
  analog1_hz: 5000
  analog2_hz: 5000
  analog3_hz: 5000
  digital0_pin: NO_PIN
  digital1_pin: NO_PIN
  digital2_pin: NO_PIN
  digital3_pin: NO_PIN
start:
  must_home: false
```

### Startup Messages

```text
ERROR: Critical error in main_init: Stepping rate 1000000 steps/sec exceeds the maximum rate
125000]
```

### What happened?

I have small lead screw 4.76mm which I need to set for about 5000 steps per mm. I am not able to do it, because I am getting the following error: ERROR: Critical error in main_init: Stepping rate 1000000 steps/sec exceeds the maximum rate
125000]
Board is not booting. Right now when I set 500 steps per mm Z axis is moving 1mm when I move it by 10mm in the web interface.

### Other Information

[image]

tomasz90, 2025-07-23:
Solved by decreasing max_rate_mm_per_min

closed 2025-07-23 by tomasz90

MitchBradley, 2025-07-23:
You could also try setting pulse_us to 1.  That increases the maximum pulse rate to 500000 pulses per second.  Earlier versions of FluidNC could not pulse that fast on I2S, but recent versions can.

Another parameter you might be able to adjust is the microstepping level of your drivers.  Since you did not tell use what drivers you are using, I cannot say for sure how to do that.  It is rarely necessary to use high microstepping levels for fine pitch screws.