bdring/FluidNC #1537: Problem: FluidNC crashes after some time

TomKamphuys, 2025-08-04:
### Wiki Search Terms

crash freeze

### Controller Board

AZDelivery ESP32 NodeMCU D1 R32 WLAN development board with CH340G and WiFi + Bluetooth IoT with micro USB compatible with Arduino

https://www.az-delivery.de/en/products/esp32-d1-r32-board?_pos=2&_sid=911b0b730&_ss=r

### Machine Description

I'm building a scanner to measure speaker output at several positions. I'm using a python program running on a raspberry pi to send a position to fluidnc and wait until it is ready and do an audio sweep at 500 to 5000 points. 

### Input Circuits

```yaml

```

### Configuration file

```yaml
board: ESPDUINO-32
name: PANT.S.

axes:
  shared_stepper_disable_pin: gpio.12:high
  
  x:
    steps_per_mm: 17.778
    max_rate_mm_per_min: 100000
    acceleration_mm_per_sec2: 5000
    max_travel_mm: 1000
    homing:
      cycle: 2
      mpos_mm: 10
      positive_direction: false
    
    motor0:
      stepstick:
        direction_pin: gpio.16
        step_pin: gpio.26

  y:
    steps_per_mm: 17.778
    max_rate_mm_per_min: 100000
    acceleration_mm_per_sec2: 5000
    max_travel_mm: 1000
    homing:
      cycle: 2
      mpos_mm: 10
      positive_direction: false

    motor0:
      stepstick:
        direction_pin: gpio.27
        step_pin: gpio.25

  z:
    steps_per_mm: 17.778
    max_rate_mm_per_min: 100000
    acceleration_mm_per_sec2: 5000
    max_travel_mm: 1000
    homing:
      cycle: 1
      mpos_mm: 10
      positive_direction: true

    motor0:
      stepstick:
        direction_pin: gpio.14
        step_pin: gpio.17

  a:
    steps_per_mm: 17.778
    max_rate_mm_per_min: 100000
    acceleration_mm_per_sec2: 5000
    max_travel_mm: 1000
    homing:
      cycle: 1
      mpos_mm: 10
      positive_direction: true

    motor0:
      stepstick:
        direction_pin: gpio.18
        step_pin: gpio.19
```

### Startup Messages

```text
websockets.client DEBUG    = connection is CONNECTING
websockets.client DEBUG    > GET / HTTP/1.1
websockets.client DEBUG    > Host: fluidnc2.local:81
websockets.client DEBUG    > Upgrade: websocket
websockets.client DEBUG    > Connection: Upgrade
websockets.client DEBUG    > Sec-WebSocket-Key: SRu4a1BqSF3al+y5UZmD+g==
websockets.client DEBUG    > Sec-WebSocket-Version: 13
websockets.client DEBUG    > Sec-WebSocket-Extensions: permessage-deflate; client_max_window_bits
websockets.client DEBUG    > User-Agent: Python/3.11 websockets/15.0.1
websockets.client DEBUG    < HTTP/1.1 101 Switching Protocols
websockets.client DEBUG    < Server: arduino-WebSocketsServer
websockets.client DEBUG    < Upgrade: websocket
websockets.client DEBUG    < Connection: Upgrade
websockets.client DEBUG    < Sec-WebSocket-Version: 13
websockets.client DEBUG    < Sec-WebSocket-Accept: v6n+Z412DpC+V6e3Qns7IQ/gNAc=
websockets.client DEBUG    = connection is OPEN
websockets.client DEBUG    > TEXT '$X\n' [3 bytes]
websockets.client DEBUG    < PING '' [0 bytes]
websockets.client DEBUG    > PONG '' [0 bytes]
websockets.client DEBUG    < TEXT 'CURRENT_ID:0' [12 bytes]
websockets.client DEBUG    < TEXT 'ACTIVE_ID:0' [11 bytes]
websockets.client DEBUG    < BINARY 5b 4d 53 47 3a 49 4e 46 4f 3a 20 43 61 75 74 69 ... 6c 6f 63 6b 65 64 5d 0a [30 bytes]
websockets.client DEBUG    < BINARY 6f 6b 0a [3 bytes]
websockets.client DEBUG    > TEXT '$/axes/x/steps_per_mm=72.752\n' [29 bytes]
websockets.client DEBUG    < BINARY 6f 6b 0a [3 bytes]
websockets.client DEBUG    > TEXT '$/axes/x/max_rate_mm_per_min=25000.0\n' [37 bytes]
websockets.client DEBUG    < BINARY 6f 6b 0a [3 bytes]
websockets.client DEBUG    > TEXT '$/axes/x/acceleration_mm_per_sec2=100.0\n' [40 bytes]
websockets.client DEBUG    < BINARY 6f 6b 0a [3 bytes]
websockets.client DEBUG    > TEXT '$/axes/y/steps_per_mm=72.752\n' [29 bytes]
websockets.client DEBUG    < BINARY 6f 6b 0a [3 bytes]
websockets.client DEBUG    > TEXT '$/axes/y/max_rate_mm_per_min=25000.0\n' [37 bytes]
websockets.client DEBUG    < BINARY 6f 6b 0a [3 bytes]
websockets.client DEBUG    > TEXT '$/axes/y/acceleration_mm_per_sec2=100.0\n' [40 bytes]
websockets.client DEBUG    < BINARY 6f 6b 0a [3 bytes]
websockets.client DEBUG    > TEXT 'G92 X0 Y230\n' [12 bytes]
websockets.client DEBUG    < BINARY 6f 6b 0a [3 bytes]
websockets.client DEBUG    > TEXT '$10=0\n' [6 bytes]
websockets.client DEBUG    < BINARY 6f 6b 0a [3 bytes]
```

### User Interface Software

Custom python program connecting through websocket

### What happened?

During the measurement(s), it stops. Fluidnc is not reacting anymore, so the measurement process freezes.

### GCode File

n.a.

### Other Information

I have tried to make the setup as small as possible in trying to find out what is wrong. I've ended up with a wifi router, a raspberry pi, the esp32 board with fluidnc and a laptop to control everything. If I only ping fluidnc I get some time-outs every now and then, but it recovers.

[image]

When I start my measurements program, the frequency of the time-outs seems to increase and fluidnc does not recover.

[image]

Pinging of the raspberry pi and router show no abnormalities.

TomKamphuys, 2025-08-05:
https://github.com/TomKamphuys/NFS/tree/feature/arcs

N.B. use the feature arcs branch

TomKamphuys, 2025-08-05:
I'm planning to output the gcode commands to a file and see if running that file from e.g. UGS has the same problems.

MitchBradley, 2025-08-05:
Are you using STA mode or AP mode?

TomKamphuys, 2025-08-05:
STA

MitchBradley, 2025-08-05:
This setup is too complicated.  I tried to run nfs_app.py on WSL, after installing the dependencies via requirements.txt and changing config.ini as instructed above.  It failed with "OSError: PortAudio library not found" from audio.py.  I eventually resolved that by installing all the packages to allow the pyaudio wheel to build, but then it failed with "Command '/usr/bin/x-www-browser' requires the firefox snap to be installed."  I don't have time to go to a deeper circle of Dependency Hell, nor to set up a Raspberry Pi for the test.

I presume that most of this infrastructure is unrelated to the specific problem at hand.  Perhaps you could construct a simpler test without all the bells and whistles.

TomKamphuys, 2025-08-05:
I understand. I'll try to reproduce it with a gcode file or a simple python script with as little dependencies as possible sending gcodes via een socket.

I'll report back.

MitchBradley, 2025-08-05:
I suspect that the problem has something to do with the network stack, which can in some cases consume a lot of memory.  If that is the case, it probably will not fail with a gcode file.

TomKamphuys, 2025-08-13:
I've done a ton of testing over the last days. At one point it seemed that the wifi signal was important. Putting the esp32 on the router gave great results. So I ordered a new router to replace the compact travel router. During shipping I continued testing and it seemed that even further away things kept going well. Once I included the steppers in the mix the problem popped up again. Even when using the new router it stopped sometimes. And now even with the steppers not connected I experienced the problem again.

This is a typical ping result:

[image]

You see the ping times increase and then it stops.

As I'm writing this comment, the communication has started again and the program continues sending gcode:

[image]

Does anyone have any idea what to test next? Where to look at?

TomKamphuys, 2025-08-29:
I've been testing a lot. It's still inconclusive. Is there anything I can watch (cpu load, memory usage, ...) or debug?

TomKamphuys, 2025-11-10:
I'm currently using an OrangePi to do the wifi communication and talk to the esp32duino with fluidnc via usb. This works flawlessly. Any wifi is fine. I've run several 5000 to 20000 point test runs and they all worked fine.