bdring/FluidNC #1542: Problem: Configure wifi webpage reports incorrect settings after save

TomKamphuys, 2025-08-10:
### Wiki Search Terms

wifi configuration setup

### Controller Board

[6-pack](https://www.az-delivery.de/nl/products/esp32-d1-r32-board)

### Machine Description

No machine at all. Laptop -> usb cable -> esp32 board

### Input Circuits

```yaml

```

### Configuration file

```yaml
n.a.
```

### Startup Messages

```text
n.a.
```

### User Interface Software

n.a.

### What happened?

I changes the wifi configuration and clicked save. I expected that the entered names would have been taken over.

### GCode File

n.a.

### Other Information

I changes the wifi configuration and clicked save. I expected that the entered names would have been taken over, but it hasn't.

[image]

The (changed) hostname 'fluidnc' -> 'fluidnc3' has been prepended to the (changed) SSID 'NearFieldScanner' -> 'fluidnc2NearFieldScanner'. That has in turn be prepended to the (changed) Access Point SSID 'FluidNC3' -> 'fluidnc3NearFieldScannerFluidNC3'

bdring, 2025-08-10:
Try setting fields manually from the console and checking the value.

http://wiki.fluidnc.com/en/features/wifi_bt#wifi-settings

Send `$STA/SSID=<your SSID>`
Then send `$STA/SSID` to check the value.

Show us the console session.

TomKamphuys, 2025-08-10:
Full Output:

`
[15:09:43] [ usb ports ] Detected a change in available ports: [COM6][COM5]
[15:09:47] [ connect ] PORT INFO: Port is now open: COM5 - Attempting to detect Controller...
[15:09:47] [ connect ] Attempting to detect Controller (1): (Autoreset)
[15:09:47] [ ] ok
[15:09:48] [ connect ] Attempting to detect Controller (2): (Ctrl+X)
[15:09:48] [ ] Grbl 3.9 [FluidNC v3.9.8 (wifi) '$' for help]
[15:09:49] [ connect ] Firmware Detected: FluidNC version 3.9.8 dated on COM5
[15:09:50] [ $CD ] board: None
[15:09:50] [ $CD ] name: Default (Test Drive no I/O)
[15:09:50] [ $CD ] meta:
[15:09:50] [ $CD ] stepping:
[15:09:50] [ $CD ] engine: RMT
[15:09:50] [ $CD ] idle_ms: 255
[15:09:50] [ $CD ] pulse_us: 4
[15:09:50] [ $CD ] dir_delay_us: 0
[15:09:50] [ $CD ] disable_delay_us: 0
[15:09:50] [ $CD ] segments: 12
[15:09:50] [ $CD ] spi:
[15:09:50] [ $CD ] miso_pin: NO_PIN
[15:09:50] [ $CD ] mosi_pin: NO_PIN
[15:09:50] [ $CD ] sck_pin: NO_PIN
[15:09:50] [ $CD ] sdcard:
[15:09:50] [ $CD ] cs_pin: NO_PIN
[15:09:50] [ $CD ] card_detect_pin: NO_PIN
[15:09:50] [ $CD ] frequency_hz: 8000000
[15:09:50] [ $CD ] kinematics:
[15:09:50] [ $CD ] Cartesian:
[15:09:50] [ $CD ] axes:
[15:09:50] [ $CD ] shared_stepper_disable_pin: NO_PIN
[15:09:50] [ $CD ] shared_stepper_reset_pin: NO_PIN
[15:09:50] [ $CD ] homing_runs: 2
[15:09:50] [ $CD ] x:
[15:09:50] [ $CD ] steps_per_mm: 80.000000
[15:09:50] [ $CD ] max_rate_mm_per_min: 1000.000000
[15:09:50] [ $CD ] acceleration_mm_per_sec2: 25.000000
[15:09:50] [ $CD ] max_travel_mm: 1000.000000
[15:09:50] [ $CD ] soft_limits: false
[15:09:50] [ $CD ] motor0:
[15:09:50] [ $CD ] limit_neg_pin: NO_PIN
[15:09:50] [ $CD ] limit_pos_pin: NO_PIN
[15:09:50] [ $CD ] limit_all_pin: NO_PIN
[15:09:50] [ $CD ] hard_limits: false
[15:09:50] [ $CD ] pulloff_mm: 1.000000
[15:09:50] [ $CD ] null_motor:
[15:09:50] [ $CD ] y:
[15:09:50] [ $CD ] steps_per_mm: 80.000000
[15:09:50] [ $CD ] max_rate_mm_per_min: 1000.000000
[15:09:50] [ $CD ] acceleration_mm_per_sec2: 25.000000
[15:09:50] [ $CD ] max_travel_mm: 1000.000000
[15:09:50] [ $CD ] soft_limits: false
[15:09:50] [ $CD ] motor0:
[15:09:50] [ $CD ] limit_neg_pin: NO_PIN
[15:09:50] [ $CD ] limit_pos_pin: NO_PIN
[15:09:50] [ $CD ] limit_all_pin: NO_PIN
[15:09:50] [ $CD ] hard_limits: false
[15:09:50] [ $CD ] pulloff_mm: 1.000000
[15:09:50] [ $CD ] null_motor:
[15:09:50] [ $CD ] z:
[15:09:50] [ $CD ] steps_per_mm: 80.000000
[15:09:50] [ $CD ] max_rate_mm_per_min: 1000.000000
[15:09:50] [ $CD ] acceleration_mm_per_sec2: 25.000000
[15:09:50] [ $CD ] max_travel_mm: 1000.000000
[15:09:50] [ $CD ] soft_limits: false
[15:09:50] [ $CD ] motor0:
[15:09:50] [ $CD ] limit_neg_pin: NO_PIN
[15:09:50] [ $CD ] limit_pos_pin: NO_PIN
[15:09:50] [ $CD ] limit_all_pin: NO_PIN
[... 91 lines cut ...]
[15:14:14] [ $STA/SSID=NearFieldScanner ] ok
[15:14:42] [ $AP/SSID=FluidNC3 ] ok
[15:14:57] [ $hostname ] $Hostname=fluidnc3 ;unknown
[15:14:57] [ $hostname ] ok
[15:15:12] [ $STA/SSID ] $Sta/SSID=fluidnc3NearFieldScanner ;unknown
[15:15:12] [ $STA/SSID ] ok
[15:15:27] [ $AP/SSID ] $AP/SSID=fluidnc3NearFieldScannerFluidNC3 ;unknown
[15:15:27] [ $AP/SSID ] ok
`

Here it what I did:

`
[15:13:45] [ $hostname=fluidnc3 ] ok
[15:14:14] [ $STA/SSID=NearFieldScanner ] ok
[15:14:42] [ $AP/SSID=FluidNC3 ] ok
[15:14:57] [ $hostname ] $Hostname=fluidnc3 ;unknown
[15:14:57] [ $hostname ] ok
[15:15:12] [ $STA/SSID ] $Sta/SSID=fluidnc3NearFieldScanner ;unknown
[15:15:12] [ $STA/SSID ] ok
[15:15:27] [ $AP/SSID ] $AP/SSID=fluidnc3NearFieldScannerFluidNC3 ;unknown
[15:15:27] [ $AP/SSID ] ok
`

Same as with the web interface.

bdring, 2025-08-10:
We will look into that.

bdring, 2025-08-10:
Try this PR

https://github.com/bdring/FluidNC/pull/1543

shampeon, 2025-12-11:
The problem with the Web Installer not saving Wifi settings still occurs with v3.9.9. I encountered it setting up a new Corgi board. I had to set the SSID/Password in the Terminal in order to preserve the settings. The Web Installer throws no errors.

```
[VER:3.9 FluidNC v3.9.9:]
[OPT:MPHS]
[MSG:Machine: Corgi Default]
```

This is the console output after seemingly successfully saving the SSID/Password/DHCP in Client mode.

```
Grbl 3.9 [FluidNC v3.9.9 (wifi) '$' for help]
$STA
$Sta/Password=********
$Sta/MinSecurity=WPA2-PSK
$Sta/IPMode=DHCP
$Sta/IP=0.0.0.0
$Sta/Gateway=0.0.0.0
$Sta/Netmask=0.0.0.0
$Sta/SSID=
$Start/Message=Grbl \V [FluidNC \B (\R) \H]
$Report/Status=1
ok
```

MitchBradley, 2025-12-11:
> The problem with the Web Installer not saving Wifi settings still occurs with v3.9.9.

3.9.9 was released on Oct 4.  The fix referenced above was created on Oct 10.  Causality is still in effect.

shampeon, 2025-12-11:
The fix (#1543) mentioned above was merged into main on August 31. Where are you getting October 10th?