bdring/FluidNC #1546: Problem: Unable to connect FluidNC-M5Dial/RJ12 version to 6XCNC Controller

shenzhenadam, 2025-08-17:
### Wiki Search Terms

fludidnc pendant

### Controller Board

FluidNC 6X CNC controller

### Machine Description

Custom

### Input Circuits

```yaml

```

### Configuration file

```yaml
board: 6x
name: 6x Default
stepping:
  engine: I2S_STREAM
  idle_ms: 254
  pulse_us: 4
  dir_delay_us: 1
  disable_delay_us: 0

axes:
  shared_stepper_disable_pin: NO_PIN
  x:
    steps_per_mm: 2560.000
    max_rate_mm_per_min: 2000.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 500.000
    soft_limits: true
    homing:
      cycle: 2
      positive_direction: false
      mpos_mm: 200.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 1000.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100
      
    motor0:
      
      limit_pos_pin: NO_PIN
      limit_neg_pin: NO_PIN
      limit_all_pin: gpio.2:high:pu
      hard_limits: false
      pulloff_mm: 1.000
      standard_stepper:
        step_pin: I2SO.2
        direction_pin: I2SO.1
        disable_pin: I2SO.0

  y:
    steps_per_mm: 2560.000
    max_rate_mm_per_min: 1000.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 600.000
    soft_limits: true
    homing:
      cycle: 3
      positive_direction: false
      mpos_mm: 400.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 1000.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: gpio.26:high:pu
      hard_limits: false
      pulloff_mm: 5.000
      standard_stepper:
        step_pin: I2SO.5
        direction_pin: I2SO.4
        disable_pin: I2SO.7
    motor1:
      limit_neg_pin: NO_PIN    
      limit_pos_pin: NO_PIN
      limit_all_pin: gpio.33:high:pu
      hard_limits: false
      pulloff_mm: 5.000
      standard_stepper:
        step_pin: I2SO.18
        direction_pin: I2SO.17
        disable_pin: I2SO.16

  z:
    steps_per_mm: 2560.000
    max_rate_mm_per_min: 2000.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 150.000
    soft_limits: false
    homing:
      cycle: 1
      positive_direction: true
      mpos_mm: 1.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 1000.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: gpio.32:high
      hard_limits: false
      pulloff_mm: 1.000
      standard_stepper:
        step_pin: I2SO.21
        direction_pin: I2SO.20
        disable_pin: I2SO.23
        
 
        
       
i2so:
  bck_pin: gpio.22
  data_pin: gpio.21
  ws_pin: gpio.17

spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

sdcard:
  card_detect_pin: NO_PIN
  cs_pin: gpio.5
  
probe:
  pin: gpio.39:low
  toolsetter_pin: gpio.36:low
  
# Using MOSFETs (Check Spindle Pin Usage
coolant:
  flood_pin: gpio.12
  mist_pin: gpio.4
  delay_ms: 0

start:
  must_home: false
  
## Begin Huanyang  
# uart1:
  # txd_pin: gpio.15
  # rxd_pin: gpio.16
  # rts_pin: gpio.14
  # baud: 9600
  # mode: 8N1

# Huanyang:
  # uart_num: 1
  # modbus_id: 1
  # tool_num: 0
  # speed_map: 0=0% 0=25% 6000=25% 24000=100%
  # off_on_alarm: false

# #begin PWM
pwm:
  pwm_hz: 5000
  direction_pin: NO_PIN
  output_pin: gpio.13
  enable_pin: gpio.14
  disable_with_s0: false
  s0_with_disable: true
  spinup_ms: 0
  spindown_ms: 0
  tool_num: 0
  speed_map: 0=0.000% 10000=100.000%
  off_on_alarm: false
  
# #begin Laser
# Laser:
  # pwm_hz: 5000
  # output_pin: gpio.4
  # enable_pin: gpio.12
  # disable_with_s0: false
  # s0_with_disable: true
  # tool_num: 1
  # speed_map: 0=0.000% 255=100.000%
  # off_on_alarm: true
  
# #begin 10V
# 10V:
  # forward_pin: gpio.15
  # reverse_pin: gpio.14
  # pwm_hz: 5000
  # output_pin: gpio.13
  # enable_pin: NO_PIN
  # direction_pin: NO_PIN
  # disable_with_s0: false
  # s0_with_disable: true
  # spinup_ms: 0
  # spindown_ms: 0  
  # tool_num: 0
  # speed_map: 0=0.000% 1000=0.000% 24000=100.000%
  # off_on_alarm: false
  
  uart2:
  txd_pin: gpio.25
  rxd_pin: gpio.27
  rts_pin: NO_PIN
  cts_pin: NO_PIN
  baud: 1000000
  mode: 8N1

uart_channel1:
  report_interval_ms: 75
  uart_num: 2
```

### Startup Messages

```text
[MSG:INFO: FluidNC v3.9.8 https://github.com/bdring/FluidNC]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.7-dirty]
[MSG:INFO: Local filesystem type is littlefs]
[MSG:INFO: Configuration file:6x_XYYZ-NEW_IO-13072025.yaml]
[MSG:ERR: Ignored key uart2]
[MSG:ERR: Ignored key txd_pin]
[MSG:ERR: Ignored key rxd_pin]
[MSG:ERR: Ignored key rts_pin]
[MSG:ERR: Ignored key cts_pin]
[MSG:ERR: Ignored key baud]
[MSG:ERR: Ignored key mode]
[MSG:INFO: Machine 6x Default]
[MSG:INFO: Board 6x]
[MSG:ERR: UartChannel: missing uart2]
[MSG:INFO: I2SO BCK:gpio.22 WS:gpio.17 DATA:gpio.21Min Pulse:2us]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
[MSG:INFO: SD Card cs_pin:gpio.5 detect:NO_PIN freq:8000000]
[MSG:INFO: Stepping:I2S_STREAM Pulse:4us Dsbl Delay:0us Dir Delay:1us Idle Delay:254ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Axis X (200.000,700.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:I2SO.2 Dir:I2SO.1 Disable:I2SO.0]
[MSG:INFO:  All Limit gpio.2:pu]
[MSG:INFO: Axis Y (400.000,1000.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:I2SO.5 Dir:I2SO.4 Disable:I2SO.7]
[MSG:INFO:  All Limit gpio.26:pu]
[MSG:INFO:   Motor1]
[MSG:INFO:     standard_stepper Step:I2SO.18 Dir:I2SO.17 Disable:I2SO.16]
[MSG:INFO:  All Limit gpio.33:pu]
[MSG:INFO: Axis Z (-149.000,1.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:I2SO.21 Dir:I2SO.20 Disable:I2SO.23]
[MSG:INFO:  All Limit gpio.32]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: STA SSID is not set]
[MSG:INFO: AP SSID FluidNC IP 192.168.0.1 mask 255.255.255.0 channel 1]
[MSG:INFO: AP started]
[MSG:INFO: WiFi on]
[MSG:INFO: Captive Portal Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]
ok
```

### User Interface Software

WebUI, UGS

### What happened?

Unable to connect FluidNC-M5Dial/RJ12 verison to 6XCNC Controller - 
1. The yaml file that I use works fine with my CNC machine, 
2. When I add the additional code at the end of the yaml file, this error message shows up in the fluidnc browser.
3. Additional code added to the end of the yaml file: 
uart2:
  txd_pin: gpio.25
  rxd_pin: gpio.27
  rts_pin: NO_PIN
  cts_pin: NO_PIN
  baud: 1000000
  mode: 8N1

uart_channel1:
  report_interval_ms: 75
  uart_num: 2
4. I have already updated the firmware to the latest on the M5Dial and the 
5. See attached error messages, yaml file, and reference images. 

### Other Information

[image]
[image]
[Error-message-17Aug2025.txt](https://github.com/user-attachments/files/21822222/Error-message-17Aug2025.txt)
[image]
[6x_XYYZ-NEW_IO-13072025.yaml](https://github.com/user-attachments/files/21822221/6x_XYYZ-NEW_IO-13072025.yaml)
[image]

breiler, 2025-08-17:
Yes, as described here:
http://wiki.fluidnc.com/en/config/uart_sections

bdring, 2025-08-17:
The startup messages tell you where the error is. It is ignoring uart2. The config file has it indented under `pwm:` and pwm does not have a uart item.

```
[MSG:ERR: Ignored key uart2]
[MSG:ERR: Ignored key txd_pin]
[MSG:ERR: Ignored key rxd_pin]
[MSG:ERR: Ignored key rts_pin]
[MSG:ERR: Ignored key cts_pin]
[MSG:ERR: Ignored key baud]
[MSG:ERR: Ignored key mode]
[MSG:INFO: Machine 6x Default]
[MSG:INFO: Board 6x]
[MSG:ERR: UartChannel: missing uart2]
```

shenzhenadam, 2025-08-17:
I updated the yaml and removed the spaces ... the pendant now connects and can control the machine via pendant .. Many thanks for your help.

closed 2025-08-17 by bdring