bdring/FluidNC #1555: Feature: Deceleration During Homing Feture

TmanEdi, 2025-08-29:
### Machine Context

I have a couple of machines with a considerable mass. When homing the lack of deceleration
 is beating up my machines. I'm just asking for an option to decelerate when reaching the switch. I'm a transplant from Mach4 to Fluid NC and a 6X CNC controller.
I'm 70yo, semi retired and have been building CNC machines for 40 years. (I LOVE this controller! and Fluid NC) So far I have implemented 6 of these controllers, the next one is a Delta. as I don't log in often please feel free to contact me via email. Edi@Murway.com

### Feature Description

I'm just asking for an option to decelerate when reaching the switch.
I know people that have retrofitted larger machines like lathes and mills will appreciate this feature.
Larger machines with ball lead screws when zipping along toward home and instantly stopping will be abusing there ball nuts and screws.

### Other Approaches

There is nothing that can be done to resolve this issue.

### How I Can Help

I would be Happy to make a $100 donation toward effort to provide this option. I have been writing code in a few languages since 1969, I know how valuable coding time is, perhaps others can help with donations.

eriknz, 2026-04-10:
I think to go along with this, it would also be nice to change how pulloff_mm works so you don't lose working space. If you have enough space after triggering your limit switch for your axis to decelerate before hitting the physical stop, you will probably want to set pulloff_mm larger than that distance so it can clear the switch when you rehome after a crash. But as it's currently set up, when it does the second homing run at the slower speed and the switch triggers, it will pull off from it that full pulloff_mm distance from the switch.
[mikeoverbay](https://github.com/mikeoverbay) has also raised the deceleration after limit switch issue here: https://github.com/bdring/FluidNC/issues/1477

bdring, 2026-04-10:
Also multi axis homing would be a pain. Currently we only have to plan one move. We would have to interrupt and replan remaining axes.

bdring, 2026-04-18:
Closing this issue. After studying the request, we think it would cause problems and reduce existing functionality.