bdring/FluidNC #1556: Problem: Alarm when i use control pins

Cyberhornet34, 2025-08-31:
### Wiki Search Terms

control pins
Alarm 11

### Controller Board

Custom board

### Machine Description

DM542, VFD,...

### Input Circuits

```yaml
-Probe/reset/start/hold on R/C filters, internal pullups activated
-Endstops on TLP521 optocouplers
```

### Configuration file

```yaml
name: CNC V3 VFD XYYZ
stepping:
  engine: I2S_STREAM
  idle_ms: 250
  pulse_us: 4
  dir_delay_us: 0
  disable_delay_us: 0
start:
  must_home: false
  check_limits: true
axes:
  x:
    steps_per_mm: 200
    max_rate_mm_per_min: 2500
    acceleration_mm_per_sec2: 50
    max_travel_mm: 700
    homing:
      cycle: -1
      positive_direction: false
    motor0:
      limit_all_pin: gpio.36
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      hard_limits: true
      pulloff_mm: 1
      standard_stepper:
        step_pin: i2so.3
        direction_pin: i2so.2
        disable_pin: i2so.1
  y:
    steps_per_mm: 200
    max_rate_mm_per_min: 2500
    acceleration_mm_per_sec2: 50
    max_travel_mm: 700
    homing:
      cycle: -1
      positive_direction: false
    motor0:
      limit_all_pin: gpio.39
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      hard_limits: true
      pulloff_mm: 1
      standard_stepper:
        step_pin: i2so.6
        direction_pin: i2so.5
        disable_pin: i2so.4
    motor1:
      limit_all_pin: NO_PIN
      limit_neg_pin: gpio.35
      limit_pos_pin: NO_PIN
      hard_limits: true
      pulloff_mm: 1
      standard_stepper:
        step_pin: i2so.13
        direction_pin: i2so.12
        disable_pin: i2so.11
  z:
    steps_per_mm: 200
    max_rate_mm_per_min: 1000
    acceleration_mm_per_sec2: 50
    max_travel_mm: 150
    homing:
      cycle: -1
    motor0:
      limit_all_pin: NO_PIN
      limit_neg_pin: NO_PIN
      limit_pos_pin: gpio.34
      hard_limits: true
      pulloff_mm: 1
      standard_stepper:
        step_pin: i2so.10
        direction_pin: i2so.9
        disable_pin: i2so.7
i2so:
  bck_pin: gpio.22
  data_pin: gpio.21
  ws_pin: gpio.17
probe:
  pin: gpio.27:pu:low
  toolsetter_pin: NO_PIN
  check_mode_start: false
  hard_stop: false
spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18
sdcard:
  card_detect_pin: NO_PIN
  cs_pin: gpio.5
10V:
  forward_pin: gpio.15
  reverse_pin: gpio.4
  pwm_hz: 5000
  output_pin: gpio.2
  enable_pin: NO_PIN
  direction_pin: NO_PIN
  disable_with_s0: false
  s0_with_disable: true
  spinup_ms: 0
  spindown_ms: 0
  tool_num: 0
  speed_map: 0=0.000% 1000=0.000% 24000=100.000%
  off_on_alarm: false
Laser:
  tool_num: 1
  output_pin: gpio.14
coolant:
  flood_pin: I2SO.8
  mist_pin: NO_PIN
  delay_ms: 0
control:
  reset_pin: gpio.26:pu:low
  feed_hold_pin: gpio.33:pu:low
  cycle_start_pin: gpio.25:pu:low
```

### Startup Messages

```text
[MSG:INFO: uart_channel0 created]

[MSG:RST]
[MSG:INFO: FluidNC v3.9.8 https://github.com/bdring/FluidNC]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.7-dirty]
[MSG:INFO: Local filesystem type is littlefs]
[MSG:INFO: Configuration file:CNCV3_VFD_XYYZ412.yaml]
[MSG:INFO: Machine CNC V3 VFD XYYZ]
[MSG:INFO: Board None]
[MSG:INFO: I2SO BCK:gpio.22 WS:gpio.17 DATA:gpio.21Min Pulse:2us]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
[MSG:INFO: SD Card cs_pin:gpio.5 detect:NO_PIN freq:8000000]
[MSG:INFO: Stepping:I2S_STREAM Pulse:4us Dsbl Delay:0us Dir Delay:0us Idle Delay:250ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Axis X (0.000,700.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:I2SO.3 Dir:I2SO.2 Disable:I2SO.1]
[MSG:INFO:  All Limit gpio.36]
[MSG:INFO: Axis Y (0.000,700.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:I2SO.6 Dir:I2SO.5 Disable:I2SO.4]
[MSG:INFO:  All Limit gpio.39]
[MSG:INFO:   Motor1]
[MSG:INFO:     standard_stepper Step:I2SO.13 Dir:I2SO.12 Disable:I2SO.11]
[MSG:INFO:  Neg Limit gpio.35]
[MSG:INFO: Axis Z (-150.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:I2SO.10 Dir:I2SO.9 Disable:I2SO.7]
[MSG:INFO:  Pos Limit gpio.34]
[MSG:INFO: reset_pin gpio.26:low:pu]
[MSG:INFO: feed_hold_pin gpio.33:low:pu]
[MSG:INFO: cycle_start_pin gpio.25:low:pu]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: STA SSID is not set]
[MSG:INFO: AP SSID FluidNC IP 192.168.0.1 mask 255.255.255.0 channel 1]
[MSG:INFO: AP started]
[MSG:INFO: WiFi on]
[MSG:INFO: Captive Portal Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]
[MSG:INFO: 10V Spindle Ena:NO_PIN Out:gpio.2 Dir:NO_PIN Fwd:gpio.15 Rev:gpio.4 Freq:5000Hz Period:8191]
[MSG:INFO: Laser Ena:NO_PIN Out:gpio.14 Freq:5000Hz Period:8191]
[MSG:INFO: Flood coolant I2SO.8]
[MSG:INFO: Probe gpio.27:low:pu]

Grbl 3.9 [FluidNC v3.9.8 (wifi) '$' for help]
[MSG:ERR: reset_pin is active at startup]
[MSG:ERR: cycle_start_pin is active at startup]
[MSG:INFO: ALARM: Control Pin Initially On]
ALARM:11
ok
[VER:3.9 FluidNC v3.9.8:]
[OPT:PHS]
[MSG:Machine: CNC V3 VFD XYYZ]
[MSG:Mode=AP:SSID=FluidNC:IP=192.168.0.1:MAC=4C-C3-82-BE-DE-B1]
ok
```

### What happened?

Hi,
At startup, i get this message:
[MSG:ERR: reset_pin is active at startup]
[MSG:ERR: cycle_start_pin is active at startup]
[MSG:INFO: ALARM: Control Pin Initially On]
ALARM:11
my control pins and probe are mounted with R/C filters
If i desactivate the alarm, i get the same alarm when i push on the reset switch
If i put reset button on the optocoupler circuit instead of a limit switch, i don't get Alarm at startup, but i still get the same alarm when i touch the reset button.
I use this board for 4 years with "GRBL esp32" with no problem, i only get this problem since i installed FluidNC.
All my switches work well in the "calibrate" section of the FluidNC installer .
I tried with another ESP vroom32, with older FluidNC version, same result.
Thank for your help ;)

Cyberhornet34, 2025-08-31:
Thanks for fast reply,
I would love that would be so simple !
here my ? command: <Idle|MPos:0.000,0.000,0.000|FS:0,0|WCO:0.000,0.000,0.000>
if i define :high, all switches are trigered ! :low is a good configuration

Cyberhornet34, 2025-08-31:
Sometimes i don't get Alarm at startup, i get IDLE but when i touch reset switch, even few minutes after startup, i get:
[MSG:ERR: reset_pin is active at startup]
[MSG:INFO: ALARM: Control Pin Initially On]
ALARM:11

MitchBradley, 2025-08-31:
Are you using external pullups on the control pins, or depending on the internal pullups?  If you are using the internal pullups, the RC time constant could be quite long.  What is the filter capacitor value?

Cyberhornet34, 2025-08-31:
Hi Mitch,
I'm using a R/C filter (low pass) with a 100R resistors and 100nf capacitors.

here's my old config that worked well in GRBL ESP32 
```
// Inverts pin logic of the control command pins based on a mask. This essentially means you can use
// normally-closed switches on the specified pins, rather than the default normally-open switches.
// The mask order is ...
// Macro3 | Macro2 | Macro 1| Macr0 |Cycle Start | Feed Hold | Reset | Safety Door
// For example B1101 will invert the function of the Reset pin.
#define INVERT_CONTROL_PIN_MASK B00001111

#define ENABLE_CONTROL_SW_DEBOUNCE     // Default disabled. Uncomment to enable.
#define CONTROL_SW_DEBOUNCE_PERIOD 32  // in milliseconds default 32 microseconds
```

MitchBradley, 2025-08-31:
FluidNC has a fixed debounce period of 5 ms.  The internal pullup can be as high as 100K, which results in a 10 ms time constant.  Using an internal pullup for any signal that is potentially connected to an external device is problematic.  Add 10K pullups to your circuit.

Cyberhornet34, 2025-08-31:
I just tested to add 10K pullups instead of internal one, i get the same issue... 
By the way, when i put reset pin on the opto circuit (who as external 10k pullup), i never get Alarm at startup, but i still get Alarm11 when i trigger reset switch after startup.
I'm going to go crazy !

[image]

bdring, 2025-08-31:
Show the console text on the reset.

I cannot test until late tomorrow.

MitchBradley, 2025-08-31:
We probably need to special-case the check for reset active on restart.

MitchBradley, 2025-08-31:
I think I know what is happening.  It will take me awhile to fix it because I am working on something else right now.

bdring, 2025-09-01:
I did a little testing and the control pins appear to be working as designed.

Try sending the $message/level=debug command. It will show you all switch state changes. This will show you if the state is changing due to noise, etc.

MitchBradley, 2025-09-01:
As I said, I think I know the cause of the problem.  I just need to find time to fix it.  I am doing many other things at the moment.

HTheatre, 2025-09-01:
Hi Bart and Mitch,

I'm the Cyberhornet34's friend. Don't worry Mitch. we red you and understood you are working on many other things at the same time. We'll be patient. So, my post is only for information which could help you and confirm your analisys.

[Here](https://drive.google.com/file/d/1IEEYucnLWkFPEK2qm4CeQsyQsLZRk2mD/view?usp=sharing) is my controller board's schematic.

Here is my config.yaml file.

```
board: SixDeCNC
name: XYZABC external drivers
meta: (09/01/2025) By HTheatre

arc_tolerance_mm: 0.002
junction_deviation_mm: 0.010
verbose_errors: false
report_inches: false
enable_parking_override_control: false
use_line_numbers: false

start:
  must_home: false
  deactivate_parking: false
  check_limits: true

stepping:
  engine: I2S_STREAM
  idle_ms: 250
  pulse_us: 4
  dir_delay_us: 0
  disable_delay_us: 0
  segments: 12

axes:
  shared_stepper_disable_pin: NO_PIN
  x:
    steps_per_mm: 60.000
    max_rate_mm_per_min: 2500.000
    acceleration_mm_per_sec2: 80.000
    max_travel_mm: 550.000
    soft_limits: false
    homing:
      cycle: 2
      allow_single_axis: true
      positive_direction: false
      mpos_mm: 0
      seek_mm_per_min: 2000.000
      feed_mm_per_min: 200.000
      settle_ms: 250
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: gpio.33:low
      hard_limits: false
      pulloff_mm: 2.000
      standard_stepper:
        disable_pin: I2SO.1
        direction_pin: I2SO.3
        step_pin: I2SO.5

  y:
    steps_per_mm: 60.000
    max_rate_mm_per_min: 2500.000
    acceleration_mm_per_sec2: 80.000
    max_travel_mm: 570.000
    soft_limits: false
    homing:
[... 277 lines cut ...]
<Idle|MPos:0.000,0.000,0.000,0.000,0.000,0.000|Bf:15,256|FS:0,0|Pn:Z>
```

And finally, here is the alarm message I get when I press the RESET button. I use momentary and normally closed switches on all the X, Y, Z, A, B, C (endstops), START and HOLD entries.
```

[SixDeCNC_schematic.pdf](https://github.com/user-attachments/files/22083805/SixDeCNC_schematic.pdf)

[SixDeCNC_schematic.pdf](https://github.com/user-attachments/files/22083818/SixDeCNC_schematic.pdf)

Grbl 3.9 [FluidNC v3.9.8 (wifi) '$' for help]
$Report/Interval=300
[MSG:WARN:: Active limit switch on Z axis motor 0]
[MSG:ERR: reset_pin is active at startup]
[MSG:INFO: ALARM: Control Pin Initially On]
ALARM:11
ALARM:11
[MSG:INFO: websocket auto report interval set to 300 ms]
ok
<Alarm|MPos:0.000,0.000,0.000,0.000,0.000,0.000|Bf:15,256|FS:0,0|Pn:Z|WCO:0.000,0.000,0.000,0.000,0.000,0.000>
[GC:G0 G54 G17 G21 G90 G94 M5 M9 T0 F0 S0]
<Alarm|MPos:0.000,0.000,0.000,0.000,0.000,0.000|Bf:15,256|FS:0,0|Pn:Z|Ov:100,100,100>
```

Thank you for all the fantastic job you do guys!