bdring/FluidNC #1558: Feature: Disable Motor/Stepper automatically after fault/estop/hard stop

Gerporgl, 2025-09-04:
### Machine Context

I could not find an way to disable steppers / motors after a fault_pin/estop_pin or reset_pin. I am trying to reproduce the same behavior I currently have on my existing machine with a mach3 board but it seems that fluidnc keep motors enabled all the time even after a fault.

### Feature Description

 I think there should be an option to disable motors after a fault. I also want to keep idle_ms to 255 (disabled) for normal usage (that is also generally recommended and my current behavior in my existing machine).

### Other Approaches

I tried briefly to create a macro that would send an alert to so a hardstop, then disable the motors, but you can't send disable motor unless you soft reset, and I could not find a way to soft reset easily in a macro and do all of that, plus it started to look messy and the wrong approach.

### How I Can Help

I bought a corgi board and a 6x, I hope this helps the project

MitchBradley, 2025-09-04:
Have you tried http://wiki.fluidnc.com/en/features/commands_and_settings#motordisable-or-md

Gerporgl, 2025-09-04:
I tried, but whenever I try to do it after a hardstop of estop or failsafe, it can't do it, it says "Grbl '$' command cannot be used unless Grbl is IDLE. Ensures smooth operation during a job." Then I need to do a softreset, but I am not sure how that can be done in a macro (is there a way to send 0x18 (CTRL+X) in a macro?)

Basically, what I'm trying to do is a failsafe that disables motors right after it stopped and disabled the spindle.