bdring/FluidNC #1560: Feature: Switch/remap driver for axis via command

E1yot, 2025-09-07:
### Machine Context

Laser with X, Y and Z-Axis, 2 different rotary units, independent Diode- and CO2 laser head

### Feature Description

It would be nice to be able to switch the axis of a stepper motor driver via command.
Use cases:
- Rotary unit: Lightburn supports rotary units on the Y, Z, and A axes. My machine has Y and Z, so the rotary unit goes on the A axis. Since I have two different rotary units, I can assign two stepper motors to the A axis or switch the cable from one to the other. In the first case, both rotary units move during operation, which isn't the best solution. A command to switch the axis assignment between stepper motor 5 and stepper motor 6 would be nice.
- Independent driver for two tools. I plan to add an independent second head to my machine. Only one head is used at a time. It would be nice to be able to configure independent stepper motors for an axis and activate the correct one according to the selected tool.

### Other Approaches

I can do this using physical switches on the machine. However, a command can automatically select the correct path.

### How I Can Help

I can help with testing.

bdring, 2025-09-07:
Create 2 separate config files. Use the $Config/Filename command to change between them.

E1yot, 2025-09-07:
Oh, I did not know this is possible. So this feature request can be closed.

MitchBradley, 2025-09-07:
no

closed 2025-09-09 by bdring