bdring/FluidNC #1577: Problem: Non constant G1 motion

DLR3D, 2025-09-27:
### Wiki Search Terms

Non uniform G1 motions

### Controller Board

Rodent

### Machine Description

6040Z CNC with ballscrews on all axis, nema 23 motors. No endstops, using G10 L20 P1 to set the position before motion.

### Input Circuits

```yaml

```

### Configuration file

```yaml
board: BTT Rodent-Z V1.0
name: 6040Z
meta: DR 6040z 1205 ballscrew

kinematics:
  Cartesian:

stepping:
  engine: I2S_STREAM
  idle_ms: 255
  pulse_us: 4
  dir_delay_us: 1
  disable_delay_us: 0

axes:
  shared_stepper_disable_pin: NO_PIN
  x:
    steps_per_mm: 640.000
    max_rate_mm_per_min: 1500.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 480.000
    soft_limits: false
    homing:
      cycle: 1
      positive_direction: false
      mpos_mm: 0
      feed_mm_per_min: 400.000
      seek_mm_per_min: 800.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.500

    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 3.000
      tmc_5160:
        step_pin: I2SO.2
        direction_pin: I2SO.1
        disable_pin: I2SO.0
        cs_pin: gpio.5
        spi_index: 1
        r_sense_ohms: 0.075
        run_amps: 2.0
        hold_amps: 1.5
        microsteps: 16
        toff_disable: 0
        toff_stealthchop: 5
        use_enable: false
        run_mode: CoolStep
        homing_mode: CoolStep
        stallguard: 16
        stallguard_debug: false
        toff_coolstep: 3
        tpfd: 4

  y:
    steps_per_mm: 640.000
    max_rate_mm_per_min: 1500.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 650
    soft_limits: false
    homing:
      cycle: 1
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 400.000
      seek_mm_per_min: 800.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.500

    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 3.000
      tmc_5160:
        step_pin: I2SO.5
        direction_pin: I2SO.4:low
        disable_pin: I2SO.7
        cs_pin: NO_PIN
        spi_index: 2
        r_sense_ohms: 0.075
        run_amps: 2.0
        hold_amps: 1.5
        microsteps: 16
        toff_disable: 0
        toff_stealthchop: 5
        use_enable: false
        run_mode: CoolStep
        homing_mode: CoolStep
        stallguard: 16
        stallguard_debug: false
        toff_coolstep: 3
        tpfd: 4

  z:
    steps_per_mm: 640.000
    max_rate_mm_per_min: 1500.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 150.000
    soft_limits: false
    homing:
      cycle: 1
      positive_direction: true
      mpos_mm: 0
      feed_mm_per_min: 400.000
      seek_mm_per_min: 800.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.500

    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 3.000
      tmc_5160:
        step_pin: I2SO.10
        direction_pin: I2SO.9:low
        disable_pin: I2SO.8
        cs_pin: NO_PIN
        spi_index: 3
        r_sense_ohms: 0.075
        run_amps: 2.0
        hold_amps: 1.5
        microsteps: 16
        toff_disable: 0
        toff_stealthchop: 5
        use_enable: false
        run_mode: CoolStep
        homing_mode: CoolStep
        stallguard: 16
        stallguard_debug: false
        toff_coolstep: 3
        tpfd: 4

  a:  #Not Used but needs to be defined so all stepper work
    # E0
    steps_per_mm: 157.750
    max_rate_mm_per_min: 18000.000
    acceleration_mm_per_sec2: 1500.000
    max_travel_mm: 325.000
    soft_limits: false
    homing:
      cycle: -1
      positive_direction: true
      mpos_mm: 150.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 800.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1.000
      tmc_5160:
        step_pin: I2SO.13
        direction_pin: I2SO.12
        disable_pin: I2SO.15
        cs_pin: NO_PIN
        spi_index: 4
        r_sense_ohms: 0.022
        run_amps: 0.5
        hold_amps: 0.5
        microsteps: 16
        toff_disable: 0
        toff_stealthchop: 5
        use_enable: false
        run_mode: CoolStep
        homing_mode: CoolStep
        stallguard: 16
        stallguard_debug: false
        toff_coolstep: 3
        tpfd: 4

i2so:
  bck_pin: gpio.22
  data_pin: gpio.21
  ws_pin: gpio.17

spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

sdcard:
  cs_pin: gpio.0
  card_detect_pin: NO_PIN
  frequency_hz: 8000000
# I2SO.3

# coolant:
#   flood_pin: gpio.2:high
#   mist_pin:  gpio.3:low
# gpio.4 # HB MOS

# probe:
#   pin: gpio.5:low:pu
#   toolsetter_pin: gpio.14:low

# Begin Huanyang
#uart1:
#  txd_pin: gpio.15
#  rxd_pin: gpio.16
#  rts_pin: gpio.14
#  baud: 9600
#  mode: 8N1

# Huanyang:
#  uart_num: 1
#  modbus_id: 1
#  tool_num: 0
#  speed_map: 0=0% 8000=22% 10000=27% 15000=40% 2000=54% 24000=63%   
#  off_on_alarm: false

PWM:
   pwm_hz: 5000
   output_pin: gpio.13
   enable_pin: gpio.04:high
   direction_pin: NO_PIN
   disable_with_s0: true
   s0_with_disable: true
   spinup_ms: 0
   spindown_ms: 0
   tool_num: 0
   speed_map: 0=0% 6000=19% 12000=39% 18000=60% 24000=100%  
   off_on_alarm: false

i2c0:
   sda_pin: gpio.27
   scl_pin: gpio.26

status_outputs:
  report_interval_ms: 500
  idle_pin: I2SO.11:low
  run_pin: I2SO.14:low
  alarm_pin: I2SO.3:low
# hold_pin: gpio.16
```

### Startup Messages

```text
the only message that shows up when sending "$ss" is the following:
ok
```

### User Interface Software

UGS

### What happened?

Hello I am using FluidNC on a Rodent motherboard, all works as expected except G1 moves across long distances.

For example starting at X200 Z0 Y0
If I send a "G1 X-200 F1000" the motor does not move in a uniform speed as it does with G0, the motor seems to be speeding up and down arround F1000, using UGS I can see the feedrate oscillating between 800 and 1200.

Video showing.
Spoken words translated to english:
"It goes slower then faster, you can also see it on the moving cutting head"

This behavior seems to be causing chatter when cutting metals.
Any idea what can be done to solve this weird behavior? 

Using the latest FluidNc installed from the website.
Config of the axis is set to 100 accel with max speed set to 2000 mm/min.

### GCode File

Manual Gcode sent
G1 X-200 F1000
G1 X200 F1000

bdring, 2025-09-27:
Use the terminal on the Web Installer to restart FluidNC and get the start messages.

bdring, 2025-09-28:
I have it running now. It seems pretty stable to me. I do not see any change in the feed rate and the motor sound is consistent.

 Try using I2S_STATIC for the stepping engine.

MitchBradley, 2025-09-28:
I do not know why UGS is suppressing the output from $ss, but we really need to see it.  Please connect with FluidTerm or WebInstaller and get the $ss results.